synthetos/g2 #327: canvas's width  of G2 chart turn to 0px

yulongWanghb, 2018-04-02:
when using G2 chart to draw path picture ,it shows normal at the first time ,but when i use jquery function  (hide、show ) ，i  find the canvas width become 0 px ,and the data showed in the picture huddled together，i try to change the width of it and failed ,this happened sometimes

yulongWanghb, 2018-04-02:
version :2.3.13
platform :windows

giseburt, 2018-04-02:
I believe you have the wrong project. This is a CNC machine firmware called
g2core.

yulongWanghb, 2018-04-03:
I think i should describle the  clearly ,i draw a path chart with G2 chart successfully in a div mark of html ,it  workes ,and  i have to hide it  by useing jquery function ($.hide（）)，but when i use ($.show ) to show the div ,the chart doesn't work , when i debug the code in chrome ,i find the width of canvas becomes 0px ,i am confused and don't know how to fix it ,can you help me with my puzzle?

giseburt, 2018-04-03:
This isn’t a web project. This is not “G2 chart.” I’m sorry, but you have
the wrong project.

yulongWanghb, 2018-04-03:
ok ,whatever thank you very much

yulongWanghb, 2018-04-03:
I 'm very sorry for my mistake ,i do have the wrong project . i intended to search a project named antivs/G2

closed 2018-04-08 by yulongWanghb