synthetos/g2 #330: documentation bug: G61/G61.1

rkoe, 2018-04-12:
The documentation is inconsistent:

- https://github.com/synthetos/g2/wiki/Gcodes says:
  G61    Exact stop mode
  G61.1 Exact path mode
- https://github.com/synthetos/g2/wiki/Configuring-0.99-System-Groups says:
  0=Exact path mode (G61)
  1=Exact stop mode (G61.1)

Which one is right?

justinclift, 2018-04-12:
Good catch @rkoe. :smile:

If no-one else gets to this sooner, I'll take a look through the source later on today to see if I can figure it out.

justinclift, 2018-04-12:
As a data point, the Consensus G-Code page (seems to be an assembly of G-Codes from several vendors, for reference) has it as:

* G61 - Exact Path Mode
* G61.1 - Exact Stop Mode

That's without checking the actual source or machine behaviour though, which I still need to get done. :smile:

justinclift, 2018-04-12:
Hmmm, this is probably going to be a task for tomorrow instead.  Sorry @rkoe. :frowning:

giseburt, 2018-04-12:
I can confirm that in [current edge](https://github.com/synthetos/g2/blob/edge/g2core/gcode_parser.cpp#L750-L751) this is correct with the concensus gcode, and that the documentation [here](https://github.com/synthetos/g2/wiki/Gcodes) is wrong.

```c++
                case 61: {
                    switch (_point(value)) {
                        case 0: SET_MODAL (MODAL_GROUP_G13, path_control, PATH_EXACT_PATH);
                        case 1: SET_MODAL (MODAL_GROUP_G13, path_control, PATH_EXACT_STOP);
                        default: status = STAT_GCODE_COMMAND_UNSUPPORTED;
                    }
                    break;
                }
```

justinclift, 2018-04-12:
Thanks @giseburt.  Just updated [that wiki page](https://github.com/synthetos/g2/wiki/Gcodes) in the various places with the corrected descriptions.

@rkoe Would you be ok to review the update, and make sure it's ok now? :smile:

justinclift, 2018-04-18:
@rkoe ping?

Hoping you have time to verify the change.  Just as a "Quality Assurance" step mind you. :smile:

rkoe, 2018-04-18:
Looks consistent now. :)
But I haven't tested it with a real machine, yet.

justinclift, 2018-04-18:
Thanks @rkoe, it was just the 2nd set of eyes to verify the doc change is all I was looking for.

I'll close this now as the doc update is done + verified.  If it turns out that the code **doesn't** actually work that way for some reason, we can make another issue out of it and figure out wtf then. 😀

closed 2018-04-18 by justinclift

aldenhart, 2018-04-18:
I'm pretty sure the code works correctly. I just checked the parse. This was probably a documentation error - caused by me. Thanks for chasing it down.