synthetos/g2 #331: feedhold does not work for G2/G3

rkoe, 2018-04-12:
Feedhold (!) and resume (~) work well for G0 and G1, but feedhold does not work for G2/G3 in the current stable version (100.26): I can send !, but the motors simply continue to run.

justinclift, 2018-04-12:
That doesn't sound good at all. :wink:

rkoe, 2018-04-18:
I tested it again with minicom, and this time it worked.
I'll do more tests, try to reproduce the error and report back.

aldenhart, 2018-04-18:
It was tested on PR#305 (UVW) and it works as expected. That build also supports Z lift on feedhold. Try it out.

PR#305 is now in edge.

rkoe, 2018-04-25:
Tested again, and it works.
Turned out that the user-interfaces (cnc.js, Goko, chilipepper) are bad and break the feedhold. With direct serial communication, it works fine.

rkoe, 2018-04-27:
Now I think I've tracked it down:
On low ct and lower xjm, feedhold does not work correctly during G2:
- $ct=0.01, $xjm=1000: works
- $ct=0.01, $xjm=50: works
- $ct=0.001, $xjm=1000: works
- $ct=0.001, $xjm=50: fails. Sending a feedhold makes g2core to switch to machine state 6 (HOLD), but the machine does not stop.

(I've used $ct=0.001, since this creates nice smooth arcs; with $ct=0.01 I literally heared the line segments from the stepper motors ;))

Minimal, reproducible example (running after g2core-reset):
```
$xam=1
$xjm=50
$ct=0.001
N001 G0 X0 Y0
N002 F500
N003 G2 X100 Y0 I50 J0
N004 M2
```

Communication logs:
- ct=0.01, xjm=50: OK
    ```
    {"r":{"fv":0.99,"fb":100.26,"fbs":"100.26-10-g1ff7","fbc":"settings_default.h","hp":3,"hv":0,"id":"0084-e739-18c6-08c","msg":"SYSTEM READY"},"f":[1,0,1]}
    $xam=1
    {"r":{"am":1},"f":[1,0,7]}
    $ct=0.001
    {"r":{"ct":0.001},"f":[1,0,10]}
    {"sr":{"stat":t,"n":t,"line":t,"posx":t,"posy":t,"posz":t,"mpox":t,"mpoy":t,"mpoz":t,"feed":t,"vel":t,"unit":t,"coor":t,"momo":t,"plan":t,"path":t,"dist":t,"admo":t,"frmo":t,"macs":t,"cycs":t,"mots":t,"hold"
    {"r":{"sr":{"stat":1,"line":0,"posx":0,"posy":0,"posz":0,"mpox":0,"mpoy":0,"mpoz":0,"feed":0,"vel":0,"unit":1,"coor":1,"momo":4,"plan":0,"path":2,"dist":0,"admo":1,"frmo":1,"macs":1,"cycs":0,"mots":0,"hold":
    {"sr":{"stat":1,"n":0,"line":0,"posx":0,"posy":0,"posz":0,"mpox":0,"mpoy":0,"mpoz":0,"feed":0,"vel":0,"unit":1,"coor":1,"momo":4,"path":2,"admo":1,"frmo":1,"macs":1}}
    N001 G0 X0 Y0
    N002 F500
    N003 G2 X100 Y0 I50 J0
    N004 M2
    {"r":{},"f":[1,0,14]}
    {"r":{},"f":[1,0,10]}
    {"r":{},"f":[1,0,23]}
    {"sr":{"stat":5,"n":3,"line":3,"posx":0.025,"posy":1.54,"mpox":0.025,"mpoy":1.54,"feed":500,"vel":500,"momo":2,"macs":5,"cycs":1,"mots":2}}
    {"sr":{"posx":0.132,"posy":3.623,"mpox":0.132,"mpoy":3.623}}
    {"sr":{"posx":0.326,"posy":5.7,"mpox":0.326,"mpoy":5.7}}
    {"sr":{"posx":0.608,"posy":7.768,"mpox":0.608,"mpoy":7.768}}
    {"sr":{"posx":0.975,"posy":9.821,"mpox":0.975,"mpoy":9.821}}
    {"sr":{"posx":1.43,"posy":11.87,"mpox":1.43,"mpoy":11.87}}
    {"sr":{"posx":1.968,"posy":13.886,"mpox":1.968,"mpoy":13.886}}
    {"sr":{"posx":2.589,"posy":15.877,"mpox":2.589,"mpoy":15.877}}
    {"sr":{"posx":3.297,"posy":17.853,"mpox":3.297,"mpoy":17.853}}
    {"sr":{"posx":4.082,"posy":19.786,"mpox":4.082,"mpoy":19.786}}
    {"sr":{"posx":4.948,"posy":21.684,"mpox":4.948,"mpoy":21.684}}
    {"sr":{"posx":5.891,"posy":23.556,"mpox":5.897,"mpoy":23.556}}
    {"sr":{"posx":6.918,"posy":25.375,"mpox":6.918,"mpoy":25.375}}
    {"sr":{"posx":8.014,"posy":27.15,"mpox":8.014,"mpoy":27.15}}
    !
    {"sr":{"stat":6,"posx":9.189,"posy":28.886,"mpox":9.189,"mpoy":28.886,"vel":491.98,"mots":3,"hold":3}}
    {"sr":{"posx":9.551,"posy":29.39,"mpox":9.551,"mpoy":29.39,"vel":0,"hold":7}}
    ```
- ct=0.001, xjm=1000: OK
    ```
    {"r":{"fv":0.99,"fb":100.26,"fbs":"100.26-10-g1ff7","fbc":"settings_default.h","hp":3,"hv":0,"id":"0084-e739-18c6-08c","msg":"SYSTEM READY"},"f":[1,0,1]}
    $xam=1
    {"r":{"am":1},"f":[1,0,7]}
    $xjm=50
    {"r":{"jm":50},"f":[1,0,8]}
    {"sr":{"stat":t,"n":t,"line":t,"posx":t,"posy":t,"posz":t,"mpox":t,"mpoy":t,"mpoz":t,"feed":t,"vel":t,"unit":t,"coor":t,"momo":t,"plan":t,"path":t,"dist":t,"admo":t,"frmo":t,"macs":t,"cycs":t,"mots":t,"hold"
    {"r":{"sr":{"stat":1,"line":0,"posx":0,"posy":0,"posz":0,"mpox":0,"mpoy":0,"mpoz":0,"feed":0,"vel":0,"unit":1,"coor":1,"momo":4,"plan":0,"path":2,"dist":0,"admo":1,"frmo":1,"macs":1,"cycs":0,"mots":0,"hold":
    {"sr":{"stat":1,"n":0,"line":0,"posx":0,"posy":0,"posz":0,"mpox":0,"mpoy":0,"mpoz":0,"feed":0,"vel":0,"unit":1,"coor":1,"momo":4,"path":2,"admo":1,"frmo":1,"macs":1}}
    N001 G0 X0 Y0
    N002 F500
    N003 G2 X100 Y0 I50 J0
    N004 M2
[... 100 lines cut ...]
    {"sr":{"posx":78.022,"posy":41.409,"mpox":78.022,"mpoy":41.409,"vel":459.63}}
    {"sr":{"posx":79.632,"posy":40.273,"mpox":79.632,"mpoy":40.273,"vel":479.17}}
    {"sr":{"posx":81.197,"posy":39.073,"mpox":81.197,"mpoy":39.065,"vel":478.74}}
    {"sr":{"posx":82.713,"posy":37.812,"mpox":82.722,"mpoy":37.805,"vel":476.9}}
    {"sr":{"posx":84.178,"posy":36.493,"mpox":84.187,"mpoy":36.485,"vel":473.68}}
    {"sr":{"posx":85.598,"posy":35.11,"mpox":85.598,"mpoy":35.11,"vel":469.61}}
    {"sr":{"posx":86.953,"posy":33.68,"mpox":86.953,"mpoy":33.68,"vel":465.47}}
    {"sr":{"posx":88.251,"posy":32.199,"mpox":88.251,"mpoy":32.199,"vel":462.11}}
    {"sr":{"posx":89.503,"posy":30.65,"mpox":89.503,"mpoy":30.65,"vel":459.84}}
    {"sr":{"posx":90.673,"posy":29.081,"mpox":90.673,"mpoy":29.081,"vel":444.35}}
    {"sr":{"posx":91.743,"posy":27.523,"mpox":91.743,"mpoy":27.523,"vel":428.63}}
    {"r":{},"f":[1,0,8]}
    {"sr":{"n":4,"posx":92.731,"posy":25.961,"mpox":92.737,"mpoy":25.953,"vel":414.04}}
    {"sr":{"posx":93.634,"posy":24.412,"mpox":93.634,"mpoy":24.412,"vel":405.64}}
    {"sr":{"posx":94.466,"posy":22.862,"mpox":94.466,"mpoy":22.862,"vel":409.25}}
    {"sr":{"posx":95.215,"posy":21.344,"mpox":95.215,"mpoy":21.344,"vel":421.67}}
    {"sr":{"posx":95.897,"posy":19.834,"mpox":95.897,"mpoy":19.834,"vel":365.96}}
    {"sr":{"posx":96.518,"posy":18.33,"mpox":96.522,"mpoy":18.321,"vel":396.16}}
    {"sr":{"posx":97.07,"posy":16.864,"mpox":97.07,"mpoy":16.864,"vel":338.09}}
    {"sr":{"posx":97.568,"posy":15.401,"mpox":97.568,"mpoy":15.401,"vel":394.39}}
    {"sr":{"posx":98.011,"posy":13.96,"mpox":98.011,"mpoy":13.96,"vel":329.86}}
    {"sr":{"posx":98.388,"posy":12.594,"mpox":98.388,"mpoy":12.594,"vel":281.62}}
    {"sr":{"posx":98.717,"posy":11.255,"mpox":98.717,"mpoy":11.255,"vel":393.23,"hold":4}}
    {"sr":{"posx":98.842,"posy":10.696,"mpox":98.842,"mpoy":10.696,"vel":0,"hold":7}}
    ```

amx1, 2018-05-28:
I don't have these issues with 101.02-17-g51f9-dirty, even G2/G3 pause / resume is working with CNCJS 1.9.15 latest master