synthetos/g2 #336: init: coolant not initialized, no early I/O-init

rkoe, 2018-04-18:
There are two issues with I/O-initialization:

1. **Coolant:** The coolant is not initialized. So, when setting the coolant to low-active, the coolant is on at power-on.

2. **late I/O-init:** The I/Os  are initialized **after** waiting 400ms for USB. So, for default-low-outputs, this results in a power-on-spike of ~0.4s. It would be better to **first** init the I/Os ("early I/O-preinit") to make this spike as short as possible, and then wait for USB.

Fixes:

1. Coolant: Add the following lines to main.cpp:
```
  #include "spindle.h"
+ #include "coolant.h"
  #include "temperature.h"

  ...

  void application_init_machine(void)
  {
      ...
      spindle_reset();
+     coolant_init();
+     coolant_reset();
      temperature_init();
```

2. early I/O-init: Initialize all I/Os directly after power-on, by moving some init-function-calls from
application_init_machine() and application_init_startup() to an earlier location. **I don't know the code in full detail, so I don't know if this has any side-effects, so some core-developer should check before merging.** My suggestion for main.cpp:
```
+ void application_init_io_early(void)
+ {
+     config_init();
+     gpio_init();
+     gpio_reset();
+     spindle_init();
+     spindle_reset();
+     coolant_init();
+     coolant_reset();
+ }

+ }
  void setup(void)
  {
      // application setup
      application_init_services();
+     // early I/O-init
+     application_init_io_early();
+
      while (SysTickTimer_getValue() < 400);  // delay 400 ms for USB to come up

      application_init_machine();
      application_init_startup();
  }
```

justinclift, 2018-04-19:
As a data point, the PR to address the first bit is in #339. :smile:

giseburt, 2018-05-06:
PR #339 has been closed. What else is needed for this issue?

justinclift, 2018-05-07:
The second part to the problem still needs thinking about, and probably some code changes (hopefully minor):

> The I/Os are initialized **after** waiting 400ms for USB. So, for default-low-outputs, this results in a power-on-spike of ~0.4s. It would be better to **first** init the I/Os ("early I/O-preinit") to make this spike as short as possible, and then wait for USB.

Potential solution:

> Initialize all I/Os directly after power-on, by moving some init-function-calls from
application_init_machine() and application_init_startup() to an earlier location.

Potential solution code from @rkoe (main.cpp):

```
+ void application_init_io_early(void)
+ {
+     config_init();
+     gpio_init();
+     gpio_reset();
+     spindle_init();
+     spindle_reset();
+     coolant_init();
+     coolant_reset();
+ }

+ }
  void setup(void)
  {
      // application setup
      application_init_services();
+     // early I/O-init
+     application_init_io_early();
+
      while (SysTickTimer_getValue() < 400);  // delay 400 ms for USB to come up

      application_init_machine();
      application_init_startup();
  }
```

Note that @rkoe mentions since he doesn't know the codebase in depth, he'd really like someone familiar with it to chime in.  If the above code snippet seems ok, he can then turn it into a PR for merging.

Also note that if @rkoe doesn't have time creating said PR, I can do it pretty easily for him too to save time. :grinning:

commit 636781db (2018-05-06): Fix coolant-init (partly fixes #336)
commit 8220e349 (2018-05-06): Merge pull request #339 from justinclift/dev-336-fix-coolant-init_v2

Fix coolant-init (partly fixes #336)