synthetos/g2 #340: Releases page is missing .elf/.bin for gShield

justinclift, 2018-04-19:
This is mostly just a reminder for myself, to add the gShield .elf and .bin to our Releases page.

Mainly because for new users it'll provide them a .bin they can use "as is" without having to go through any of the compiling/development steps.

That's preferable, because this will get beginners to "Motors are moving!" earlier on in the process.  Better for motivation. :smile:

giseburt, 2018-04-19:
BTW: That would be done by editing the travis file. TravisCI automatically pushes builds to the Releases pages.

https://github.com/synthetos/g2/blob/master/.travis.yml

justinclift, 2018-04-19:
Thanks @giseburt.  That looks really straightforward. :smile:

closed 2023-02-17 by ril3y