synthetos/g2 #344: (msg...) / gcode messages

rkoe, 2018-04-26:
The Gcode-wiki-page (https://github.com/synthetos/g2/wiki/Gcodes) lists a "gcode message": (msg ...)

> Gcode messages are comments that begin with the characters msg
> (case insensitive). These will be echoed to the operator.

How are they echoed to the operator?
g2core does not send any message; so is this the task of the g-code-sender, and should the wiki-text be modified to:

>  Gcode messages are comments that begin with the characters msg
> (case insensitive). The g-code-sender should echo them to the operator.

Example:
```
G0X0
(msg test message)
G0X100
```

giseburt, 2018-04-26:
I believe you've found a bug. You should get back: {r:{msg:"test message"}, f:[…]}

-Rob

rkoe, 2018-04-26:
But when should I get this back?

- when executed: I would expect to get it back when this line is executed in g2core. But then, `{r:{msg:...}}` looks wrong and would break line mode.
- when sent: This would fit to `{r:{msg:...}}`, but then it would break the g-code-sequence.

giseburt, 2018-04-26:
@rkoe That's a valid question. I believe it should be in the {r:...} for the line itself, so it would be when queued, but not in line with motion.

aldenhart, 2018-04-27:
You may be right that the message is returned on parsing - I have not checked. But think the more correct behaviour is to queue the message response to the motion, then send the message synchronized with the motion. This way it supports messages to the operator to do things like change tools, turn things on and off along with commands like M0 or M1. 

If that's the case, the message would need to be returned "out of band", such as in a status report or exception report. Or a new OOB category simply called "message".

MitchBradley, 2018-07-11:
I don't think that the message is returned at all.  There is a function cm_message() that adds a message to the response object, but that function is not called anywhere that I can find.