synthetos/g2 #345: line mode questions / bugs?

rkoe, 2018-04-26:
g2core recommends to use the line mode (https://github.com/synthetos/g2/wiki/g2core-Communications). But this seems to work only partially:

- "Every command will return one and only one response." is wrong.
  Some command return more then one {"r":...}.
  

  E.g.:
  ```
  {"m": n}
  {"q": n}
  {"o": n}
  $$
  ```

  Are there more? And how can this be fixed/worked around?

- lines_to_send should probably also reset after sending a %.
  Correct? If this is correct, then the wiki-page should be updated.

- Directly after connecting to g2core via USB, I get a {r:...} message.
  This should probably be ignored for the lines_to_send-calculation.

  Example:
  ```
  {"r":{"fv":0.99,"fb":100.26,...}
  ```

- Could {"qr":n} be used as alternative to line mode?

- Is it intensional, that {"qr":n}-answers are sometimes delayed very long, especially during G2/G3? (I noticed ~20 seconds, sometimes).

giseburt, 2018-04-26:
> "Every command will return one and only one response." is wrong.
> Some command return more then one {"r":...}
> 
> E.g.:
> 
> {"m": n}
> {"q": n}
> {"o": n}
Ah, you've found the uber-groups. Those are shortcuts for calling multiple other calls, and should be mentioned in the wiki as exceptions to this rule. Currently, they should be specifically avoided in line mode without special consideration, as they do not have a clear way to know when they are done.

{m:n} can be replaced with
{m1:n}
{m2:n}
{m3:n}
{m4:n}
{m5:n}

Similar for q and o.

We have discussed having the uber-groups output a special response as the last response so that you can tell when it's done. Perhaps an empty non-error response, or such. {"$":n} in particular is very useful for inspection of a machine, but you have to call it in isolation and have a timeout to determine when it's done sending, which is not an ideal interaction (in my opinion.)
> $$
Non-JSON (anything starting with a $) is not considered part of line-mode, as that is for human interaction.
> Are there more? And how can this be fixed/worked around?
In fact, there are a few more uber groups. Some may be only in edge, however:

{"di":n}
{"do":n}
{"$":n} ( the JSON form of "$$")
> lines_to_send should probably also reset after sending a %.
Yes, along with recovering from an alarm state with {clr:n} - those are also omission on the wiki, and should be corrected.
> Correct? If this is correct, then the wiki-page should be updated.
> 
Indeed. Thank you for pointing these out.
> Directly after connecting to g2core via USB, I get a {r:...} message.
> This should probably be ignored for the lines_to_send-calculation.
> 
> Example:
> 
> {"r":{"fv":0.99,"fb":100.26,...}
This is the startup message, and is the client's indication that the machine is alive and responding, as well as giving vital information about what protocol it speaks. This also should have been mentioned on the wiki.

Once this message comes in, then line mode can begin.
> Could {"qr":n} be used as alternative to line mode?
> 
> Is it intensional, that {"qr":n}-answers are sometimes delayed very long, especially during G2/G3? (I noticed ~20 seconds, sometimes).

The qr information is the deprecated earlier attempt at what became line-mode. The qr mode has a few fatal flaws, as you pointed out one of them.

Thank you for these notes.

-Rob

rkoe, 2018-04-27:
Thanks for clarification.

> We have discussed having the uber-groups output a special response as the last response so that you can tell when it's done.

Hmm, but that probably doesn't help with line-mode, since I guess ueber-group-responses and other responses could be mixed.
**Or could they?**
If not -- so, if the responses are _always in the same order as the requests_ -- I could determine the uber-group-response-end by sending an additional request directly after the uber-group-request and wait for its answer. (e.g. send `{"fb":n}` and wait for `{"r":{"fb"...`)

But I would prefer an additional _1st_ answer which tells me how many more responses to expect. Then I could simply do `lines_to_send -= RECEIVED_NUMBER`, and line mode would be fine.

> Non-JSON (anything starting with a $) is not considered part of line-mode, as that is for human interaction.

It would be nice to be able to mix JSON and text mode commands for debugging or for doing special things in minimal user-interfaces. But this seems to work nice already, since I can use text-mode-commands in JSON-mode (ej=1), without breaking line-mode (except for the uber-groups, which need special handling here as well).

giseburt, 2018-05-13:
So we are considering a few alterations to the response format to indicate that the response is not done.

Basically the concept would be that a reply somehow indicates that it continues to the next reply line. The host software would then simply not alter `lines_to_send` in that case and know the the next response line is to be merged with this one.