synthetos/g2 #348: Motors should remain powered during feedhold

aldenhart, 2018-05-14:
Currently motors will become unpowered if their power timeout expires during a feedhold. This should not happen. Motors should remain in the same powered state (on/off) that they entered feedhold for the duration of the feedhold.

giseburt, 2018-05-14:
This brings up an interesting question: Should feedhold have a safety timeout in this case?

lutorm, 2018-10-09:
I had a related question: Is it possible to define motors that are normally always-powered but that are disabled after a machine stop?

I'm now using 3 ClearPath servos and have them set to always powered so the system doesn't lose position. If any of the three motors fault (can't keep position, torque saturation, etc) I assert a FAST_STOP since having one motor stop while the others continue would be a recipe for disaster. Ideally in this situation I'd like g2 to *disable* the motors after coming to a stop. It seems like it would be useful to be able to define a power management setting specific to feedholds.

aldenhart, 2018-10-09:
@lutorm - Presumably this means they are disabled for this following conditions: M2, M30, as well as alarm and shutdown conditions; but not for in-program stops such as M0, M1, or feedholds. Is this what you are thinking?

lutorm, 2018-10-09:
I'm not really sure how I want it to work... ;-)

I can see how you would want to keep the motors powered in a feedhold, because you want to keep position. The stop signal I'm sending seems more like an "alarm" than a "feedhold", but I *do* want a controlled stop because I want to be able to reset the motor and resume without the machine losing position.

I think normally I want the motors to be powered, even after a program end, so the machine keeps position. It's just when you stop due to an abnormal condition that it seems more prudent to power them down.