synthetos/g2 #349: Possible bug with 101.02 - error "Soft limit - X min"

justinclift, 2018-05-17:
Found what **seems** like a bug with 101.02 (current edge) today, when trying to cut out a ~reasonably simple circle, diamond, square test with machining wax.

[image]
[image]

* With 11.02, the job stops with "Soft limit - X min" or "Soft limit - Z min"
  * Ran the job 3 times.  Twice gave the "Soft limit - Z min" error.  Once gave the "Soft limit - X" error.
* With 100.26 (current master), the job runs through to completion
  * Ran the job twice.  Both times ran from start to finish without issue.

Name|Value
---|---
Host computer|Windows 8 x64
G-Code sender|CNCjs 1.9.15 (Windows x64 desktop application version)
CNC hardware| Shapeoko 3, with g2core replacing the stock electronics/controller
g2core hardware| Ardunio Due with simple self built controller (4x A4988 stepper drivers)
g2core config| [settings_shapeoko3.h](https://github.com/justinclift/g2/blob/b9158edc2fd9f918f83a1f67a7d42f0d88763ba8/g2core/settings/settings_shapeoko3.h) (it's the shapeoko2 dual Y one, with [some minor adjustments](https://github.com/justinclift/g2/commit/b9158edc2fd9f918f83a1f67a7d42f0d88763ba8))
G-Code file| [circle diamond square - 10mm end mill.txt](https://github.com/synthetos/g2/files/2014034/circle.diamond.square.-.10mm.pass.txt)

This error seems especially weird, because as far as I can tell both soft and hard limits are disabled in my [settings config](https://github.com/justinclift/g2/blob/b9158edc2fd9f918f83a1f67a7d42f0d88763ba8/g2core/settings/settings_shapeoko3.h#L42-L43).

After the first run failed (using 11.02), I suspected it was potentially due to static electricity buildup or similar due to my crappy wiring/soldering.  So, reran things with the bench power supply not providing current to the motors.  Error still occured.  Not a static electricity problem after all.

It turned out to be a bit more complex to reproduce than the above sounds.  The machine needs to be physically homed first (eg with motors given power), then jogged into the starting position (X: 45, Y 385, Z: 4) then setting G54 0,0,0 at that point.  After this, power to the motors can be turned off as the error will still occur.

If the G-code file is sent to g2 without the above physical homing + G54 setup dance, then no error shows up. (this was a bit time consuming to work out)

Logs of the errors below
---
### Initial error log
Firmware 101.02 - commit 194c9ab

Machine starting co-ordinates unknown.  May not have been X: 45, Y 385, Z: 4

I **think** the error occurred roughly 5 or 6 mins into the job.  Wasn't watching carefully enough at the time though.

```
...
{"sr":{"posz":1.24815,"vel":702.87,"dist":1,"stat":5}}
{"qr":48}
{"sr":{"posz":2,"vel":0,"dist":0,"stat":3}}
feeder> G91
{"r":{},"f":[1,0,4]}
feeder> G0 Z1
{"r":{},"f":[1,0,6]}
feeder> G90
{"qr":47}
{"r":{},"f":[1,0,4]}
{"sr":{"posz":2.24815,"vel":702.87,"dist":1,"stat":5}}
{"qr":48}
{"sr":{"posz":3,"vel":0,"dist":0,"stat":3}}
feeder> G91
{"r":{},"f":[1,0,4]}
feeder> G0 Z1
feeder> G90
{"r":{},"f":[1,0,6]}
{"qr":47}
{"r":{},"f":[1,0,4]}
{"sr":{"posz":3.40021,"vel":794.93,"dist":1,"stat":5}}
{"qr":48}
{"sr":{"posz":4,"vel":0,"dist":0,"stat":3}}
{"sr":{"stat":3}}
> X-1.653 Y-170.831 Z-4.347
{"err":{"code":225,"msg":"Soft limit - Z min","line":14366,"data":"X-1.653 Y-170.831 Z-4.347"}}
```
---

### Second error
Firmware 101.02 - commit 194c9ab

[... 350 lines cut ...]
{"sr":{"posx":236.98056,"posy":410.59744,"posz":36.76469}}
{"sr":{"posx":225.38371,"posy":409.05117,"posz":34.78549}}
{"sr":{"posx":213.61116,"posy":407.4815,"posz":32.77631}}
{"sr":{"posx":201.8386,"posy":405.91183,"posz":30.76713}}
{"sr":{"posx":190.06605,"posy":404.34216,"posz":28.75794}}
{"sr":{"posx":178.2935,"posy":402.77249,"posz":26.74876}}
{"sr":{"posx":166.52095,"posy":401.20281,"posz":24.73958}}
{"sr":{"posx":154.9241,"posy":399.65655,"posz":22.76039}}
{"sr":{"posx":143.15155,"posy":398.08688,"posz":20.7512}}
{"sr":{"posx":131.37899,"posy":396.51721,"posz":18.74202}}
{"sr":{"posx":119.60644,"posy":394.94754,"posz":16.73284}}
{"sr":{"posx":107.83389,"posy":393.37786,"posz":14.72365}}
{"sr":{"posx":96.06134,"posy":391.80819,"posz":12.71447}}
{"sr":{"posx":84.46449,"posy":390.26196,"posz":10.73528}}
{"sr":{"posx":72.69194,"posy":388.69229,"posz":8.72609}}
{"sr":{"posx":60.91938,"posy":387.12261,"posz":6.71691}}
{"sr":{"posx":49.73348,"posy":385.6311,"posz":4.80785,"vel":5672.33}}
{"sr":{"posx":45.0953,"posy":385.01269,"posz":4.01627,"vel":491.06}}
{"qr":48}
{"sr":{"posx":45,"posy":385,"posz":4,"vel":0,"stat":3}}
{"sr":{"stat":3}}
> G0 X-2.43 Y-173.863
{"err":{"code":221,"msg":"Soft limit - X min","line":13,"data":"G0 X-2.43 Y-173.863"}}
>
```

amx1, 2018-05-28:
@justinclift 

I am just checking the test cases. This one runs also without errors with my config and my custom shield.

Firmware: 101.02-17-g51f9-dirty, CNCJS 1.9.15 latest master, Soft and Hard limits enabled! Only needed digital inputs enabled to avoid random interrupts

justinclift, 2018-05-28:
That sounds interesting. :smile:

When you say "Only needed digital inputs enabled to avoid random interrupts", does that mean I should change something in [my config](https://github.com/justinclift/g2/blob/b9158edc2fd9f918f83a1f67a7d42f0d88763ba8/g2core/settings/settings_shapeoko3.h)?

amx1, 2018-05-28:
check what happens if you set all not used inputs to disabled
like #define DI1_MODE  IO_MODE_DISABLED

if your custom shield does not set the inputs to a defined value by pull-up resistors this might fire random interrupts

justinclift, 2018-05-28:
Oh interesting thought.  Thanks @amx1 I'll give that a shot. :smile:

aldenhart, 2018-06-01:
Justin, did that solve the issue?

justinclift, 2018-06-01:
Haven't checked yet.  Been head down migrating servers for a different project + in-depth reviewing some of the PR's there too.

Likely be finished with those bits over the weekend, which should free up some schedule. :smile:

aldenhart, 2018-06-02:
Cool. I'm interested in your experience.

justinclift, 2018-06-04:
@aldenhart Just tried a run of this with latest edge (101.03) with the motors disabled, with the [same settings](https://github.com/justinclift/g2/commit/ca4e95de8d589dfd22ca7b376d2d5685df73559e) applied.

It ran through to completion without error, so I'm kind of hopeful your fix the other day (5c4ba36db4a2746a27f5c0b03cb5220497a1a9a5) solved the problem I was hitting.

I'll try it again tommorrow morning with the motors enabled and actually cutting something, and report back. :smile:

justinclift, 2018-06-05:
Yep, this issue seems to be fixed in 101.03. :smile:

With 101.02, this particular gcode file had a 100% failure rate.  Now, after a few soft runs (eg motors off) and one hard run (cutting stuff) it's not yet failed.

[image]

Thanks @aldenhart and @amx1. :smile:

closed 2018-06-05 by justinclift