synthetos/g2 #350: Build Error with Xcode 9.3.1

BDFife, 2018-05-19:
This issue looks like it might be similar to #70. I'm seeing the following error when I try to compile per the instructions on the Wiki (first, I ran into a build error that was fixed by deleting the ./Motate/Tools/osx folder. 

Now I'm seeing:

 "../Motate/MotateProject/motate/../../Tools/osx/gcc-arm-none-eabi/bin"/arm-none-eabi-g++  -Wall -Wchar-subscripts -Wcomment -Wformat=2 -Wmain -Wparentheses -Wunreachable-code -Wsequence-point -Wreturn-type -Wswitch -Wtrigraphs -Wunused -Wuninitialized -Wno-unknown-pragmas -Wfloat-equal -Wundef -Wshadow -Wpointer-arith -Wwrite-strings -Wformat -Wmissing-format-attribute -Wno-deprecated-declarations -Wredundant-decls -Wlong-long -Wno-format-nonliteral -Wno-format-security -mcpu=cortex-m7 -mthumb -ffunction-sections -fdata-sections -std=gnu++14 -fno-rtti -fno-exceptions -fstrict-volatile-bitfields -fdiagnostics-show-option  -DBOARD=gquadratic-b   -g3 -Os -I../Motate/MotateProject/motate/cmsis -I../Motate/MotateProject/motate/cmsis/TARGET_Atmel/sams70/include -I../Motate/MotateProject/motate/cmsis/TARGET_Atmel -I../Motate/MotateProject/motate/cmsis/TARGET_Atmel/preprocessor -I../Motate/MotateProject/motate/cmsis/TARGET_Atmel/sams70/include -I../Motate/MotateProject/motate/cmsis/TARGET_Atmel/sams70/source/templates -I../Motate/MotateProject/motate/Atmel_sam_common -I../Motate/MotateProject/motate/Atmel_sams70 -I../Motate/MotateProject/motate/platform/atmel_sam -I. -I../Motate/MotateProject/motate -I./board/gquadratic -Idevice/step_dir_driver -Idevice/step_dir_hobbyservo -Idevice/neopixel -DMOTATE_BOARD="gquadratic-b" -DSETTINGS_FILE="settings_test.h" -DMOTATE_CONFIG_HAS_USBSERIAL=0 -D__SAMS70N19__ -D__sams70__ -D__ARM__ -DGIT_VERSION=\"100.26-10-g1ff7-dirty\" -DGIT_EXACT_VERSION=100.26 -DDEBUG=1 -DIN_DEBUGGER=1 -MMD -MF build/TestQuadratic-gquadratic-b/dep/MotateUtilities.o.d -MT build/TestQuadratic-gquadratic-b/motate/MotateUtilities.o -xc++ -c -o build/TestQuadratic-gquadratic-b/motate/MotateUtilities.o ../Motate/MotateProject/motate/MotateUtilities.cpp
arm-none-eabi-g++: internal compiler error: Killed: 9 (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
make: *** [build/TestQuadratic-gquadratic-b/motate/MotateUtilities.o] Error 4
Command /usr/bin/make failed with exit code 2

I jumped into the terminal and trying to build MotateUtilities.cpp with the same parameters has the same result. I also tried the steps in #70 and verified there isn't another arm-none-eabi-g++. Any advice on what I should try to get this unstuck?

Thanks,
Brian Fife

justinclift, 2018-05-19:
Interesting.  I'll have a go on an OSX desktop later on today, and see if I hit the same problem.

In the meantime, which branch are you using - `edge` (the default) - or something else?

Also, what's the full path on your box to your checked out clone of the git repo?  Just in case there are spaces or something in the path name that are mucking things up. :smile:

BDFife, 2018-05-19:
Thanks!

Path is /Users/bdfife/Projects/g2
I did a ‘standard’ git checkout a few months ago, and pulled latest as part
of my debugging process when the build didn’t work for me. The main repo
thinks it is up to date.

justinclift, 2018-05-19:
No worries.  What's your output of `git branch -a`, when run in that repo?

justinclift, 2018-05-19:
Oh, also what are the parameters you're using with the `make` command?

BDFife, 2018-05-19:
Git:

Athens:g2 bdfife$ git branch -a

* master

  remotes/origin/HEAD -> origin/master

  remotes/origin/dev-168-gquintic

  remotes/origin/dev-169-motor-enable-polarity

  remotes/origin/dev-212-common-enable

  remotes/origin/dev-298-nested-json-read-fail

  remotes/origin/dev-304-uvw-9axis

  remotes/origin/dev-307-spring-comp

  remotes/origin/dev-omco-issue-2

  remotes/origin/digital-dro

  remotes/origin/edge

  remotes/origin/edge-087

  remotes/origin/edge-090

  remotes/origin/edge-100

  remotes/origin/gh-pages

  remotes/origin/master

  remotes/origin/omc-072.65

Athens:g2 bdfife$

Make command uses whatever is default for Xcode. In the case of the
specific file being built, I think the parameters are:

"Found that we're in OS X"

Compiling cpp ../Motate/MotateProject/motate/MotateUtilities.cpp

    -> build/TestQuadratic-gquadratic-b/motate/MotateUtilities.o

"../Motate/MotateProject/motate/../../Tools/osx/gcc-arm-none-eabi/bin"/arm-none-eabi-g++
-Wall -Wchar-subscripts -Wcomment -Wformat=2 -Wmain -Wparentheses
-Wunreachable-code -Wsequence-point -Wreturn-type -Wswitch -Wtrigraphs
-Wunused -Wuninitialized -Wno-unknown-pragmas -Wfloat-equal -Wundef
-Wshadow -Wpointer-arith -Wwrite-strings -Wformat
-Wmissing-format-attribute -Wno-deprecated-declarations -Wredundant-decls
-Wlong-long -Wno-format-nonliteral -Wno-format-security -mcpu=cortex-m7
-mthumb -ffunction-sections -fdata-sections -std=gnu++14 -fno-rtti
-fno-exceptions -fstrict-volatile-bitfields -fdiagnostics-show-option
-DBOARD=gquadratic-b   -g3 -Os -I../Motate/MotateProject/motate/cmsis
-I../Motate/MotateProject/motate/cmsis/TARGET_Atmel/sams70/include
-I../Motate/MotateProject/motate/cmsis/TARGET_Atmel
-I../Motate/MotateProject/motate/cmsis/TARGET_Atmel/preprocessor
-I../Motate/MotateProject/motate/cmsis/TARGET_Atmel/sams70/include
-I../Motate/MotateProject/motate/cmsis/TARGET_Atmel/sams70/source/templates
-I../Motate/MotateProject/motate/Atmel_sam_common
-I../Motate/MotateProject/motate/Atmel_sams70
-I../Motate/MotateProject/motate/platform/atmel_sam -I.
-I../Motate/MotateProject/motate -I./board/gquadratic
-Idevice/step_dir_driver -Idevice/step_dir_hobbyservo -Idevice/neopixel
-DMOTATE_BOARD="gquadratic-b" -DSETTINGS_FILE="settings_test.h"
-DMOTATE_CONFIG_HAS_USBSERIAL=0 -D__SAMS70N19__ -D__sams70__ -D__ARM__
-DGIT_VERSION=\"100.26-10-g1ff7-dirty\" -DGIT_EXACT_VERSION=100.26
-DDEBUG=1 -DIN_DEBUGGER=1 -MMD -MF
build/TestQuadratic-gquadratic-b/dep/MotateUtilities.o.d -MT
build/TestQuadratic-gquadratic-b/motate/MotateUtilities.o -xc++ -c -o
build/TestQuadratic-gquadratic-b/motate/MotateUtilities.o
../Motate/MotateProject/motate/MotateUtilities.cpp

arm-none-eabi-g++: internal compiler error: Killed: 9 (program cc1plus)

Please submit a full bug report,

with preprocessed source if appropriate.

See <http://gcc.gnu.org/bugs.html> for instructions.

make: *** [build/TestQuadratic-gquadratic-b/motate/MotateUtilities.o] Error
4

Command /usr/bin/make failed with exit code 2

==

-Brian

justinclift, 2018-05-19:
Thanks @BDFife, that helps. :smile:

justinclift, 2018-05-19:
> I'll have a go on an OSX desktop later on today ...

This is going to have to be a tomorrow thing instead.  It's getting late here now and I'm just too sleepy atm. :frowning:

justinclift, 2018-05-20:
Trying out the build on Xcode here now... also fails:

[image]

It's not immediately clear to me why there are references to "gQuintic PrintrBot" and "gQuadratic" in the error messages.  I'm definitely in the master branch, and from what I can tell at the moment (I'm not familiar with Xcode personally) it should be building for Arduino Due + gShield.

Investigating now. :smile:

BDFife, 2018-05-20:
> Hmmm, what seems to be happening is that clicking the play button builds
> for *every* target, not the single one that's clearly selected. So our
> Xcode docs look like they need updating. 😦
>
> For reference I'm using OSX 10.11.6 (El Capitan), Xcode 8.2.1. Yeah, it's
> a bit older than yours.
>
> As a workaround for now, the command line instructions (for OSX and Linux)
> seem ok:
>
>    - https://github.com/synthetos/g2/wiki/Compiling-g2core-on-
>    Linux-and-OS-X-(command-line)#build-the-sources
>
> $ make PLATFORM=DUE BOARD=gShield
> "Found that we're in OS X"
> Compiling c ../Motate/MotateProject/motate/cmsis/TARGET_Atmel/sam3x/source/templates/exceptions.c
>     -> build/gShield/motate/cmsis/TARGET_Atmel/sam3x/source/templates/exceptions.o
> Compiling c ../Motate/MotateProject/motate/cmsis/TARGET_Atmel/sam3x/source/templates/system_sam3x.c
>     -> build/gShield/motate/cmsis/TARGET_Atmel/sam3x/source/templates/system_sam3x.o
> Compiling c ../Motate/MotateProject/motate/cmsis/TARGET_Atmel/sam3x/source/templates/gcc/startup_sam3x.c
>     -> build/gShield/motate/cmsis/TARGET_Atmel/sam3x/source/templates/gcc/startup_sam3x.o
> Compiling c ../Motate/MotateProject/motate/platform/atmel_sam/cortex_handlers.c
>     -> build/gShield/motate/platform/atmel_sam/cortex_handlers.o
> Compiling c ../Motate/MotateProject/motate/platform/atmel_sam/hooks.c
>     -> build/gShield/motate/platform/atmel_sam/hooks.o
> [... 97 lines cut ...]
>    text	   data	    bss	    dec	    hex	filename
>  168568	      0	  35152	 203720	  31bc8	bin/gShield/g2core.elf
> Build 100.26 100.26-10-g1ff7-dirty
>
> Are you ok to go with the command line approach?

Yes, no problem with the command line. The only reason why I tried Xcode in
the first place was that I assumed from the flow of the document it was the
preferred approach :)

I’ll give this a go now.

BDFife, 2018-05-20:
Still fails trying to build for V9Test from command line. I'll try a clean checkout. 

What OS/Branch has the best chance of success? I'm happy to set up a linux VM and build via that, also, if that's the most well-trodden path.

BDFife, 2018-05-20:
Command line OSX build **from a clean checkout** worked perfectly for me on the first run. Thanks!

closed 2018-05-20 by BDFife

justinclift, 2018-05-20:
You're welcome. :smile:

I've just adjusted the wiki page to de-emphasise the Xcode instructions, so most people on OSX will use the CLI ones instead. :smile:

With the branches, `master` is generally considered pretty stable.  I'm not personally sure if it's suitable for 3D printing with though.  One way to find out. :wink:

The `edge` branch is where development stuff gets merged.  eg people do development either on `edge` itself (for simple things and many PRs), or in longer running special branches (eg `dev-168-gquintic`).

When a longer-running-special-branch is considered "ready", that gets merged into `edge`.

Every now and then, when `edge` looks to be pretty stable and isn't showing bugs, it'll get promoted to `master`.

So... if the `master` branch does what you're needing, stick with that.  If it doesn't, then join us in the more err... leading `edge` branch. 😬