synthetos/g2 #353: Xcode Wiki Instructions out of Date

BDFife, 2018-05-20:
See PR #350 

Xcode build instructions didn't work for me when I followed the instructions on the wiki. https://github.com/synthetos/g2/wiki/Compiling-g2core-on-OS-X-(with-Xcode)

Per @justinclift, the Xcode project may be set up incorrectly. The way the docs are set up right now a naive user (that's me) might assume that Xcode is the preferred build method for OSX.

Consider deprecating the Xcode instructions until someone gets them up to date.

justinclift, 2018-05-20:
Heh Heh Heh

Already done.  Well, I think it is.  Would you be ok to look over my changes and confirm they're good from your point of view? :smile:

BDFife, 2018-05-20:
Perfect, thank you.

closed 2018-05-20 by BDFife