synthetos/g2 #354: changing of status report not working

amx1, 2018-05-21:
I have some problems with the actual edge version. I was not able to enable posx, posy, ... print out by {"sr":{"posx":true, "posy":true,...}}. it does not change the pre compiled default values. The edge version is not working properly with CNCJS, because of the missing coordinates. CNCJS sends all the config stuff during the initialization phase, but this fails with the actual edge version. 
  
The work around was to change my  settings file then compile and flash a new bin file. 

Firmware Version: {"r":{"fv":0.99,"fb":101.03,"fbs":"101.02-21-g9de1-dirty","fbc":"settings_cnc3040.h","hp":"ArduinoDue","hv":"na","id":"0084-8ce7-0084-6bd","msg":"SYSTE
M READY"},"f":[1,0,1]}

I would like to ask somebody to check, if it is a general problem or just a question of my configuration.

justinclift, 2018-05-21:
Thanks @amx1.  If no-one gets to it sooner, I'll probably have time to investigate this either today or tomorrow.

Just for reference:

* Which CNCjs version are you using, and how did you install it?
  * eg npm? git clone? download of desktop app?
* Which OS is this on?
  * Guessing Linux (from quick searching and finding [this](https://github.com/cncjs/cncjs/issues/210))?  If so, which distro.

That's just in case it's useful for trying to match your install for debugging. :smile:

amx1, 2018-05-21:
@justinclift, it is not an issue of CNCjs. I am using the latest master by git clone and npm install on a pi3 with 64bit linux (https://github.com/bamarni/pi64) for production and development on a standard 64bit ubuntu 16.04.

 It is definitely a problem of g2. I checked this by communicating to the g2 controller directly with a terminal program. (minicom) You will not see the problem normally with CNCjs, because of the default definition in settings_default.h

#ifndef STATUS_REPORT_DEFAULTS                              // {sr: See Status Reports wiki page
#define STATUS_REPORT_DEFAULTS "line","posx","posy","posz","posa","feed","vel","unit","coor","dist","admo","frmo","momo","stat"
// Alternate SRs that report in drawable units
//#define STATUS_REPORT_DEFAULTS "line","vel","mpox","mpoy","mpoz","mpoa","coor","ofsa","ofsx","ofsy","ofsz","dist","unit","stat","homz","homy","homx","momo"
#endif 

I changed the STATUS_REPORT_DEFAULTS in my settings a while ago, because I needed some other information for a different program. This was not an issue, because you could change the report by sending {sr:{posx:t, ...}} which is done during the initialization phase of CNCjs, but it is not working anymore. 

for example: sending {sr:{posx:t}} or {"sr":{"posx":true}} leads to:
{"r":{"sr":{"posx":false}},"f":[1,0,15]} 

It should report posx only but it does does not change anything,

g2core version 101.02-21-g9de1-dirty

justinclift, 2018-05-21:
No worries.  Wasn't thinking it was a CNCjs problem, I just (personally) like to grab the versions of things in case I need to run the same stuff up in a VM to test with.  I've just found that it helps to ask up front first is all. :wink:

justinclift, 2018-05-27:
(Finally got time to test this)

Yep, this does look like a bug in the `edge` branch.

* When changing the status report settings in `master` through manually setting `sr` values, they take effect and are used.
* No so in the `edge` branch, where manually giving `sr` values does nothing. :frowning_face: 

Tested just now through Minicom on CentOS 7 x64,  with fresh compiles of both master and edge, using `make BOARD=gShield`

The example status report lines I tried setting were:

* `{sr:{line:t,posx:t,posy:t,posz:t,vel:t,unit:t,stat:t}}`
* `{sr:{line:t,posx:t}}`
* `{sr:{posx:t,posy:t,posz:t}}`

All three worked in `master` but not `edge`.

### Edge branch

```
Welcome to minicom 2.6.2

OPTIONS: I18n 
Compiled on Jun 10 2014, 03:20:53.
Port /dev/ttyACM0, 20:57:15

Press CTRL-A Z for help on special keys

{"r":{"fv":0.99,"fb":101.03,"fbs":"101.02-17-g51f9","fbc":"settings_default.h","hp":"ArduinoDue","hv":"na","id":"0084-d639-18c6-29c","msg":"SYSTEM READY"},}

?
{"r":{"sr":{"line":0,"posx":0,"posy":0,"posz":0,"posa":0,"feed":0,"vel":0,"unit":1,"coor":1,"dist":0,"admo":1,"frmo":1,"momo":4,"stat":1}},"f":[1,0,2]}
{sr:{line:t,posx:t,posy:t,posz:t,vel:t,unit:t,stat:t}}
{"r":{"sr":{"line":1,"posx":false,"posy":false,"posz":false,"vel":false,"unit":1,"stat":1}},"f":[1,0,56]}
?
{"r":{"sr":{"line":0,"posx":0,"posy":0,"posz":0,"posa":0,"feed":0,"vel":0,"unit":1,"coor":1,"dist":0,"admo":1,"frmo":1,"momo":4,"stat":1}},"f":[1,0,2]}
{sr:{line:t,posx:t}}
{"r":{"sr":{"line":0,"posx":0,"posy":0,"posz":0,"posa":0,"feed":0,"vel":0,"unit":1,"coor":1,"dist":0,"admo":1,"frmo":1,"momo":4,"stat":1}},"f":[1,0,23]}
?
{"r":{"sr":{"line":0,"posx":0,"posy":0,"posz":0,"posa":0,"feed":0,"vel":0,"unit":1,"coor":1,"dist":0,"admo":1,"frmo":1,"momo":4,"stat":1}},"f":[1,0,2]}
{sr:n}
{"r":{"sr":{"line":0,"posx":0,"posy":0,"posz":0,"posa":0,"feed":0,"vel":0,"unit":1,"coor":1,"dist":0,"admo":1,"frmo":1,"momo":4,"stat":1}},"f":[1,0,8]}
{sr:{posx:t,posy:t,posz:t}}
{"r":{"sr":{"posx":false,"posy":false,"posz":false}},"f":[1,0,29]}
{sr:n}
{"r":{"sr":{"line":0,"posx":0,"posy":0,"posz":0,"posa":0,"feed":0,"vel":0,"unit":1,"coor":1,"dist":0,"admo":1,"frmo":1,"momo":4,"stat":1}},"f":[1,0,8]}
```

Clearly, giving `sr` lines resulted in no change to the status report format generated.  No idea why, as no error was indicated.

### Master branch

```
Welcome to minicom 2.6.2

OPTIONS: I18n 
Compiled on Jun 10 2014, 03:20:53.
Port /dev/ttyACM0, 21:08:18

Press CTRL-A Z for help on special keys

{"r":{"fv":0.99,"fb":100.26,"fbs":"100.26-10-g1ff7","fbc":"settings_default.h","hp":3,"hv":0,"id":"0084-d639-18c6-29c","msg":"SYSTEM READY"},"f":[1,0,1]}

?
{"r":{"sr":{"line":0,"posx":0,"posy":0,"posz":0,"posa":0,"feed":0,"vel":0,"unit":1,"coor":1,"dist":0,"admo":1,"frmo":1,"momo":4,"stat":1}},"f":[1,0,2]}
{sr:{line:t,posx:t,posy:t,posz:t,vel:t,unit:t,stat:t}}
{"r":{"sr":{"line":0,"posx":0,"posy":0,"posz":0,"vel":0,"unit":1,"stat":1}},"f":[1,0,56]}
{"sr":{"line":0,"posx":0,"posy":0,"posz":0,"vel":0,"unit":1,"stat":1}}
?
{"r":{"sr":{"line":0,"posx":0,"posy":0,"posz":0,"vel":0,"unit":1,"stat":1}},"f":[1,0,2]}
{sr:{line:t,posx:t}}
{"r":{"sr":{"line":0,"posx":0}},"f":[1,0,22]}
?
{"r":{"sr":{"line":0,"posx":0}},"f":[1,0,2]}
{sr:n}
{"r":{"sr":{"line":0,"posx":0}},"f":[1,0,8]}
{sr:{posx:t,posy:t,posz:t}}
{"r":{"sr":{"posx":0,"posy":0,"posz":0}},"f":[1,0,29]}
?
{"r":{"sr":{"posx":0,"posy":0,"posz":0}},"f":[1,0,2]}
```

Here, the given `sr` values do take effect.

amx1, 2018-05-28:
@giseburt , @aldenhart 

I think I found the issue. It seems to be a problem of the json parser in "_json_parser_execute"

static stat_t _json_parser_execute(nvObj_t *nv) {
    do {
        if (nv->valuetype == TYPE_PARENT) {
            // anything?
        } else if (nv->valuetype == TYPE_NULL) {           // means GET the value

if you roll it back this part to 100.26 it is working again, but I don't if this leads to other effects.

static stat_t _json_parser_execute(nvObj_t *nv) {
    do {
        if (nv->valuetype == TYPE_NULL) {           // means GET the value
            ritorno(nv_get(nv));                    // ritorno returns w/status on any errors

Best Axel

aldenhart, 2018-05-29:
Thanks for the detailed diagnosis. I'll see if I can reproduce this, and add these cases to the regression tests.

aldenhart, 2018-05-29:
I believe you are correct. I have an issue branch that appears to fix it but I want to test more fully before pulling it. Your diagnosis led me right to the problem.

aldenhart, 2018-05-30:
Closing this issue. See PR #356 and release 101.03

closed 2018-05-30 by aldenhart