synthetos/g2 #357: compiling - some parameters of settings_xxx.h file not taken into account

amx1, 2018-05-31:
@aldenhart 

there is still a problem with the configuration. The following parameters of the settings_xxx.h file are not taken into account properly. I tried a lot of different combinations. Setting the coolant polarity parameters to 1 also forced the LIMIT variables to change the value. COOLANT_PAUSE_ON_HOLD is ignored at all.   

#define SOFT_LIMIT_ENABLE			0				// 0=off, 1=on
#define HARD_LIMIT_ENABLE			0				// 0=off, 1=on
#define SAFETY_INTERLOCK_ENABLE	0				// 0=off, 1=on

#define COOLANT_MIST_POLARITY 		0     				// 0=active low, 1=active high
#define COOLANT_FLOOD_POLARITY	0				// 0=active low, 1=active high
#define COOLANT_PAUSE_ON_HOLD 	0

amx1, 2018-05-31:
sorry missed to say - clean checkout of today / the problem with status report is fixed
Build 101.02-20-gf038a-dirty

justinclift, 2018-05-31:
Ahhh.  I wonder if that means the commits to edge yesterday will change things for the weird limit switch trip I reported.  I'll try to get time to test it again in the next few days.

amx1, 2018-05-31:
I checked also the old version g2-edge 100.02 80f3fcc. The problem can be reproduced by setting COOLANT_FLOOD_POLARITY.

Examples 

#define SOFT_LIMIT_ENABLE					0	// 0=off, 1=on
#define HARD_LIMIT_ENABLE					0	// 0=off, 1=on
#define SAFETY_INTERLOCK_ENABLE			0	// 0=off, 1=on

#define COOLANT_MIST_POLARITY 				0     	// 0=active low, 1=active high
#define COOLANT_FLOOD_POLARITY			1	// 0=active low, 1=active high
#define COOLANT_PAUSE_ON_HOLD 			0

result $sys in text mode

[sl]  soft limit enable           1 [0=disable,1=enable]                                                                                                    
[lim] limit switch enable         1 [0=disable,1=enable]                                                                                                    
[saf] safety interlock enable     1 [0=disable,1=enable]  

$co

[coph] coolant pause on hold      0 [0=no,1=pause_on_hold]                                                                                                  
[comp] coolant mist polarity      0 [0=low is ON,1=high is ON]                                                                                              
[cofp] coolant flood polarity     1 [0=low is ON,1=high is ON]                                                                                              
[com]  coolant mist               0 [0=OFF,1=ON]                                                                                                            
[cof]  coolant flood              0 [0=OFF,1=ON]      

changing to

#define SOFT_LIMIT_ENABLE					0	// 0=off, 1=on
#define HARD_LIMIT_ENABLE					0	// 0=off, 1=on
#define SAFETY_INTERLOCK_ENABLE			0	// 0=off, 1=on

#define COOLANT_MIST_POLARITY 				1     	// 0=active low, 1=active high
#define COOLANT_FLOOD_POLARITY			0	// 0=active low, 1=active high
#define COOLANT_PAUSE_ON_HOLD 			0

$sys 

[sl]  soft limit enable           0 [0=disable,1=enable]                                                                                                    
[lim] limit switch enable         0 [0=disable,1=enable]                                                                                                    
[saf] safety interlock enable     0 [0=disable,1=enable]    

$co
[coph] coolant pause on hold      0 [0=no,1=pause_on_hold]                                                                                                  
[comp] coolant mist polarity      1 [0=low is ON,1=high is ON]                                                                                              
[cofp] coolant flood polarity     0 [0=low is ON,1=high is ON]                                                                                              
[com]  coolant mist               0 [0=OFF,1=ON]                                                                                                            
[cof]  coolant flood              0 [0=OFF,1=ON]        

changing to

#define SOFT_LIMIT_ENABLE					1	// 0=off, 1=on
#define HARD_LIMIT_ENABLE					0	// 0=off, 1=on
#define SAFETY_INTERLOCK_ENABLE			1	// 0=off, 1=on

#define COOLANT_MIST_POLARITY 				1     	// 0=active low, 1=active high
#define COOLANT_FLOOD_POLARITY			0	// 0=active low, 1=active high
#define COOLANT_PAUSE_ON_HOLD 			0

result

[sl]  soft limit enable           0 [0=disable,1=enable]                                                                                                    
[lim] limit switch enable         0 [0=disable,1=enable]                                                                                                    
[saf] safety interlock enable     0 [0=disable,1=enable]

[coph] coolant pause on hold      0 [0=no,1=pause_on_hold]                                                                                                  
[comp] coolant mist polarity      1 [0=low is ON,1=high is ON]                                                                                              
[cofp] coolant flood polarity     0 [0=low is ON,1=high is ON]                                                                                              
[com]  coolant mist               0 [0=OFF,1=ON]                                                                                                            
[cof]  coolant flood              0 [0=OFF,1=ON]

aldenhart, 2018-06-01:
Created a PR for the fix. It's in edge now.

closed 2018-06-01 by aldenhart