synthetos/g2 #359: G2core - how to home BOTH motors on my Y-Axis?

grantnlee, 2018-06-03:
I am upgrading my GRBL based DIY CNC to leverage G2Core.  One of the goals in the upgrade it to be able to have home switches on both of the motors/ballscrews that drive the Y-axis in order to help fine tune the calibration of the X-Y squareness.

I can't seem to figure out whether g2core will do this, when two motors operate the same axis.

When I setup the two Y-axis motors, I mapped them both to the Y-axis.   But the homing setup driven off the definition of the axis, not the motor/ballscrew.  So I am being forced down the path of a single limit switch that will define the home for both of the Y motor/ballscrews at the same time...

Is it possible and how should I set up my Y-axis, when I have two motors and want to home each of them independently, yet operate the CNC with those two in lockstep?

 Thanks! Jay

justinclift, 2018-06-03:
Oh, that's an interesting question.

There's definitely capability for having multiple limit switches in a fairly user defined way.

But yeah.  Any axis seems like it can only be tied to a single one (eg Y axis for a [Shapeoko 2](https://github.com/synthetos/g2/blob/f5a6753f8e9bd863460badcecef8cca51406f4da/g2core/settings/settings_shapeoko2dualY.h#L146)), rather than passing it multiple values.

That'll probably be a useful feature to add. :smile:

justinclift, 2018-06-03:
Yeah, just tried passing a comma separated list of digital input # values (`$xhi=1,2`) to my X axis homing config here.  Didn't work. Oh well. :frowning:

giseburt, 2018-06-03:
So we are looking at such interesting homing situations in an upcoming
kinematics update.

For now our solution is a two-part solution:

1) home the axis normally, where both motors are enabled and one of the
inputs is set as the homing input.
2) disable one motor, and switch the homing input to the one on the motor
that’s still enabled.

So this is a case for M100, so you can script this. As an example if your Y
axis is set to motors 2 and 3, and the homing inputs are set to match the
motors, where motor 2 homes to input 2 and motor 3 Gaines to input 3, then
this would be the gcode sequence (adjust the motor numbers, input numbers,
and power modes as needed):

G28.2 Z0 X0 ; home Z and X
M100 ({xhi:2, 2pm:2, 3pm:2}) ; switch y homing to input 2, and power mode
to 2 for both motors
G28.2 Y0 ; now home y to input 2, moving both motors
M100 ({xhi:3, 2pm:0}) ; switch y to home to input 3, disable motor 2
G28.2 Y0 ; now home just motor 3 to input 3
M100 ({xhi:2, 2pm:2}) ; enable motor 2 again, power mode 2
; now the machine is fully homed and ready to use

Now to square up your Y axis you aware up the limit switches on inputs 2
and 3.

Note that if this sequence gets interrupted you MUST ensure you enable
motor 3 again before moving the Y axis or it could damage your gantry.

Homing the Y the second time with only motor 3 is only safe after homing
with both Y motors and assuming that they are at least close enough to
square that the machine can move safely.

grantnlee, 2018-06-03:
Thanks, makes sense.  I'll cut over to g2core with the standard homing and then experiment with that approach once I am steady state.

Given it's not in the code base yet, it does make me think there might be a different approach to implementing the feature.  Maybe a "X-Y squaring" or "two motor alignment" routine operates separately from the homing routine.  (Let home continue to function like it does today.)  

Whereas my homing search velocity is typically on the order of 1000 mm/min, it does feel like I might want to apply a higher degree of caution when running a function that stops one of the two motors and keeps the other going.  And maybe limit the incremental travel of the second motor/ballscrew to no more than a few more mm more than the first motor in order to find the second switch.  Instead of complicating the homing function, maybe this would be best implemented independently.  

Appreciate the insight.  I'll share the results if I get to experiment with this and learn something more.
Jay

giseburt, 2018-06-03:
Thank you Jay. Please do share you’re results and we’ll use it to inform
how we officially support that.

An alternative to disabling the motor is to switch it to an axis you’re not
using, that way it holds position. It would be done at the same times using
M100.

RonHunn, 2018-07-04:
Hi, If you can stand having one motor skip steps for a bit you could wire both homing switches in series. The homing sequence won't complete until both motors have made the switches.
A bit dirty but would work.

closed 2019-12-12 by ril3y