synthetos/g2 #362: Adding IO pins

brettbalder, 2018-06-11:
I am using a Due and building my own shield to interface with my machine. I am capable of changing the number of input pins from 9 to 12, but cannot seem to add any more than that. I'd like 16 inputs in total: 2 for each axis limit (XYZABC), a probe, and three buttons to control resetting, feedhold, and cycle start. 

So far I have:
1) changed the #define D_IN_CHANNELS from 9 to 16 in gpio.h
2) added ioDigitalInputExt<kInput#_PinNumber , #> _din#; for pin# 1-16 at line 220 in gpio.cpp
3) added _din#.reset(); at line 400 in gpio.cpp
4) added pin_number kInput#_PinNumber = SomeDuePinNumber; at line 200 in my shield-pinout.h

Am I missing something?

giseburt, 2018-06-11:
Yes, the big table in config_app.cpp and the totals after it need updated
to extend the JSON interface.

Btw, simplifying the setup of the config system is on our radar.

justinclift, 2018-06-11:
@brettbalder As a thought, with your rotational axes (ABC) do they definitely need 2 limit switches?

Just asking because for the A axis stuff I've been looking at recently, only one is needed.  The A axis (in my case) is able to rotate 360° continuously, so there aren't limits it's dangerous to hit.  eg I just need something that can provide a reliable home point (not hard).

:smile:

brettbalder, 2018-06-11:
Got it working, thank you gentleman.

@justinclift No, I'm not even sure if I'll need to add rotational axes in the future. I'm working on the early stages of a machine and want to allow for potential expansion for when other people continue my work when I leave. It is likely that the inputs will be used by something, though.

closed 2018-06-11 by brettbalder

justinclift, 2018-06-11:
No worries at all. :smile:

justinclift, 2018-06-11:
@brettbalder Your Craniobot repo looks interesting:

&nbsp; &nbsp; https://github.com/brettbalder/CraniobotCourier

It sounds like it'll be able to probe a given area, generate a non-planar surface from the data, then transform desired operations to match the surface.  Is that a correct understanding of the README info?

Also, are we ok to add it to the "Who uses g2core?" page? :smile:

&nbsp; &nbsp; https://github.com/synthetos/g2/wiki/g2core-in-use

brettbalder, 2018-06-11:
Yup, I am actually updating that README as we speak. I'm not sure how much I can disclose, but the gist of it is that a CNC is being used to automate microsurgical procedures in research animals. The end game is providing neurological researchers with a much easier way of conducting their research. But yes, you can include it :)

justinclift, 2018-06-11:
Thanks.  Just added it. :smile: