synthetos/g2 #363: typedef problem in config_app.cpp

amx1, 2018-06-17:
@aldenhart 

I think that there is another typedef problem in config_app. The heater enable commands are not working properly. 

he1e is defined as boolean, he2e and he3e are defined as int and the default_enable are defined as boolean. 

--------
config_app.cpp  / line 565 ++

    { "he1","he1e", _bip, 0, tx_print_nul, cm_get_heater_enable,   cm_set_heater_enable,   nullptr, H1_DEFAULT_ENABLE },

    { "he2","he2e", _iip, 0, tx_print_nul, cm_get_heater_enable,   cm_set_heater_enable,   nullptr, H2_DEFAULT_ENABLE },

------------------
temperature.cpp / line 639 ++

stat_t cm_set_heater_enable(nvObj_t *nv)
{
    switch(_get_heater_number(nv)) {
        case '1': { pid1._enable = nv->value_int; break; }
        case '2': { pid2._enable = nv->value_int; break; }
        case '3': { pid3._enable = nv->value_int; break; }
        default: { return(STAT_INPUT_VALUE_RANGE_ERROR); break; }
    }
    return (STAT_OK);
}

  
Best amx1

aldenhart, 2018-06-17:
Thanks for the heads up. I'll take a look. I'll probably combine this with the step diagnostics Issue #360. If see any other problems please let me know.

justinclift, 2018-06-17:
@amx1 Out of curiosity, any interest in being added to the g2 team here on GitHub?

It gives write access to things (eg the issues section, git source), though for non-trivial source code changes it's still best to do stuff via PR for commenting/review. :smile:

amx1, 2018-06-19:
@justinclift 

thank you for the offer, but I am quite busy at the moment and don't have the time to work on it continuously.

Best

amx1

justinclift, 2018-06-19:
No worries @amx1. :smile: