synthetos/g2 #364: 2 Due rx tx not working

JasonTitcomb, 2018-06-23:
Hi all and thanks to all that have come before me.

I am trying to connect 2 Due boards via rx tx and I am stumped.
Inspired by [this ](https://github.com/tchepperz/ArdSketch/blob/master/serialGrblSkch/serialGrblSkch.ino)sketch I want to use one Due for HMI and one for G2core.

G2(100.26 bin) is installed and working on Due#2 and I have a small test sketch on Due#1
I get no response when G2 is flashed to the Due.
I have done so much testing that I believe it has something to do with G2core and Due.

Here is why I have done.
1. I have tested my sketch and wiring with a TinyG V8 hardware and it works.
2. I have tested with grbl on Uno and it works.
3. I can get 2 Due to talk to each other with my own test sketch on each.
4. I have even tried a new Due and still will not work.

Something to do with the two usb ports on Due?
Any ideas?
Regards,
Jason

justinclift, 2018-06-24:
Ahhh, interesting idea. :smile:

Just to get a better idea of the bits you've already tested, with Due #2... have you been able to successfully talk to it over usb from a PC/Pi/etc?  eg send it command, and have it return info

It sounds like "yes", but I'm asking just to be sure first. :grin: 

Next thought... which settings were used?  With g2core, the behaviour of things is changable via some of the settings that get compiled in.  eg json vs text mode, whether to use 1 or 2 usb ports, (etc)

Did you use one of the pre-built g2core .bin images for flashing (if so, which one?), or is it one you've compiled?  If it's one you've compiled, are you ok to point us at your settings file?

JasonTitcomb, 2018-06-24:
Thanks for responding.
pre-built g2core .bin 100.26 just downloaded it.
No, I can't get them to talk. That's the problem.
I bet the usb 1 or 2 has something to do with it.
pic.twitter.com/xy6jgon14j

justinclift, 2018-06-24:
Thanks, that's helpful stuff. :smile:

Btw, I just added your pic directly to the comment instead, to make things easier.

Ok, so the very first thing I'd (personally) do, is start out super simple and validate things from the most basic point (then work from there).

So, disconnect the boards from each other, and just leave the one with g2core 100.26 connected via usb.  The left side one in your pic seems like the one to choose for this.

With those connected together, in theory you *should* be able to communicate to that single board over USB using (say) CoolTerm (since you're on Windows).

Oh... since you're on Windows you need to install the drivers first, so windows recognises the "TinyG2" comms ports.  There's a bit of the wiki page has the .inf file for Windows, and the walk-though-video showing how, if you haven't done it yet. :smile:

&nbsp; &nbsp; https://github.com/synthetos/g2/wiki/Connecting-to-g2core#windows

With that driver installed, plugging in a single usb to the port you - as you already have in the photo above - should be enough for CoolTerm (or whichever program you're using) to see it.

Anyway, that's the starting point I'd aim for first.  When that's working, then I'd look into the tx/rx bits next.

Err... does that help? :smile:

giseburt, 2018-06-24:
I can’t give a thorough response at the moment but the RTS and CTS lines
are vital.

justinclift, 2018-06-24:
Oh good point.

Yeah, once the "single Due running g2core" is known to work, then to experiment with the tx/rx lines the RTS and CTS will need to be included as well. :smile:

JasonTitcomb, 2018-06-24:
Okay, I will look into RTS CTS. 

But as I said...
I have tested my sketch and wiring with a TinyG V8 hardware and **it works fine**.
I have tested my sketch with my sketch on a Mega with grbl on an Uno and **it works fine.**
The G2 Due responds fine through the native USB port.
Isn't the native USB port the same as rx0 and tx0?
I'm sure it will be something simple.

justinclift, 2018-06-24:
> The G2 Due responds fine through the native USB port.

Ahhh ok.  Sounds like you're up to experimenting with the RTS CTS stuff then.

And yeah, hopefully it is something simple with that.

closed 2022-05-27 by JasonTitcomb