synthetos/g2 #365: Not updating positions in status report

breiler, 2018-06-29:
Sorry for this noob issue... But I've been experimenting for a couple of hours and I'm not getting anywhere.

I've tried loading my Due with the g2core-gShield-100.26.bin (I've also tried the latest build 101.03) and connected to it using MacOSX and CoolTerm. I can post commands and get nice responses from the hardware. I don't have any shields connected to it, just running it stand alone.

But if I try to move the machine using `G0 X10` the position isn't updated in the status report. I've tried connecting to it using Chilipeppr and I get the same thing there.

[image]

Is there a setting that I'm missing, or is my hardware bad (bought a cheaper Due from China: https://www.banggood.com/Arduino-Compatible-DUE-R3-32-Bit-ARM-With-USB-Cable-p-906466.html?rmmds=search&cur_warehouse=CN).

Settings: 
```json
$$
{"r":{"sys":{"fb":100.26,"fbs":"100.26-10-g1ff7","fbc":"settings_default.h","fv":0.99,"hp":3,"hv":0,"id":"0084-e739-29c6-29c","jt":0.75,"ct":0.01,"sl":0,"lim":1,"saf":1,"m48e":1,"mfoe":0,"mfo":1,"mtoe":0,"mto":1,"mt":2,"spep":1,"spdp":0,"spph":1,"spdw":1,"ssoe":0,"sso":1,"cofp":1,"comp":1,"coph":1,"tv":1,"ej":1,"jv":2,"qv":0,"sv":1,"si":250,"gpl":0,"gun":1,"gco":1,"gpa":2,"gdi":0}},"f":[1,0,3]}
{"r":{"1":{"ma":0,"sa":1.8,"tr":1.25,"mi":8,"su":1280,"po":0,"pm":0,"pl":0}},"f":[1,0,1]}
{"r":{"2":{"ma":1,"sa":1.8,"tr":40,"mi":8,"su":40,"po":0,"pm":0,"pl":0}},"f":[1,0,1]}
{"r":{"3":{"ma":2,"sa":1.8,"tr":1.25,"mi":8,"su":1280,"po":1,"pm":0,"pl":0}},"f":[1,0,1]}
{"r":{"4":{"ma":3,"sa":1.8,"tr":360,"mi":8,"su":4.44444,"po":0,"pm":0,"pl":0}},"f":[1,0,1]}
{"r":{"x":{"am":0,"vm":1000,"fr":1000,"tn":0,"tm":0,"jm":1000,"jh":1000,"hi":0,"hd":0,"sv":500,"lv":100,"lb":4,"zb":2}},"f":[1,0,1]}
{"r":{"y":{"am":0,"vm":1000,"fr":1000,"tn":0,"tm":0,"jm":1000,"jh":1000,"hi":0,"hd":0,"sv":500,"lv":100,"lb":4,"zb":2}},"f":[1,0,1]}
{"r":{"z":{"am":0,"vm":1000,"fr":1000,"tn":0,"tm":0,"jm":500,"jh":500,"hi":0,"hd":0,"sv":250,"lv":25,"lb":4,"zb":2}},"f":[1,0,1]}
{"r":{"a":{"am":0,"vm":288000,"fr":288000,"tn":-1,"tm":-1,"jm":288000,"jh":288000,"ra":0.199,"hi":0,"hd":0,"sv":144000,"lv":28800,"lb":5,"zb":2}},"f":[1,0,1]}
{"r":{"b":{"am":0,"vm":288000,"fr":288000,"tn":-1,"tm":-1,"jm":288000,"jh":288000,"ra":0.199,"hi":0,"hd":0,"sv":144000,"lv":28800,"lb":5,"zb":2}},"f":[1,0,1]}
{"r":{"c":{"am":0,"vm":288000,"fr":288000,"tn":-1,"tm":-1,"jm":288000,"jh":288000,"ra":0.199,"hi":0,"hd":0,"sv":144000,"lv":28800,"lb":5,"zb":2}},"f":[1,0,1]}
{"r":{"di1":{"mo":0,"ac":0,"fn":0}},"f":[1,0,1]}
{"r":{"di2":{"mo":0,"ac":0,"fn":0}},"f":[1,0,1]}
{"r":{"di3":{"mo":0,"ac":0,"fn":0}},"f":[1,0,1]}
{"r":{"di4":{"mo":0,"ac":0,"fn":0}},"f":[1,0,1]}
{"r":{"di5":{"mo":0,"ac":0,"fn":4}},"f":[1,0,1]}
{"r":{"di6":{"mo":0,"ac":0,"fn":0}},"f":[1,0,1]}
{"r":{"di7":{"mo":0,"ac":0,"fn":0}},"f":[1,0,1]}
{"r":{"di8":{"mo":0,"ac":0,"fn":0}},"f":[1,0,1]}
{"r":{"di9":{"mo":1,"ac":0,"fn":0}},"f":[1,0,1]}
{"r":{"do1":{"mo":1}},"f":[1,0,1]}
{"r":{"do2":{"mo":1}},"f":[1,0,1]}
{"r":{"do3":{"mo":1}},"f":[1,0,1]}
{"r":{"do4":{"mo":1}},"f":[1,0,1]}
{"r":{"do5":{"mo":1}},"f":[1,0,1]}
{"r":{"do6":{"mo":1}},"f":[1,0,1]}
{"r":{"do7":{"mo":1}},"f":[1,0,1]}
{"r":{"do8":{"mo":1}},"f":[1,0,1]}
{"r":{"do9":{"mo":1}},"f":[1,0,1]}
{"r":{"he1":{"e":false,"p":9,"i":0.12,"d":400,"st":0,"t":-1,"op":0,"tr":-1,"at":false,"an":0,"fp":1,"fm":0.4,"fl":40,"fh":40}},"f":[1,0,1]}
{"r":{"he2":{"e":false,"p":9,"i":0.12,"d":400,"st":0,"t":-1,"op":0,"tr":-1,"at":false,"an":0,"fp":0,"fm":0,"fl":0,"fh":0}},"f":[1,0,1]}
{"r":{"he3":{"e":false,"p":9,"i":0.12,"d":400,"st":0,"t":-1,"op":0,"tr":-1,"at":false,"an":0,"fp":0,"fm":0,"fl":0,"fh":0}},"f":[1,0,1]}
{"r":{"p1":{"frq":100,"csl":7900,"csh":12800,"cpl":0.13,"cph":0.17,"wsl":0,"wsh":0,"wpl":0.1,"wph":0.1,"pof":0.1}},"f":[1,0,1]}
{"r":{"g54":{"x":0,"y":0,"z":0,"a":0,"b":0,"c":0}},"f":[1,0,1]}
{"r":{"g55":{"x":0,"y":0,"z":0,"a":0,"b":0,"c":0}},"f":[1,0,1]}
{"r":{"g56":{"x":0,"y":0,"z":0,"a":0,"b":0,"c":0}},"f":[1,0,1]}
{"r":{"g57":{"x":0,"y":0,"z":0,"a":0,"b":0,"c":0}},"f":[1,0,1]}
{"r":{"g58":{"x":0,"y":0,"z":0,"a":0,"b":0,"c":0}},"f":[1,0,1]}
{"r":{"g59":{"x":0,"y":0,"z":0,"a":0,"b":0,"c":0}},"f":[1,0,1]}
{"r":{"g92":{"x":0,"y":0,"z":0,"a":0,"b":0,"c":0}},"f":[1,0,1]}
{"r":{"g28":{"x":0,"y":0,"z":0,"a":0,"b":0,"c":0}},"f":[1,0,1]}
{"r":{"g30":{"x":0,"y":0,"z":0,"a":0,"b":0,"c":0}},"f":[1,0,1]}
```

justinclift, 2018-06-29:
Ahhh, this sounds similar to when I was first experimenting with g2core, and hasn't enabled the motors (virtually) in the settings.

Gimme a few minutes and I'll jump on a box with a g2 controller connected.  I can't remember the JSON bit off hand, as I use text mode personally when doing stuff in a terminal. :wink:

justinclift, 2018-06-29:
Ahhh, yep.  I think that's the problem you're hitting as well.  In your x/y/z lines, it's showing the motors as disabled:

```
{"r":{"x":{"am":0,"vm":1000,"fr":1000,"tn":0,"tm":0,"jm":1000,"jh":1000,"hi":0,"hd":0,"sv":500,"lv":100,"lb":4,"zb":2}},"f":[1,0,1]}
{"r":{"y":{"am":0,"vm":1000,"fr":1000,"tn":0,"tm":0,"jm":1000,"jh":1000,"hi":0,"hd":0,"sv":500,"lv":100,"lb":4,"zb":2}},"f":[1,0,1]}
{"r":{"z":{"am":0,"vm":1000,"fr":1000,"tn":0,"tm":0,"jm":500,"jh":500,"hi":0,"hd":0,"sv":250,"lv":25,"lb":4,"zb":2}},"f":[1,0,1]}
```

That doesn't mean there's anything wrong, it's just the default settings for now, there is ongoing discussions to change that.

breiler, 2018-06-29:
Awesome it works, thanks a bunch! :clap:

aldenhart, 2018-06-30:
I added a Troubleshooting page to the Getting Started section where we can collect these kinds of issues.
https://github.com/synthetos/g2/wiki/Troubleshooting

breiler, 2018-06-30:
@aldenhart I personally am fine with all the motors disabled from the start. For the end user I think this should be handled by the UI giving the user a proper feedback. And for UI-implementors this is just a documentation issue (which you have solved for now with the troubleshooting guide).

This may actually not be a bug/issue, but instead a feature. =)
By having them disabled from the beginning will enable the UI-implementors to check if the GCode contain unintentional axis movement and warn the user. Just a thought for the ongoing discussion...