synthetos/g2 #366: Compiling on Win10 fails

JasonTitcomb, 2018-07-01:
Greetings.
I am attempting to compile g2core. I am running Win10 64 bit home 1803.
Running Atmel Studio 7 and have forked the code.
I am not strong with git but I cloned the repo to my machine and It looks like I got everything including the Motate folder contents.

The first compile error was related to Motate and downloading arm-none-eabi.
Don't even know what that is but it was failing so I manually installed it.

Then for some reason the compiler was looking for gcc-arm-none-eabi folder not arm-none-eabi so I just renamed the folder and that got me past the first compile error. ?
Man, this stuff is not for wimps.

Now I have one(I hope) more compile error and I am asking for some help.
Here is my output from the build if your interested.
Thanks in advance for any input.

------ Build started: Project: g2core, Configuration: G2v9k ARM ------
Build started.
Project "g2core.cppproj" (default targets):
Target "PreBuildEvent" skipped, due to false condition; ('$(PreBuildEvent)'!='') was evaluated as (''!='').
Target "CoreBuild" in file "C:\Program Files (x86)\Atmel\Studio\7.0\Vs\Compiler.targets" from project "C:\g2\g2core\g2core.cppproj" (target "Build" depends on it):
	Task "RunCompilerTask"
		Shell Utils Path C:\Program Files (x86)\Atmel\Studio\7.0\shellUtils
		C:\Program Files (x86)\Atmel\Studio\7.0\shellUtils\make.exe -C "C:\g2\g2core" -f "Makefile" CONFIG=ShopbotTestV9 COLOR=0 VERBOSE=2 DEBUG=2 
		make: Entering directory 'C:/g2/g2core'
		"Found that we're in Atmel Studio"
		DEVICE_INCLUDE_DIRS+USER_INCLUDE_DIRS:
		../Motate/MotateProject/motate/cmsis 
		../Motate/MotateProject/motate/cmsis/TARGET_Atmel/sam3x/include 
		../Motate/MotateProject/motate/cmsis/TARGET_Atmel 
		../Motate/MotateProject/motate/cmsis/TARGET_Atmel/preprocessor 
		../Motate/MotateProject/motate/cmsis/TARGET_Atmel/sam3x/include 
		../Motate/MotateProject/motate/cmsis/TARGET_Atmel/sam3x/source/templates 
		../Motate/MotateProject/motate/Atmel_sam_common 
		../Motate/MotateProject/motate/Atmel_sam3x 
		../Motate/MotateProject/motate/platform/atmel_sam 
		. 
		../Motate/MotateProject/motate 
		./board/G2v9 
		device/step_dir_driver
		DEVICE_LIB_DIRS+USER_LIB_DIRS: 
		../Motate/MotateProject/motate/platform/atmel_sam
		SOURCE_DIRS: 
		. 
		../Motate/MotateProject/motate 
		./board/G2v9 
		device/step_dir_driver
		C_SOURCES: 
		CXX_SOURCES: 
		./alarm.cpp 
		./canonical_machine.cpp 
		./config.cpp 
		./config_app.cpp 
		./controller.cpp 
		./coolant.cpp 
		./cycle_feedhold.cpp 
		./cycle_homing.cpp 
		./cycle_jogging.cpp 
		./cycle_probing.cpp 
		./encoder.cpp 
		./gcode_parser.cpp 
		./gpio.cpp 
		./help.cpp 
		./json_parser.cpp 
		./kinematics.cpp 
		./main.cpp 
		./marlin_compatibility.cpp 
		./persistence.cpp 
		./plan_arc.cpp 
		./plan_exec.cpp 
		./plan_line.cpp 
		./plan_zoid.cpp 
		./planner.cpp 
		./pwm.cpp 
		./report.cpp 
		./spindle.cpp 
		./stepper.cpp 
		./temperature.cpp 
		./text_parser.cpp 
		./util.cpp 
		./xio.cpp 
		../Motate/MotateProject/motate/MotateUtilities.cpp 
		../Motate/MotateProject/motate/main.cpp 
		./board/G2v9/board_stepper.cpp 
		./board/G2v9/board_xio.cpp 
		./board/G2v9/hardware.cpp
		FIRST_LINK_SOURCES: 
		./main.cpp 
		../Motate/MotateProject/motate/Atmel_sam_common/SamDMA.cpp 
		../Motate/MotateProject/motate/Atmel_sam_common/SamPins.cpp 
		../Motate/MotateProject/motate/Atmel_sam_common/SamPower.cpp 
		../Motate/MotateProject/motate/Atmel_sam_common/SamSPI.cpp 
		../Motate/MotateProject/motate/Atmel_sam_common/SamServiceCall.cpp 
		../Motate/MotateProject/motate/Atmel_sam_common/SamTimers.cpp 
		../Motate/MotateProject/motate/Atmel_sam_common/SamUART.cpp 
		../Motate/MotateProject/motate/Atmel_sam_common/SamUniqueID.cpp 
		../Motate/MotateProject/motate/Atmel_sam3x/SamUSB.cpp 
		../Motate/MotateProject/motate/platform/atmel_sam/syscalls.cpp
		The system cannot find the path specified.
		"Compiling c ../Motate/MotateProject/motate/cmsis/TARGET_Atmel/sam3x/source/templates/exceptions.c"; echo "    -> build/ShopbotTestV9-g2v9k/motate/cmsis/TARGET_Atmel/sam3x/source/templates/exceptions.o" 
		"../Motate/MotateProject/motate/../../Tools/win32/gcc-arm-none-eabi/bin"/arm-none-eabi-gcc  -Wall -Wchar-subscripts -Wcomment -Wformat=2 -Wimplicit-int -Werror-implicit-function-declaration -Wmain -Wparentheses -Wsequence-point -Wreturn-type -Wswitch -Wtrigraphs -Wunused -Wuninitialized -Wno-unknown-pragmas -Wfloat-equal -Wundef -Wshadow -Wpointer-arith -Wbad-function-cast -Wwrite-strings -Wsign-compare -Waggregate-return -Wformat -Wmissing-format-attribute -Wno-deprecated-declarations -Wredundant-decls -Wnested-externs -Winline -Wlong-long -Wunreachable-code -Wno-format-nonliteral -Wno-format-security -mcpu=cortex-m3 -mthumb -ffunction-sections -fdata-sections -std=gnu99  -DBOARD=g2v9k  -g3 -Os -I../Motate/MotateProject/motate/cmsis -I../Motate/MotateProject/motate/cmsis/TARGET_Atmel/sam3x/include -I../Motate/MotateProject/motate/cmsis/TARGET_Atmel -I../Motate/MotateProject/motate/cmsis/TARGET_Atmel/preprocessor -I../Motate/MotateProject/motate/cmsis/TARGET_Atmel/sam3x/include -I../Motate/MotateProject/motate/cmsis/TARGET_Atmel/sam3x/source/templates -I../Motate/MotateProject/motate/Atmel_sam_common -I../Motate/MotateProject/motate/Atmel_sam3x -I../Motate/MotateProject/motate/platform/atmel_sam -I. -I../Motate/MotateProject/motate -I./board/G2v9 -Idevice/step_dir_driver -DMOTATE_BOARD="G2v9k" -DSETTINGS_FILE="settings_shopbot_test.h" -D__SAM3X8C__ -D__sam3x__ -D__ARM__ -DGIT_VERSION=\"101.03-6-g7a25d-dirty\" -DGIT_EXACT_VERSION=101.03 -DDEBUG=1 -DIN_DEBUGGER=1 -MMD -MF build/ShopbotTestV9-g2v9k/dep/exceptions.o.d -MT build/ShopbotTestV9-g2v9k/motate/cmsis/TARGET_Atmel/sam3x/source/templates/exceptions.o -c -o build/ShopbotTestV9-g2v9k/motate/cmsis/TARGET_Atmel/sam3x/source/templates/exceptions.o ../Motate/MotateProject/motate/cmsis/TARGET_Atmel/sam3x/source/templates/exceptions.c
C:\g2\Motate\MotateProject\motate\Motate.mk(475,1): error: recipe for target 'build/ShopbotTestV9-g2v9k/motate/cmsis/TARGET_Atmel/sam3x/source/templates/exceptions.o' failed
		process_begin: CreateProcess(NULL, ../Motate/MotateProject/motate/../../Tools/win32/gcc-arm-none-eabi/bin/arm-none-eabi-gcc -Wall -Wchar-subscripts -Wcomment -Wformat=2 -Wimplicit-int -Werror-implicit-function-declaration -Wmain -Wparentheses -Wsequence-point -Wreturn-type -Wswitch -Wtrigraphs -Wunused -Wuninitialized -Wno-unknown-pragmas -Wfloat-equal -Wundef -Wshadow -Wpointer-arith -Wbad-function-cast -Wwrite-strings -Wsign-compare -Waggregate-return -Wformat -Wmissing-format-attribute -Wno-deprecated-declarations -Wredundant-decls -Wnested-externs -Winline -Wlong-long -Wunreachable-code -Wno-format-nonliteral -Wno-format-security -mcpu=cortex-m3 -mthumb -ffunction-sections -fdata-sections -std=gnu99 -DBOARD=g2v9k -g3 -Os -I../Motate/MotateProject/motate/cmsis -I../Motate/MotateProject/motate/cmsis/TARGET_Atmel/sam3x/include -I../Motate/MotateProject/motate/cmsis/TARGET_Atmel -I../Motate/MotateProject/motate/cmsis/TARGET_Atmel/preprocessor -I../Motate/MotateProject/motate/cmsis/TARGET_Atmel/sam3x/include -I../Motate/MotateProject/motate/cmsis/TARGET_Atmel/sam3x/source/templates -I../Motate/MotateProject/motate/Atmel_sam_common -I../Motate/MotateProject/motate/Atmel_sam3x -I../Motate/MotateProject/motate/platform/atmel_sam -I. -I../Motate/MotateProject/motate -I./board/G2v9 -Idevice/step_dir_driver -DMOTATE_BOARD=G2v9k -DSETTINGS_FILE=settings_shopbot_test.h -D__SAM3X8C__ -D__sam3x__ -D__ARM__ -DGIT_VERSION=\"101.03-6-g7a25d-dirty\" -DGIT_EXACT_VERSION=101.03 -DDEBUG=1 -DIN_DEBUGGER=1 -MMD -MF build/ShopbotTestV9-g2v9k/dep/exceptions.o.d -MT build/ShopbotTestV9-g2v9k/motate/cmsis/TARGET_Atmel/sam3x/source/templates/exceptions.o -c -o build/ShopbotTestV9-g2v9k/motate/cmsis/TARGET_Atmel/sam3x/source/templates/exceptions.o ../Motate/MotateProject/motate/cmsis/TARGET_Atmel/sam3x/source/templates/exceptions.c, ...) failed.
		make (e=2): The system cannot find the file specified.
		make: *** [build/ShopbotTestV9-g2v9k/motate/cmsis/TARGET_Atmel/sam3x/source/templates/exceptions.o] Error 2
		make: Leaving directory 'C:/g2/g2core'
	Done executing task "RunCompilerTask" -- FAILED.
Done building target "CoreBuild" in project "g2core.cppproj" -- FAILED.
Done building project "g2core.cppproj" -- FAILED.

Build FAILED.
========== Build: 0 succeeded or up-to-date, 1 failed, 0 skipped ==========

giseburt, 2018-07-02:
I’m not sure what you mean by “arm-none-eabi” ...”it was failing so I
manually installed it.” Can you possibly expand on (log file lines, etc) on
what it was complaining of when failing?

It downloads and installs (in a way that won’t collide with anything else
in your system) a very specific set of the ARM tool chain. If you chose a
newer one or older one than it was building then there may be issues.

What branch are you trying to build? I expect it’s likely the default
branch, edge. If that’s the case then it’s the 2016 Q2 update available
here now:
https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads

I don’t really recommend manual installation.

JasonTitcomb, 2018-07-02:
User error as usual.
Went through it all again and I was missing the step in the instructions that said 
>If your build fails with errors stating that it is unable to find gcc-arm-none-eabi, or similar

I edited the the Makefile as it said to point to my git install and now it compiles.
Thank you.

closed 2018-07-02 by JasonTitcomb