synthetos/g2 #368: Have to configured the Axis and Motors ?

shczm, 2018-07-10:
For a DUE, it seems that the axes and motors are deactivated in the default file, so there will be no movement. How to activate them?

Thanks!

breiler, 2018-07-10:
I had the same issue:
https://github.com/synthetos/g2/wiki/troubleshooting

shczm, 2018-07-10:
Thank you, Breiler!
I checked the configuration, am=1
[image]

shczm, 2018-07-10:
[image]

justinclift, 2018-07-10:
Cool.  That looks like the `x`, `y`, `z`, and `a` axis are enabled.  Would you be ok to grab a dump of all the settings (use the `$$` command) and paste them here as text?  The screenshot in your comment above is extremely hard to read (very small text size).

justinclift, 2018-07-10:
Btw for text, if you put it inside \`\`\` marks like this:

\`\`\`
foo
bar
baz
\`\`\`

It'll be displayed like this:

```
foo
bar
baz
```

That's probably the right display approach for settings dumps. :smile:

justinclift, 2018-07-10:
@shczm Out of curiosity, did you compile g2core yourself or are you using one of the downloads from the Releases page?  If it's one of the downloads, do you remember which one exactly?  That'll help, just in case it's one of the one's that uses a different pin out setting. :smile:

shczm, 2018-07-10:
>>> $$
[unhandled message] {"r":{"sys":{"fb":101.03,"fv":0.99,"fbs":"101.02-17-g51f9-dirty","fbc":"settings_shopbot_test.h","hp":"g2v9","hv":"k","id":"0084-d639-18c6-18c","jt":0.8,"ct":0,"zl":0.5,"sl":1,"lim":1,"saf":1,"m48":1,"froe":0,"fro":1,"troe":0,"tro":1,"mt":2,"tv":1,"ej":1,"jv":4,"qv":0,"sv":1,"si":250,"gpl":0,"gun":0,"gco":2,"gpa":2,"gdi":0}},"f":[1,0,3]}
{"r":{"1":{"ma":0,"sa":1.8,"tr":0.5,"su":3200.00024,"mi":8,"po":1,"pm":2,"pl":0.375,"ep":0,"sp":1}},"f":[1,0,1]}
{"r":{"2":{"ma":1,"sa":1.8,"tr":0.5,"su":3200.00024,"mi":8,"po":1,"pm":2,"pl":0.375,"ep":0,"sp":1}},"f":[1,0,1]}
{"r":{"3":{"ma":2,"sa":1.8,"tr":0.5,"su":3200.00024,"mi":8,"po":0,"pm":2,"pl":0.375,"ep":0,"sp":1}},"f":[1,0,1]}
{"r":{"4":{"ma":3,"sa":1.8,"tr":48,"su":33.33333,"mi":8,"po":0,"pm":2,"pl":0.375,"ep":0,"sp":1}},"f":[1,0,1]}
{"r":{"x":{"am":1,"vm":100,"fr":100,"tn":0,"tm":14,"jm":2,"jh":394,"hi":1,"hd":0,"sv":60,"lv":30,"lb":0.125,"zb":0.25}},"f":[1,0,1]}
{"r":{"y":{"am":1,"vm":100,"fr":100,"tn":0,"tm":11.5,"jm":2,"jh":394,"hi":3,"hd":0,"sv":60,"lv":30,"lb":0.125,"zb":0.25}},"f":[1,0,1]}
{"r":{"z":{"am":1,"vm":100,"fr":100,"tn":0,"tm":6,"jm":2,"jh":2,"hi":6,"hd":1,"sv":60,"lv":30,"lb":0.125,"zb":0.125}},"f":[1,0,1]}
{"r":{"u":{"am":0,"vm":39,"fr":39,"tn":0,"tm":0,"jm":39,"jh":39,"hi":0,"hd":0,"sv":20,"lv":3.94,"lb":0.15748,"zb":0.07874}},"f":[1,0,1]}
{"r":{"v":{"am":0,"vm":39,"fr":39,"tn":0,"tm":0,"jm":39,"jh":39,"hi":0,"hd":0,"sv":20,"lv":3.94,"lb":0.15748,"zb":0.07874}},"f":[1,0,1]}
{"r":{"w":{"am":0,"vm":39,"fr":39,"tn":0,"tm":0,"jm":20,"jh":20,"hi":0,"hd":0,"sv":10,"lv":0.98,"lb":0.15748,"zb":0.07874}},"f":[1,0,1]}
{"r":{"a":{"am":1,"vm":9144,"fr":48000,"tn":-1,"tm":-1,"jm":6000,"jh":6000,"ra":1,"hi":0,"hd":0,"sv":1524,"lv":762,"lb":3.175,"zb":9.525}},"f":[1,0,1]}
{"r":{"b":{"am":0,"vm":9144,"fr":9144,"tn":-1,"tm":-1,"jm":6000,"jh":6000,"ra":1,"hi":0,"hd":0,"sv":1524,"lv":762,"lb":3.175,"zb":9.525}},"f":[1,0,1]}
{"r":{"c":{"am":0,"vm":9144,"fr":9144,"tn":-1,"tm":-1,"jm":6000,"jh":6000,"ra":1,"hi":0,"hd":0,"sv":1524,"lv":762,"lb":3.175,"zb":9.525}},"f":[1,0,1]}
{"r":{"di1":{"mo":1,"ac":2,"fn":1}},"f":[1,0,1]}
{"r":{"di2":{"mo":2,"ac":2,"fn":1}},"f":[1,0,1]}
{"r":{"di3":{"mo":1,"ac":2,"fn":1}},"f":[1,0,1]}
{"r":{"di4":{"mo":1,"ac":2,"fn":1}},"f":[1,0,1]}
{"r":{"di5":{"mo":0,"ac":0,"fn":0}},"f":[1,0,1]}
{"r":{"di6":{"mo":1,"ac":3,"fn":3}},"f":[1,0,1]}
{"r":{"di7":{"mo":1,"ac":1,"fn":2}},"f":[1,0,1]}
{"r":{"di8":{"mo":0,"ac":7,"fn":0}},"f":[1,0,1]}
{"r":{"di9":{"mo":1,"ac":0,"fn":0}},"f":[1,0,1]}
{"r":{"do1":{"mo":2}},"f":[1,0,1]}
{"r":{"do2":{"mo":2}},"f":[1,0,1]}
{"r":{"do3":{"mo":2}},"f":[1,0,1]}
{"r":{"do4":{"mo":2}},"f":[1,0,1]}
{"r":{"do5":{"mo":2}},"f":[1,0,1]}
{"r":{"do6":{"mo":2}},"f":[1,0,1]}
{"r":{"do7":{"mo":2}},"f":[1,0,1]}
{"r":{"do8":{"mo":2}},"f":[1,0,1]}
{"r":{"do9":{"mo":2}},"f":[1,0,1]}
{"r":{"he1":{"e":false,"at":true,"p":9,"i":0.12,"d":400,"st":0,"t":-1,"op":0,"tr":-1,"an":0,"fp":0,"fm":0.4,"fl":0,"fh":0}},"f":[1,0,1]}
{"r":{"he2":{"e":true,"at":true,"p":9,"i":0.12,"d":400,"st":0,"t":-1,"op":0,"tr":-1,"an":0,"fp":0,"fm":0,"fl":0,"fh":0}},"f":[1,0,1]}
{"r":{"he3":{"e":true,"at":true,"p":9,"i":0.12,"d":400,"st":0,"t":-1,"op":0,"tr":-1,"an":0,"fp":0,"fm":0,"fl":0,"fh":0}},"f":[1,0,1]}
{"r":{"p1":{"frq":100,"csl":7900,"csh":12800,"cpl":0.13,"cph":0.17,"wsl":0,"wsh":0,"wpl":0.1,"wph":0.1,"pof":0.1}},"f":[1,0,1]}
{"r":{"g54":{"x":0,"y":0,"z":0,"u":0,"v":0,"w":0,"a":0,"b":0,"c":0}},"f":[1,0,1]}
{"r":{"g55":{"x":0,"y":0,"z":0,"u":0,"v":0,"w":0,"a":0,"b":0,"c":0}},"f":[1,0,1]}
{"r":{"g56":{"x":0,"y":0,"z":0,"u":0,"v":0,"w":0,"a":0,"b":0,"c":0}},"f":[1,0,1]}
{"r":{"g57":{"x":0,"y":0,"z":0,"u":0,"v":0,"w":0,"a":0,"b":0,"c":0}},"f":[1,0,1]}
{"r":{"g58":{"x":0,"y":0,"z":0,"u":0,"v":0,"w":0,"a":0,"b":0,"c":0}},"f":[1,0,1]}
{"r":{"g59":{"x":0,"y":0,"z":0,"u":0,"v":0,"w":0,"a":0,"b":0,"c":0}},"f":[1,0,1]}
{"r":{"g92":{"x":0,"y":0,"z":0,"u":0,"v":0,"w":0,"a":0,"b":0,"c":0}},"f":[1,0,1]}
{"r":{"g28":{"x":0,"y":0,"z":0,"u":0,"v":0,"w":0,"a":0,"b":0,"c":0}},"f":[1,0,1]}
{"r":{"g30":{"x":0,"y":0,"z":0,"u":0,"v":0,"w":0,"a":0,"b":0,"c":0}},"f":[1,0,1]}

justinclift, 2018-07-10:
Ahhh.  That's using a different settings file: `settings_shopbot_test.h`, which from memory uses different pin outs.

It sounds like you're using a plain Due board with no motors connected.  If that's the case, try flashing this .bin file, which will use the "standard" default pin outs

* https://github.com/synthetos/g2/releases/download/101.03/g2core-gShield-101.03.bin

When that's flashed, you'll need to turn on the motors again.  You should then see signals on pin `2` when X axis movement occurs. :smile:

shczm, 2018-07-10:
It works!
Thank you very much,  Justinclift!