synthetos/g2 #371: Activating/configurig pwm output from heater groups on arduino due

blwn23, 2018-07-27:
I am using g2core on an Arduino Due to control a small cnc router, which works very well. I've compiled the firmware from the latest edge branch with a custom pinout adapted from the gShield pinount.

Now i am trying to use G2core's heater groups, and i've hit a snag: i can't get the heater output pwm to work. It seems to me like a wrong pin mode configuration?

For testing purposes i have "stolen" the spindle-pwm pin, since i know this pin works and is capable of pwm-output. (other pins showed the same behaviour):
`kSpindle_PwmPinNumber    = -1; // to deactivate spindle output`
`kOutput1_PinNumber = 11; // to use the spindle pin as extruder output`

To get a temperature reading i have connected the hotend thermistor (with 4700R resistors as pullup and series) to an ADC pin :
`kADC1_PinNumber  = 64; `

Flashed onto the Arduino Due i get the following response:
`{he1:n}`
`{"r":{"he1":{"e":false,"at":true,"p":9,"i":0.12,"d":400,"st":0,"t":33.4,"op":0,"tr":77688.304,"an":3874,"fp":0,"fm":0.4,"fl":0,"fh":0}},"f":[1,0,9]}`
which tells me that the thermistor readout is working.

Activating the heater group does nothing
`{he1:{st:150}}`
`{he1:{e:0}}`
`{he1:n}`
`{"r":{"he1":{"e":true,"at":true,"p":9,"i":0.12,"d":400,"st":150,"t":33.7,"op":1,"tr":76521.835,"an":3871,"fp":0,"fm":0.4,"fl":0,"fh":0}},"f":[1,0,9]}`
Which tells me that the heating control loop is generating an output, but nothing changes on the designated output pin. The pin itself is working, since i can control my spindle VFD through the same pin.

What am i missing?

(full settings file: https://paste.ee/p/IKh0f, full pinout file: https://paste.ee/p/OsTY8)

justinclift, 2018-07-27:
Oops, sorry for the delay in answering @blwn23.  It looks like you've got the basics up and running (same as me), but I've personally not gotten around to looking at the heater stuff (yet - it's on my ToDo list).

Hopefully one of the main developers has some time over the weekend to look into this and assist.  :smile:

giseburt, 2018-07-28:
Hi @blwn23,

Since the temperature controls were added outside the normal GPIO section (this is being fixed in our next release), there are two oddities to be aware of.

The first is that in the pinout File you want to indicate (to the GPIO subsystem) that pwm is NOT available on those pins, so it doesn’t configure them. Instead they will be configured over in temperature.cpp, but only if...

See [here](https://github.com/synthetos/g2/blob/edge/g2core/temperature.cpp#L256) where if TEMPERATURE_OUTPUT_ON == 1 it sets up the heater output pins. You would want to put that in the pinout file, like [this](https://github.com/synthetos/g2/blob/edge/g2core/board/printrboardg2/printrboardG2v3-pinout.h#L104).

blwn23, 2018-07-28:
@justinclift no worries, i actually didn't expect such a fast reply!

@giseburt thanks! that explanation helps. `TEMPERATURE_OUTPUT_ON == 1` is actually already activated in my pinout file. That means `#define OUTPUT1_PWM 0` gets overwritten and should be `0`, correct?

I have poked around in this a bit more today, and have found some arduino pins, A4 and A6 (digital pins D58 & D60) seem to work. i think i have choosen an unsuitable pin before. 

Looking at `Motate/motate/Atmel_sam3x/motate_chip_pin_functions.h`
i suspect that the heater groups need a dedicated  `Motate::TimerChannel`?
Is there some documentation about which pins are pwm-capable without interfering with other functions? 

During my "poking around", i came across a potential typo in temperature.cpp:
in the template `HeaterFan`, the line 458 `float high_temp = MIN_FAN_TEMP;` should probably be `float high_temp = MAX_FAN_TEMP;`

giseburt, 2018-07-28:
Yes, define OUTPUTn_PWM that you want to use for the heater as 0.

Motate uses the lookup table in the pin_functions file you mentioned to
setup the correct timer automatically. However if two pins use the same
timer of the same type then you’ll get weird behavior.

Other than PWM, the system only uses a few times. They’re all listed in the
hardware.h file for your board. When possible they use timers that can’t be
attached to pins.

-Rob