synthetos/g2 #372: freertos, hal, stm32

vaughanatworld, 2018-07-29:
I am interested in porting g2core to run on freertos.
stcubemx auto generates a hal and configures freertos for the stm32 line of arm cortex machines. This might simplify Motate.
Before I go any further, is there interest and/or has there been any progress made along this line of reasoning?
Thanks
Brian

vaughanatworld, 2018-07-31:
Looking for a free generic development environment leads me to eclipse CDT and gnu compilers. Atmel Studio uses the Visual Studio project mechanism. Has anyone ported g2core to eclipse. Alternatively has anyone converted g2core projects to cmake? I understand cmake can be used with eclipse.

giseburt, 2018-08-01:
We recommend Atmel Studio as an available GUI, but it’s not an integral
part of the project. At its base g2core and Motate are makefile based
projects, and it automatically downloads the GCC precompiled by Arm in
order to control exactly which compiler is used. (Otherwise we spend a lot
of extra time maintaining backwards-compatibility with old compilers, and
waiting for the average available compiler to support newer language
features. This way we can use C++17 compiler features today, and we know
exactly what compiler bugs we can expect to deal with.)

Personally I use Atom and the command line to build and debug the project.
I occasionally use XCode, but used to much more. I find Eclipse overly
complex but it should work fine, and am by no means opposed to anyone else
adding and maintaining eclipse project files. We only ask that you document
it in the wiki and keep the files up to date.

As for cmake we have looked at it a few times. The only two issues with
cmake (not deal breakers, necessarily) are
1. It adds another dependency to track and install, and potential issue for
version mismatch. I believe there are ways around that, such as having
cmake build makefiles and then checking those into the project. Which leads
to:
2. With the current make configuration we are able to build for all
configurations and for all available boards simultaneously on the same
machine from the same directory without collision and with completely
independent output. I found this to be non-obvious how to achieve with
cmake. I would greatly appreciate it if anyone who knows cmake better than
I do could provide some assistance.

I’ve also been meaning to look into Visual Studio Code, and CLion. CLion
not bing free may be a deal breaker though.

vaughanatworld, 2018-08-02:
Got g2core to compile for SAM under SW4STM32 using the makefiles. Minimal fiddling.
My interest is in open development tools (no $) and as generic as possible.
The most general choices seem to be GNU/LLVM; Eclipse/Visual Studio; Anything else is becoming a niche player. 
The two areas that need standardization is a software package management system and project management. There are lots of solutions. It is a matter of the community picking one. Eclipse has lots of packages but it takes some effort to paw through the pile to get a coherent development environment. IBM is trying to create some order with with their market place.
Microsoft, with hind sight, looks poised to take the lead with VSC. That is good for you with Atmel Studio morphing into VSC.
For a duffer like me, I'm looking for example/sample code that I can mash into something interesting. For now, that means pulling code from vendor specific environments to a generic environment. 

PS. Nice work on g2core. I built a hand wired shield to interface to a donated PhoenixGS.
Now I'm building a g2core controller for a "RCFoamCutter" (STM32 based, just because).

mnesarco, 2019-10-23:
Hi @vaughanatworld I am interested in what you did to port g2core to STM32. Do you plan to share your work in a g2core fork? Or a kind of tutorial/blog to reproduce your success?

Thanks in advance.

mhlong10, 2019-10-24:
I would be interested in an STM32 port.  About a year ago I made an attempt at porting g2core.  I also picked up a couple of stm32nucleo-F411RE boards for $13 each.  My plan was to use these on a new CNC I was building.  Ultimately I spent enough time designing and building the CNC hardware that I just wanted to get it up and running, so I punted on the port, picked up a DUE and built a custom shield.

justinclift, 2019-10-25:
@mhlong10 Would you be ok to take a few photos of your custom shield?  Just wanting to see how my dodgy hand soldered first attempt compares to it.

Maybe we should have a competition, as a few people have mentioned making their own ones already... :grinning:

mnesarco, 2019-10-25:
I have found this project: https://github.com/thomast777/grbl32 it looks amazing too. GRBL 1.1f on 32bits ARM with support up to 6 axis. It is based on STM32. I wonder what are the advantages of g2core against GRBL 1.1 these days (2019). It looks like 6th order acceleration planning is the big difference, the JSON protocol is nice but GRBL protocol is more "standard".

I am using g2core + modified (trivial 3.3v mod) Protoneer's cncShield v3 and I am happy with the results, but in future builds I want to try GRBL1.1 with STM32.

mhlong10, 2019-10-25:
This shows the entire control box.  The shield is on the bottom left.  On the far left is the Raspberrypi (I used USB to connect to DUE as that seems more robust than serial port).  Top is 500 watt 36v supply and to the right of that are a couple of regulators for supplying 5v and 12v.  On the bottom right are all the stepper drivers, 3 closed loop for X & Y and 2 open loop for Z & A.  I used aviation connectors for all external connections to machine.

[image]

This shows a closer up view of the shield.

[image]

mhlong10, 2019-10-25:
One other thing to note is that g2core on DUE kicks butt.  I was doing some speed tests with my machine and was easily able to hit 30,000mm/min with 8x micro steps on 16mm ball screws drive.  That is 1,875 RPM and 50,000 steps/second on multiple axis at the same time.  I decided to cut that speed in half for normal jog use since it is actually scary and dangerous to have the gantry coming at you that fast.

Here is the machine custom designed and built machine:
[image]

And a video of the speed test before I put the spindle or waste board on:

https://www.youtube.com/watch?v=tgK-xsQH6wA

justinclift, 2019-10-26:
Wow, that is *so* much better put together than mine!

(Mine was *super* dodgy in comparison! :wink:)

justinclift, 2019-10-26:
@mhlong10 With the A axis stuff you're doing, what's the approach you're going with?

Not seeing anything obviously 4th-axis in the video, and am wanting to do similar myself in near-ish future. :smile:

mhlong10, 2019-10-26:
@justinclift  I have enough height that I mount the rotary axis right on my waste board and then connect its stepper to the controller via an aviation connector.  For shorter lengths (less than 24") I mount across the X axis and for longer spindles I mount in Y axis.

On another note, the carving area also goes beyond the Y at the front of the bed by about 1.5 inches which means I can carve edges and ends of boards.  I did this recently for tenons and also pockets for hinges.

justinclift, 2019-10-26:
Awesome.  That's probably the approach I'll copy too, at least initially anyway. :smile:

Tardusas, 2021-11-22:
@vaughanatworld could you share your experience in porting g2core?

I am interested in porting g2core to stm32f429vgt6 on skr 2 board. 

Thanks!

closed 2023-02-17 by ril3y