synthetos/g2 #373: 3D printing Hardware - Duet3D

jonbob0687, 2018-08-03:
I too am working towards g2core firmware to control a 3d printer similar to @pbannister. I have been using g2core for several years on my CNC mill and love the firmware. I run it on a due with a custom shield driving copley servo drives.

I am still in the research and design phase of my project and looking at hardware options for the 3d printer.

has anyone looked at running g2core on the Duet3D hardware? it looks like a really great hardware option. From the surface, it seems like it's meant to be mostly hardware compatible with the Due which makes me think it wouldn't take too much effort to get up and running. obviously it has a rich set of dedicated IO for 3d printing.

https://duet3d.dozuki.com/Wiki/Hardware_Overview

Is their any update on gQuintic release date? at least it seems like there is an active community behind the Duet3D project.

There are plenty of other firmwares out there for 3D printing. I like g2core because of the dedication to accurate, smooth motion control and planning strategies. the features are well implemented and it's come a long way even since i started using it. I would really like to see it become a more popular option in the 3D printing communities as a solid 3D printing firmware option.

myk3y, 2018-10-09:
You can compile for the gquintic-b board right now:  make BOARD=gquintic-b

jonbob0687, 2018-10-10:
Thanks Mike. I was actually searching for a way to purchase a gquintic board as they seem quite illusive on the interwebs.

I have received my Duet3D Wifi and I have to say, the user interface is exceptional. The browser based UI to load files directly to SD card space and configure the firmware without re-compiling is awesome since i tend to tinker a lot. im currently running their RTOS based reprapfirmware. Personally, i feel like the lack of SD card or other non-volatile memory on the DUE is one of the biggest downsides to the g2core firmware at this point. It took 5 min to go through the web-configurator tool and setup and save all the hardware configuration for the firmware. I didn't have to download any toolchains to compile and flash anything.

When i get some time, i will see about building g2core for the Duet3D board since i have one now. Currently, my design time is consumed with the actual printer build, and not so much the firmware. I need to update my CNC as well since i think it's running edge 0.97 or thereabouts.
https://filameister.blogspot.com/

myk3y, 2018-10-10:
At £145, too rich for me, considering I can do the same job with a £10 pi clone and a Due and a cnc shield.

But, I will have a deeper look at how their ESP12 integrates. In my experience the ESP12E is just not up to the task of solid WiFi serial comms. The ESP32, sure, but not the 12E.

drphil3d, 2018-12-21:
printrboard g2 would work

closed 2023-02-17 by ril3y