synthetos/g2 #374: Feed value is interpreted as mm/min when using G20

breiler, 2018-08-05:
I'm trying to figure out how the feed speed is interpreted by g2core. In this GCode I'm switching to G20 and then a do a G1 movement two inches at the feed rate of 1000 (which I would assume is 1000 inches/min). In the status report it says that the velocity achieved is 39.37 inches/min.

```

>>> G20G91G1X-2F1000
{"r":{},"f":[1,0,17]}
>>> G90 G21
{"r":{},"f":[1,0,8]}
{"sr":{"posx":1.902,"mpox":48.299,"vel":39.37,"unit":0,"stat":5,"dist":1}}
{"sr":{"posx":1.736,"mpox":44.099}}
{"sr":{"posx":1.571,"mpox":39.899}}
{"sr":{"posx":1.405,"mpox":35.699}}
{"sr":{"posx":1.241,"mpox":31.525}}
{"sr":{"posx":1.076,"mpox":27.325}}
{"sr":{"posx":0.91,"mpox":23.125}}
{"sr":{"posx":0.745,"mpox":18.925}}
{"sr":{"posx":0.58,"mpox":14.726}}
{"sr":{"posx":0.415,"mpox":10.551}}
{"sr":{"posx":0.25,"mpox":6.351}}
{"sr":{"posx":0.085,"mpox":2.151}}
{"sr":{"posx":-0,"mpox":-0,"vel":0,"unit":1,"stat":3,"dist":0}}
```

If I use the gcode with feed rate 800 `G20G91G1X2F800`, I get the reported velocity of 31.5 inches/min.

Is there a unit conversion missing for the feed rate?

I'm using the version g2core 100.26

aldenhart, 2018-08-05:
Thanks for pointing this out. We'll take a look

breiler, 2018-08-05:
Wow, thanks for digging into this! I'll try to fix the code generating this gcode as you suggested.
Thanks!

closed 2023-02-17 by ril3y