synthetos/g2 #375: M100(...) and status report

smhty, 2018-08-18:
Hi Everyone,

I have a problem with the current edge version:
**Description**
If I change outputs via M100 command, for example M100({out1: 1}), I don't see any update in the status report. 
The interesting part is that if I send  M100({do1mo:1}) then I see the update in sr. 
I have already included both "out1" and "do1mo" in my status report, and if I manually call {sr:n}, I can see the "out1": 1 in the report. I also make sure that "out1" is 0 before sending the M100({out1: 1}) command

### sample log ### 
send:  M100({out5:1})
receive:  {'r': {}, 'f': [1, 0, 15]}
receive:  {'qr': 47, 'qi': 1, 'qo': 0}
receive:  {'qr': 48, 'qi': 0, 'qo': 1}
receive:  {'sr': {'stat': 3}}

send: M100({do1mo:1})
receive:  {'r': {}, 'f': [1, 0, 16]}
receive:  {'qr': 47, 'qi': 1, 'qo': 0}
receive:  {'qr': 48, 'qi': 0, 'qo': 1}
receive:  {'sr': {'stat': 3, 'do1mo': 1}}
### sample log ### 

I would like to check this with you, and see if you guys have any suggestion.

Thank you very much

smhty, 2018-08-18:
Hi Justin, 
Good point.
I actually send the command in line format (not json), and I don't see similar behavior in the master version.

justinclift, 2018-08-18:
Ouch.  Sounds like things are in a bit of a buggy state atm then.  Ugh. :frowning:

giseburt, 2018-08-19:
This is a bug. There should be a:

sr_request_status_report(SR_REQUEST_TIMED);

After right before the output exits, here:
https://github.com/synthetos/g2/blob/edge/g2core/gpio.cpp#L510

That will tell the status report subsystem to update and output anything
that’s changed.

If you feel comfortable with gut you can make that change, test it, and
open a pull request.

Either way thank you for pointing this out. I’m not sure how we didn’t
catch it before. Several of us (I believe ShopBot as well) use outputs in
status reports, so I guess we were only getting the update when something
else changed.

smhty, 2018-08-19:
@justinclift, I already did it. As you mentioned, it compiles fine but still doesn't update outputs.
I will do it again tomorrow, and let you know if I see any difference.
Thanks again for all your help, and suggestion.

justinclift, 2018-08-19:
Heh Heh Heh

My bad.  I didn't see your comment in time mentioning that.  I was creating the PR at the same time.

It sounds like that PR will need a bit more work then. :wink: