synthetos/g2 #379: New setting Ramps 13 board stepping motor not turning

geargh, 2018-09-25:
I made new setting for ramps 13 board like below and motor enable signal seems working but stepping motor not turning.
What's problem?
I used mk4duo with lamps 13 as these pin number setting and stepping motor turning well except temperature sensor.

Settins.h file:
Comment these lines for use default.
/*
#ifdef SETTINGS_FILE
#define SETTINGS_FILE_PATH <settings/SETTINGS_FILE>
#include SETTINGS_FILE_PATH
#endif
*/

gShield-pinout.h changed like below.

// We don't have all of the inputs, so we don't define them.
#define INPUT1_AVAILABLE 1
#define INPUT2_AVAILABLE 1
#define INPUT3_AVAILABLE 1
#define INPUT4_AVAILABLE 1
#define INPUT5_AVAILABLE 1
#define INPUT6_AVAILABLE 1
#define INPUT7_AVAILABLE 1
#define INPUT8_AVAILABLE 1
#define INPUT9_AVAILABLE 1
#define INPUT10_AVAILABLE 1
#define INPUT11_AVAILABLE 1
#define INPUT12_AVAILABLE 1
#define INPUT13_AVAILABLE 0

#define ADC0_AVAILABLE 0
#define ADC1_AVAILABLE 0
#define ADC2_AVAILABLE 0
#define ADC3_AVAILABLE 0

#define XIO_HAS_USB 1
#define XIO_HAS_UART 0
#define XIO_HAS_SPI 0
#define XIO_HAS_I2C 0

#define TEMPERATURE_OUTPUT_ON 0

// Some pins, if the PWM capability is turned on, it will cause timer conflicts.
// So we have to explicity enable them as PWM pins.
// Generated with:
// perl -e 'for($i=1;$i<14;$i++) { print "#define OUTPUT${i}_PWM 0\n";}'
#define OUTPUT1_PWM 0
#define OUTPUT2_PWM 0
#define OUTPUT3_PWM 0
#define OUTPUT4_PWM 0
#define OUTPUT5_PWM 0
#define OUTPUT6_PWM 0
#define OUTPUT7_PWM 0
#define OUTPUT8_PWM 0
#define OUTPUT9_PWM 0
#define OUTPUT10_PWM 0
#define OUTPUT11_PWM 0
#define OUTPUT12_PWM 0
#define OUTPUT13_PWM 0

namespace Motate {

// NOT ALL OF THESE PINS ARE ON ALL PLATFORMS
// Undefined pins will be equivalent to Motate::NullPin, and return 1 for Pin<>::isNull();

pin_number kSerial_RXPinNumber = 0;
pin_number kSerial_TXPinNumber = 1;
//    pin_number kSerial_RTSPinNumber                      =  8;   // added later
//    pin_number kSerial_CTSPinNumber                      =  9;   // added later

[... 2391 lines cut ...]
#ifndef USER_DATA_C0
#define USER_DATA_C0 0
#endif
#ifndef USER_DATA_C1
#define USER_DATA_C1 0
#endif
#ifndef USER_DATA_C2
#define USER_DATA_C2 0
#endif
#ifndef USER_DATA_C3
#define USER_DATA_C3 0
#endif

#ifndef USER_DATA_D0
#define USER_DATA_D0 0
#endif
#ifndef USER_DATA_D1
#define USER_DATA_D1 0
#endif
#ifndef USER_DATA_D2
#define USER_DATA_D2 0
#endif
#ifndef USER_DATA_D3
#define USER_DATA_D3 0
#endif

justinclift, 2018-09-25:
It's been ages since I looked at this personally, so I'm not 100% sure... but is there any chance it's the "motors do not move" problem mentioned here?

&nbsp; &nbsp; https://github.com/synthetos/g2/wiki/troubleshooting#motors-do-not-move

geargh, 2018-09-25:
I already read that thread.
I have not such problem. Already x, y. z is 1.
This is connection message.
{"ej":1}
{"sr":{"stat":true,"vel":true,"feed":true,"unit":true,"coor":true,"momo":true,"plan":true,"dist":true,"mpox":true,"mpoy":true,"mpoz":true,"mpoa":true,"posx":true,"posy":true,"posz":true,"posa":true}}
{"r":{"ej":1},"f":[1,0,10]}
{"sr":""}
{"qr":""}
{"sys":""}
{"x":""}
{"y":""}
{"z":""}
{"a":""}
{"r":{"sr":{"stat":3,"vel":0,"feed":1270,"unit":1,"coor":1,"momo":0,"plan":0,"dist":0,"mpox":0,"mpoy":0,"mpoz":0,"mpoa":0,"posx":0,"posy":0,"posz":0,"posa":0}},"f":[1,0,201]}
{"b":""}
{"c":""}
{"1":""}
{"2":""}
{"3":""}
{"4":""}
{"p1":""}
{"G54":""}
{"G55":""}
{"G56":""}
{"G57":""}
{"r":{"sr":{"stat":3,"vel":0,"feed":1270,"unit":1,"coor":1,"momo":0,"plan":0,"dist":0,"mpox":0,"mpoy":0,"mpoz":0,"mpoa":0,"posx":0,"posy":0,"posz":0,"posa":0}},"f":[1,0,11]}
{"r":{"qr":48},"f":[1,0,11]}
{"r":{"sys":{"fb":101.03,"fv":0.99,"fbs":"","fbc":"settings_shapeoko2.h","hp":"ArduinoDue","hv":"na","id":"0084-e739-18c6-9ce","jt":0.75,"ct":0.01,"zl":0,"sl":0,"lim":1,"saf":1,"m48":1,"froe":0,"fro":1,"troe":0,"tro":1,"mt":2,"tv":1,"ej":1,"jv":2,"qv":0,"sv":1,"si":250,"gpl":0,"gun":1,"gco":1,"gpa":2,"gdi":0}},"f":[1,0,12]}
{"G58":""}
{"G59":""}
{"r":{"x":{"am":1,"vm":50000,"fr":50000,"tn":0,"tm":500,"jm":5000,"jh":20000,"hi":1,"hd":0,"sv":3000,"lv":100,"lb":4,"zb":2}},"f":[1,0,10]}
{"r":{"y":{"am":1,"vm":50000,"fr":50000,"tn":0,"tm":500,"jm":5000,"jh":20000,"hi":3,"hd":0,"sv":3000,"lv":100,"lb":4,"zb":2}},"f":[1,0,10]}
{"r":{"z":{"am":1,"vm":1200,"fr":1200,"tn":-95,"tm":200,"jm":500,"jh":500,"hi":6,"hd":1,"sv":800,"lv":25,"lb":4,"zb":4}},"f":[1,0,10]}
{"r":{"a":{"am":3,"vm":77760,"fr":9720,"tn":0,"tm":10,"jm":648000,"jh":648000,"ra":5.30516,"hi":0,"hd":0,"sv":2000,"lv":2000,"lb":5,"zb":2}},"f":[1,0,10]}
{"r":{"b":{"am":3,"vm":77760,"fr":9720,"tn":0,"tm":10,"jm":648000,"jh":648000,"ra":5.30516,"hi":0,"hd":0,"sv":2000,"lv":2000,"lb":10,"zb":2}},"f":[1,0,10]}
{"r":{"c":{"am":0,"vm":450000,"fr":450000,"tn":-1,"tm":-1,"jm":45000,"jh":45000,"ra":57.29571,"hi":0,"hd":0,"sv":225000,"lv":45000,"lb":5,"zb":2}},"f":[1,0,10]}
{"r":{"1":{"ma":0,"sa":1.8,"tr":0,"su":inf,"mi":32,"po":0,"pm":2,"pl":0.5,"ep":0,"sp":0}},"f":[1,0,10]}
{"r":{"2":{"ma":1,"sa":1.8,"tr":0,"su":inf,"mi":32,"po":0,"pm":2,"pl":0.5,"ep":0,"sp":1}},"f":[1,0,10]}
{"r":{"3":{"ma":2,"sa":1.8,"tr":0,"su":inf,"mi":32,"po":0,"pm":2,"pl":0.5,"ep":0,"sp":1}},"f":[1,0,10]}
{"r":{"4":{"ma":6,"sa":1.8,"tr":0,"su":inf,"mi":32,"po":0,"pm":2,"pl":0.5,"ep":0,"sp":1}},"f":[1,0,10]}
{"r":{"p1":{"frq":100,"csl":0,"csh":1000,"cpl":0.13,"cph":0.17,"wsl":0,"wsh":1000,"wpl":0.13,"wph":0.17,"pof":0.1}},"f":[1,0,11]}
{"r":{"g54":{"x":0,"y":0,"z":0,"u":0,"v":0,"w":0,"a":0,"b":0,"c":0}},"f":[1,0,12]}
{"r":{"g55":{"x":0,"y":0,"z":0,"u":0,"v":0,"w":0,"a":0,"b":0,"c":0}},"f":[1,0,12]}
{"r":{"g56":{"x":0,"y":0,"z":0,"u":0,"v":0,"w":0,"a":0,"b":0,"c":0}},"f":[1,0,12]}
{"r":{"g57":{"x":0,"y":0,"z":0,"u":0,"v":0,"w":0,"a":0,"b":0,"c":0}},"f":[1,0,12]}
{"r":{"g58":{"x":0,"y":0,"z":0,"u":0,"v":0,"w":0,"a":0,"b":0,"c":0}},"f":[1,0,12]}
{"r":{"g59":{"x":0,"y":0,"z":0,"u":0,"v":0,"w":0,"a":0,"b":0,"c":0}},"f":[1,0,12]}
{"sr":{"stat":3}}

geargh, 2018-09-26:
Yes, It will better.
I didn't format because I make it  for only test.

closed 2023-02-16 by ril3y