synthetos/g2 #380: Universal Gcode Platform

jschoch, 2018-10-03:
How does the config loading work for UGS?  I get a "no firmwares found" error.

Also it isn't really at all clear how to use it with g2core.  Are there instructions anywhere?  A video would be very helpful.

justinclift, 2018-10-04:
Ahhh.   Just to check, which version of UGS are you using? :smile:

The support for g2 is in the nightly builds, not yet in the main UGS releases.  Just to check, I grabbed the "UGS Platform" download from here and tried it successfully:

&nbsp; &nbsp; https://github.com/winder/Universal-G-Code-Sender#downloads

With that (in theory) there should be an option to set the controller type being connected to.  eg GRBL, TinyG (g2core), (etc).  eg:

[image]

Does that help? :smile:

jschoch, 2018-10-04:
I can connect but I get a "no firmwares" error, the image below, or once I was able to get a firmware setting page but i don't really know how.  I can guess it was after trying cncjs or some other gui.  I'm using the latest nightly version of UGS platform 64bit and the latest 101.03  .bin on a due.

[image]

I love the 32 bit processor and motion planning but honestly the experience of trying to understand how to get g2 up and running has been extremely frustrating.

justinclift, 2018-10-04:
Yeah, I very much get where you're coming from.  Once you have it up and running it tends to just keep on working, but the learning curve isn't super simple. :frowning: 

As a thought, since I don't personally have time to get into this properly, I'd recommend pinging the UGS folk.  At least @breiler (added the TinyG/g2core support to UGS) might be able to point you in the right direction. :smile:

breiler, 2018-10-04:
I would recommend that you try the g2core 100.26 version. I've had some issues with the later versions.

closed 2023-02-16 by ril3y