synthetos/g2 #381: problem flashing defualt .bin

jds598, 2018-10-08:
Every time I attempt to flash the default .bin file I get an error stating permission denied. I am on windows 10. I am novice when it comes to using command prompt but have tried everything I can think of, please help!

justinclift, 2018-10-08:
Ahhh.  It sounds like when you start the Command Prompt, you're *not* started it with Administrative privileges.

Personally, I only have Win 7 available, so I'm not sure about Win 10.

In Win 7, you can click the Start button to bring up a menu, and on that is a search tool for locating the command you want to run.  If I type in `cmd`, then it displays and entry for `cmd.exe`.  When I right click that (to bring up the options menu for the entry), one of the options is to run it with "Administrative" privileges (or wording to that effect).  That's the one you'd want.

Win 10 should have something similar.  Doing a quick search online showed up several websites for what seems like correct instructions.  Here's a random one from the list,:

&nbsp; &nbsp; https://www.intowindows.com/command-prompt-as-administrator-in-windows-10/

Hopefully that helps. :smile:

jds598, 2018-10-08:
Thanks for the quick reply! Ill give it a try tomorrow and let you know if it works.

justinclift, 2018-10-09:
Well, we're making progress. :smile:

The Administrative console was a good first attempt, so we're just running through the options until success is achieved.

The behaviour now sounds like what I've seen with my normal setup (Linux based), when I accidentally have some other program using the Arduino when I do the flashing.  eg the Ardunio IDE, Minicom (similar to [CoolTerm](http://freeware.the-meiers.org/) on windows).

With a com port (COM10 in your case), only one program can (safely) access it at a time.

Is there any chance some other program is using it while you're running the bossac command?  If so, close that other program and try again.  In theory (!) it should then work. :smile:

jds598, 2018-10-09:
First I'd like to say thanks for the help I really appreciate it. I did figure out the problem, it was me not following the directions close enough. I didn't put the name of the .bin in the command line 😑, once I did that it worked!

justinclift, 2018-10-09:
Awesome. Glad it turned out to be a simple solution, they're the best ones. :smile:

closed 2018-10-09 by justinclift