synthetos/g2 #383: Due USB comms not working when self-powered

lutorm, 2018-10-09:
I don't know if this is an issue with the Due hardware, with g2, or with the Macbook Pro I use to connect to it, but I've noticed that if I turn power on to the controller and then plug in the USB, no ports are detected. If I plug the USB in first, ports are detected and I can subsequently power the controller on without interrupting it. If I subsequently unplug the USB and plug it back in, I'm back to no ports.

justinclift, 2018-10-09:
Interesting.  Haven't heard of that happening before.

Not sure what could be causing it though.  Maybe something going a bit wrong with the g2 code's internal boot sequence?

Hmmm, which version of g2 do you have flashed to the Due?  If it's not the 100.26 release (ye old "stable" master release) then give that one a go and see if it still happens.  That master version (in theory) should be fairly well battle tested by now. :smile:

lutorm, 2018-10-14:
I'm using 7a25dbf (101.03, g2/edge as of June). It's not easy for me to change version because I need to re-port my config, and my changes don't apply cleanly to the master.

justinclift, 2018-10-15:
k, no worries.  We'll need one of the main developers to investigate then. :smile:

turboproc, 2018-12-28:
Having a similar issue with one of my DUE's running Repetier, so doesn't look g2 related.

lutorm, 2021-04-15:
I was looking at this again and I suspect it's this issue: https://forum.arduino.cc/index.php?topic=256771.75

MitchBradley, 2021-04-15:
On my g2core/due setup connected to a Raspberry Pi running CNCjs, I have a Pi GPIO connected to the Due's reset line.  I have a shell script that pulses the reset before starting CNCjs.  Without that, the Due will not start g2core.  Instead, it comes up in programming mode.

lutorm, 2021-04-15:
I can confirm that one if the fixes in the thread above (soldering a 10k resistor across a transistor) works, my Due is now accessible even if it's first powered up with the DC jack.

However, if I unplug the USB and then plug it back in, it still does not respond. It enumerates the ttyACM0 port when ?I plug in the USB, but connecting to it with a terminal I get nothing. Is reconnecting the USB supposed to work?

It would be great if I could do this because I'd like to be able to leave the controller powered on and just reconnect to it without having to re-home and re-upload configs.

closed 2023-02-16 by ril3y