synthetos/g2 #384: ask question about G2 for porting to STM32F415 MCU

wjzhang, 2018-10-12:
Hi:
    I'm now try porting tinyG2 edge branch to STM32F415 MCU.  
    i use USART to communicate with PC.
   I have some questions:
1) the MoatateBuffer:  the RXBuffer to start transfer, it call UART/UARTUSB/... module startRXTransfer() function, it seem unbreakable, it need receive all request datas.  first time it about 1024 bytes. then return. 
why use this design? 

2) the  dda_timer, exec_timer, fwd_plan_timer  are just 3 simple timer, No necessary use necessary.  right?

justinclift, 2018-10-12:
@aldenhart @giseburt ping

wjzhang, 2018-10-14:
sorry, type error. question 2: "No necessary use TimerChannel"

giseburt, 2018-10-15:
Hello,

Sorry for the slow response.

1. The transfer is generally handled using DMA or DMA-like interrupts, and so there are two asynchronous loops nested here. The outer-most loop exists completely in g2core, and consists of other parts of the system calling readline (which there is a global readline that calls the individual port readlines and eventually the `LineRXBuffer` one [here](https://github.com/synthetos/g2/blob/edge/g2core/xio.cpp#L821)) and from there the process goes roughly:
  1. If the buffer is not full, call `LineRXBuffer::_restartTransfer()` which is actually [`RXBuffer::_restartTransfer()`](https://github.com/synthetos/Motate/blob/41e5b92a98de4b268d1804bf6eadf3333298fc75/MotateProject/motate/MotateBuffer.h#L230) which handles all the logic to determine what contiguous space in the circular buffer can be filled next and then calls [`_owner->startRXTransfer(write_pos, transfer_size, write_pos_extra, transfer_size_extra)`](https://github.com/synthetos/Motate/blob/41e5b92a98de4b268d1804bf6eadf3333298fc75/MotateProject/motate/MotateBuffer.h#L298)
  2. `_owner` is the object that handles the hardware directly. It can be a UART, USART, USB serial, etc. RXBuffer doesn’t care, really. RXBuffer just asks that something setup the hardware to fill this buffer asynchronously.
  3. When the transfer is “done” (as done as it will get, either fully or as much as the hardware can for now) then the function set in `_owner->setRXTransferDoneCallback(...)` gets called by `_owner`. Note that `RXBuffer` sets that.
  4. At some point later `_restartTransfer()` is called again from the outer logic and the loop continues.

`TXBuffer` Works in much the same way, except g2core doesn’t really wrap any extra logic around it, so it’s used directly. (`LineRXBuffer` handles the line and special character processing that `RXBuffer` knows nothing about, and shouldn’t.)

2. I’m not sure about this question. Yes, they are somewhat simple timers, in that they simply setup the hardware timer to fire an interrupt at the requested frequency at the requested interrupt level. However, they are vital to the functionality of g2core, so they must be used and available functionality.

giseburt, 2018-10-15:
Note I updated my last comment, so if reading via email you might want to check the website.

wjzhang, 2018-10-15:
giseburt, Thanks!  I see.  the _owner->startRXTransfer() is prepare receive datas,  and use _owner->getRXTransferPosition() update real receive datas.

Tardusas, 2021-11-15:
Hi, has anybody ported G2 to STM32F407VGT6, is it possible?

I have an SKR 2.0 board with STM32F407VGT6 I would love to use g2core with TMC2209 without any additional shields.

@wjzhang have you managed to port it to STM32F415?

wjzhang, 2021-11-15:
@Tardusas:
 No. It's postponed.

closed 2023-02-17 by ril3y