synthetos/g2 #385: g2 upload to Due -> flash operation exceeds flash size

patrick-brouille-habeille-com, 2018-11-02:
Hello,
Got a weird stuff...
Compiling g2 OK as an elf file.
Using bossac on Widows 10 to upload firmware lead to an error message after a second:
**"flash operation exceeds flash size"**
Tried several times:
 - push reset on board (>1s),
 - wait 10s at least,
 - unplug,
 - prepare bossac command line,
 - plug in usb,
 - hit enter as soon as Widows recognise
Also tried with an elf file downloaded from Github, same result.
I have no Atmel ICE and not the budget to buy one.
Google was not my friend since I found no one having this issue...
Please help,
Thanks

turboproc, 2018-12-28:
Are you sure you're flashing the right file? Build process results in a g2core.bin file which has a size of approx  190kB and a g2core.elf file which has a size of 4Mb. Made this mistake once myself so recognizing the issue.

patrick-brouille-habeille-com, 2019-01-06:
Hello,

Thanks for your answer.

Had to switch to another solution. Could not wait for answer. Bought a 
Mesa Electronic board (7i96) with Linux CNC. Works great out of the box.

No more G2 for me.

Wish you great success.

Best regards,

Patrick