synthetos/g2 #386: Compatible with BCNC

MARIOBASZ, 2018-11-21:
Is there intention to make G2 Core work in BCN?
I find it useful platform, in fact I make some small contributions, and specifically I would like to take advantage of the advantages of jerk control in trochoidal milling
The work must be done from here? From there? On the whole?

justinclift, 2018-11-21:
Support for bCNC would definitely be of interest from our point of view.

The primary/initial author of bCNC (@vlachoudis) expressed interest in adding TinyG/g2core support in https://github.com/vlachoudis/bCNC/issues/83. :smile:

When I tried to follow up though, there's been no response. :frowning:

So, not really sure how to get things moving forward.  Any ideas? :smile:

justinclift, 2018-11-21:
Ahhh, this looks related too: https://github.com/vlachoudis/bCNC/issues/844