synthetos/g2 #396: "msg":"Internal error - mp_exec_move()"

rarens, 2019-01-15:
@justinclift 
I seem to get that error randomly what causes it?

rarens, 2019-01-15:
I also get this error randomly:
{"er":{"fb":101.03,"st":157,"msg":"Arc specification error - missing offset(s) - arc offsets missing or zero"}}

justinclift, 2019-01-15:
Sorry, @rarens, I'm definitely not the guy to ask for this.  I don't have any of my g2core nor CNC gear with me (I've moved country), and haven't really looked at the code base in any depth in months.

It sounds like you're hitting bugs, which isn't a good sign. :frowning:

justinclift, 2019-01-15:
Oh... that arc specification error.  I **think** that's been mentioned before somewhere.

What are you using to generate the GCode?  If it's something like Fusion 360, from memory there should be an option in it to generate stuff with arcs differently.  I don't remember the exact setting off hand, but take a look and have a muck around with them?  eg try toggling options / changing decimal units of precision / etc

rarens, 2019-01-15:
Sorry @justinclift I meant to tag @giseburt.  Yes I do use fuse to generate gcode ill do some poking around.

rarens, 2019-01-15:
Yes im using the tinyg post, and Ive tried it with every combo of flags.  I would say its the gcode if it errors at the same line every time but sometimes it line 200 and sometimes 10000 and sometimes it runs without a hitch.  Ill continue to run tests and see if there is some condition that causes it.

rarens, 2019-01-17:
@giseburt  I noticed the me_exec_error usually gets thrown when the queue report indicates 47 or 48 open slots.  Does this have anything to do with it?

ep1cman, 2019-07-31:
Hi, I have been having an issue with this error also randomly occurring. I am running off the latest code and it seems to happen sporadically. Some time it will happen every few commands sent, other times it can run fine for hours. Do you have any advice on tracking down the cause of this error?

justinclift, 2019-07-31:
@MitchBradley Any ideas? :smile:

MitchBradley, 2019-08-01:
@ep1cman  - could you be more specific about "latest code" - i.e. give a branch and commit tag? I just want to make sure I'm not accidentally looking in the wrong place.

The clue from @rarens seems interesting to me.  I can't find anywhere in the code where a bf structure is allocated but bf->func is not set, but perhaps there is an edge or race condition where a bf is freed while still being processed, or something like that.  One diagnostic would be add some printouts prior to that panic - print the other bf fields, for example.  That might tell us whether the bf is completely clear, as it would be after having been freed, or whether just bf->func is NULL with the other fields valid.

MitchBradley, 2019-08-01:
I am starting to like the race condition hypothesis.  mp_exec_move() does not depend on any of the other bf fields unless bf->block_type == BLOCK_TYPE_ALINE - which would be false if the bf struct had been cleared by its reset() method.  Unfortunately, I have not studied the control flow and interrupt structure well enough to reason about it.

ep1cman, 2019-08-01:
So looking it with a jtag debugger attached it definitely is a race condition between `mp_aline` and `mp_exec_move`.

`mp_exec_move` is executed by a timer interrupt that can happen in the middle of mp_aline, halfway through setting up the buffer. `mp_commit_write_buffer` should protect against this except there is a bug in `mp_get_run_buffer` where it only check if the buffer state is `MP_BUFFER_EMPTY` but should also check if it is `MP_BUFFER_INITIALIZING` state.

After implementing this change, the issue seems to go away. My particular usecase that delivers dynamic movements, and therefore kept the queue almost empty caused this to happen quite often.

I will submit a pull request with the fix in a minute

PR #421 (2019-08-01): Fixed race condition between `mp_exec_move` and `mp_aline`
There exists a race condition between `mp_exec_move` and `mp_aline`.
When `mp_aline` is modifying a buffer it is possible for the timer interrupt to trigger `mp_exec_move` halfway though the buffer being constructed.

This should be guarded against by using `mp_commit_write_buffer` however, `mp_get_run_buffer` does not properly check if it able to get the run buffer, if the buffer remaining is the one being written to. Both `MP_BUFFER_EMPTY` and `MP_BUFFER_INITIALIZING` both indicate that the buffer should not be used. 

Fixes #396 

A [... cut]
closed 2019-08-01 by justinclift