synthetos/g2 #401: Compile Under WSL

MitchBradley, 2019-02-08:
I want to compile on Windows using the new "Windows Subsystem for Linux" feature, which lets you run Linux programs on Windows with much less overhead than with a VM. You can use the Linux cross compilers, make, etc, so you don't have to deal with Atmel Studio and all that bloat. Access to the Windows filesystem is straightforward, avoiding the irritations of dealing with VM or Docker mount points.

The main problem so far is that the Linux toolchain that Motate currently uses - gcc-arm-none-eabi-5_4-2016q2 - is an ELF32 binary, which WSL doesn't (and reportedly never will) support.  There are newer toolchains that are ELF64 binaries, but there are some problems:

1. The new toolchains are now hosted at developer.arm.com instead of launchpad.net , and the URLs to access them are formatted differently.  This affects the scripting in Motate/Tools/Makefile .  For testing, I downloaded and installed the toolchain manually.  Going forward, that Makefile needs to be revised.
2. The new toolchain exposed some coding errors as detailed in #400 .
3. The new toolchain has problems with lines like (in SamCommon.h):

`constexpr Uart* const UART0_DONT_CONFLICT = UART;
`
The error is "error: reinterpret_cast from integer to pointer"

I made the problem go away by changing "constexpr" to "static" in the places where the left hand side was a pointer and the right hand side was derived from an integer.  I'm not sure if that is the best way to handle it.

The current status is that I have a clean build for the Othermill target.  The next step is to build for my hardware and see if it works.

giseburt, 2019-03-07:
Hi Mitch,

Sorry for the long delay.

If you look at branch [dev-168-gquintic](https://github.com/synthetos/g2/tree/dev-168-gquintic) and follow to the motate branch it uses (a slightly older commit in [sams70-port](https://github.com/synthetos/Motate/tree/sams70-port)) You'll see that it's using a newer compiler from developer.arm.com as you mentioned.

It does use a newer compiler, but I believe the one you mention in #400 is newer yet, and I haven't tested with that yet. (I'll comment over there shortly.)

Out of curiosity, were you able to get it to work? (You mentioned you needed to test it.)

MitchBradley, 2019-03-13:
Sorry for the delay in responding - I thought I had replied immediately but something didn't  go through...
Yes, my WSL-hosted compilation is working.  The machine is running well with the resulting binary.

ril3y, 2023-02-17:
https://github.com/synthetos/g2/commit/925b4ecf750d0b3494bc509876b4a6024a3ac52f

closed 2023-02-17 by ril3y