synthetos/g2 #406: collect2: fatal error: ld terminated with signal 9 [Killed: 9]

spothus, 2019-04-01:
I've got this error when compiling as per explaining in the wiki:

collect2: fatal error: ld terminated with signal 9 [Killed: 9]

[image]

I've no idea what is happening.
Thanks

spothus, 2019-04-01:
I tried with a more simple directory but the result is not any better:

[image]

It looks like it is gcc related ?

giseburt, 2019-04-01:
I looked around and signal 9 means that the process used too much ram and
the system killed it.

On OS X that usually means your boot drive is too full for it to
effectively use as swap space.

Use Activity Monitor to see if you have anything else open that’s eating a
bunch of RAM and try to free up RAM and boot drive space.

spothus, 2019-04-02:
Thanks for your answer.

I tried but it looks like I've enough memory...

[image]

karoria, 2019-04-02:
Try with command line. In my case I got succeeded with command line. I can't recall the error that time I was getting with XCode. That might be because XCode is using more RAM.

karoria, 2019-04-02:
You will need to install command line tools for xcode.

spothus, 2019-04-02:
same thing with command line and I've installe the commande line tools for Xcode

karoria, 2019-04-02:
Bossac and motate paths are defined? I mean it should have the hierarchical folders which you need to manually put there. I am not from software or electronics background so dont know much about signal 9. But i can only suggest to use command line with above paths well defined.

spothus, 2019-04-04:
I've installed Github desktop and downloaded the project via this app. It works both via make and Xcode now... Anyway I don't know what was the issue. The only thing was that the Motate folder was empty when I downloaded the zip file and I had to perform the submodule command.
Thanks for your time

ralpha, 2021-09-22:
I fixed this issue before by switching linker from `ld` (GNU, default) or `lld` (LLVM):
To switch to `lld` I used: `.cargo/config.toml`: (make sure `lld` is installed)
```toml
[target.x86_64-unknown-linux-gnu]
rustflags = [
    "-C", "link-arg=-fuse-ld=lld",
]
```
https://github.com/xd009642/tarpaulin/issues/517#issuecomment-923429731

I don't know if this helps, but hope it does.