synthetos/g2 #407: Compatibility with g2coregui

unquenchable53, 2019-04-02:


breiler, 2019-04-03:
I would like to comment on your third question. But first a disclaimer, I'm one of the developers of Universal Gcode Sender so I'm a bit biased. 😃 

There seems to be a trend in having automated workflows, we've got several questions about this the last couple of weeks. So I took a stab at it in UGS, developing a Command Line Interface (CLI) that would make it possible to send a .gcode-file through a script. https://github.com/winder/Universal-G-Code-Sender/issues/1171

There is also a brand new web based API waiting on a code review which would allow you to automate such a process with a bit of coding/scripting skills (could be done using curl-commands). https://github.com/winder/Universal-G-Code-Sender/issues/1192

Both features are still very new and there is still a lot of documentation and test work to be done. But if you have any questions regarding these I would gladly be of assistance.

There is a software comparison list if you need some help searching for other options. For instance, have a look at CNC.js which have a ton of features and looks pretty easy to extend.
https://github.com/winder/Universal-G-Code-Sender/wiki/Software-comparison

unquenchable53, 2019-04-06:
@breiler That would be an amazing feature to have. We plan to use image recognition algorithms to make the microscope stage move on its own based on what it sees. We need a closed feedback loop (Gcode - UGS - Motors - Image Capture - New Gcode - UGS - Motors) and having a CLI on our Gcode sender would be necessary. 

1) Can UGS have a feature which constantly (every ~0.05s) "sends" whatever gets written on a notepad or .gcode file? This way, we can use a simple copy command from CMD to write any gcode statements onto the .gcode file. 
2) Is there a beta version of the CLI that I could try out? 

Thanks!

breiler, 2019-04-07:
Sounds like an interesting project 😄 

There is no plan on having such a feature that will send gcode files from a directory like in g2coregui. 
You could however run the the CLI with something like this when your file has been created (for more options see this page https://github.com/breiler/Universal-G-Code-Sender/tree/feature/cli-pendant/ugs-cli):
`./start-cli.sh -c TinyG -d ttyUSB0 -f filename.gcode`

You could also start the daemon and communicate with the server using the API:
`./start-cli.sh -d -c TinyG -d ttyUSB0`

Then you could send POST-commands to execute Gcode using the API:
`curl -v -XPOST -H "Content-Type: application/json" --data '{"commands":"G0 X1"}' http://localhost:8080/api/v1/machine/sendGcode`

I'm waiting for code reviews for it to be merged into the master branch. But I have my development branch here where I'm still working on these features. Download the UniversalGcodeSender.zip which contains the UGS Classic edition): https://github.com/breiler/Universal-G-Code-Sender/releases/tag/v.2.3-cm

Remember that this is still under development.

unquenchable53, 2019-04-11:
Here's what I did. My computer runs Windows 10 so I installed Cygwin, opened the terminal, navigated to the newly downloaded UniversalGcodeSender folder (downloaded UniversalGcodeSender.zip from https://github.com/breiler/Universal-G-Code-Sender/releases/tag/v.2.3-cm), opened UniversalGcodeSender and opened a connection (COM4, GRBL, Baud: 115200) and ran this command on Cygwin- 
`$ bash start-cli.sh -c GRBL -f newfile.gcode`  
and nothing happened. The terminal just displayed the $ prompt for new line. 
newfile.gcode is in the same folder as the newly downloaded UGS folder. 

I also tried 
`$ ./start-cli.sh -c GRBL -d ttyUSB0 -f newfile.gcode` and 
`$ # java -jar UniversalGcodeSender.jar --controller GRBL --port /dev/ttyUSB0 --baud 115200 --print-progressbar` 
and had the same result. I tried all these commands after closing the connection from UniversalGcodeSender and had the same results. 

What am I doing wrong? Do I need some other prerequisite software? I'm new to all of this so I'm pretty sure I'm making a rudimentary blunder.

breiler, 2019-04-11:
Sorry I made a type with the port-select parameter. It shouldn't be `-d <portname>`, instead `-p <portname>`

I don't think that you can use /dev/ttyUSB0 as a port on Windows/Cygwin.
Close everything and try this to see if it can detect your COM-ports:
`./start-cli.sh -l`

Make sure you you have the file **newfile.gcode** in the same directory and run this. It should connect to COM4, send the file and then exit: 
`./start-cli.sh -c GRBL -p COM4 -b 115200 -pp -f newfile.gcode`

You could also skip Cygwin, but as there isn't a bat-file yet, so you need to start the java program manually:
`java -Xmx256m -cp UniversalGcodeSender.jar com.willwinder.ugs.cli.TerminalClient -c GRBL -p COM4 -b 115200 -pp -f newfile.gcode`

unquenchable53, 2019-04-11:
@breiler 
`./start-cli.sh -1` and 
`./start-cli.sh -c GRBL -p COM4 -b 115200 -pp -f newfile.gcode' 
show no results. Only the $prompt for next command.

`java -Xmx256m -cp UniversalGcodeSender.jar com.willwinder.ugs.cli.TerminalClient -c GRBL -p COM4 -b 115200 -pp -f newfile.gcode' 
works like a charm. The sound of the motors whirring was sweet music to my ears. Thanks so much for your help.

1) When will this feature be committed to Master?
2) Will I have to make any changes to the way I send the gcode file after the commit and downloading latest UGS? i.e. Will I have to make any changes to the 
`java -Xmx256m -cp UniversalGcodeSender.jar com.willwinder.ugs.cli.TerminalClient -c GRBL -p COM4 -b 115200 -pp -f newfile.gcode' 
after downloading the latest version? 

Thanks a lot, you've been so helpful.

breiler, 2019-04-11:
I figure I'll have my branch merged with master in a week or two. But I don't know when we'll have a new build. 

I'm not planning to change the command line arguments.

closed 2019-04-12 by unquenchable53