synthetos/g2 #411: Active 5 and 6 axis

qiwalker, 2019-04-13:
Hi....
I already flash g2core to my due
But only detect 4 axis.. How to active axis 5 and 6 

Thanks
Rizki

MitchBradley, 2019-04-13:
Look at this commit from my fork https://github.com/MitchBradley/g2/commit/b93e22160b00d048da1fef2982b24036e338f38e

qiwalker, 2019-04-14:
Hi mr.mitch

Great work.... 
Can u send me .bin file
And one more question and request
How to enable corexy ?

Thanks
Rizki

MitchBradley, 2019-04-14:
My .bin file would not work for you because my motor configuration is different than yours.  g2core must be configured before compilation, depending on your drivers and motors. My Due has a gShield, but the pinout is different than normal because I have external drivers for some axes instead of using the X and Y drivers on the gShield.  And I have limit switches on unusual pins, and other differences from  the 'stock' gShield pinout file.
What do you mean by "corexy"?

MitchBradley, 2019-04-14:
Instructions for compiling in the cloud:

Create an account on https://cloud.google.com . If you follow the steps below, you should remain in the "free tier" and not incur any monetary charges.

When you get to the dashboard/console page, click on the "Activate Cloud Shell" button at the top, as shown.
[image]

In the shell window, clone the git repository and compile.  Here is the sequence that I used, but you should use your fork instead of mine.

```
git clone --recursive https://github.com/MitchBradley/g2
cd g2/g2core
make BOARD=gShield SETTINGS_FILE=settings_othermill.h
```
Here are some changes you will need to make

- Clone from your repo instead of mine
- Patch your repo with the commit that I referenced above https://github.com/MitchBradley/g2/commit/b93e22160b00d048da1fef2982b24036e338f38e
- Also apply this patch https://github.com/MitchBradley/g2/commit/1b74b1ceaf94d2fa8adaaff3b53cf69af5df705d which fixes this problem https://github.com/synthetos/g2/issues/367
- Create a settings file for your machine and use it instead of the one I listed above

Then you can use "Download file" from the ... menu to get the binary file.  The download dialog requires a full path name to the binary file.  Get the prefix by typing "pwd".  In my case the prefix was "/home/wombatpie/g2/g2core".  The suffix is derived by replacing ".elf" with ".bin" in the "filename" that was displayed at the end of the "make" step.  In my case, the filename was "bin/gShield/g2core.elf", so the full pathname to download was "/home/wombatpie/g2/g2core/bin/gShield/g2core.bin"

qiwalker, 2019-04-19:
Hi mr.mitch

Core xy is invers kinematic, type of belt drive... 
Im still confuse with g2core firmware 😭😭

Anyway thank you so much for your guidance

Now im try to move from grbl to g2core firmware..

Thanks
Rizki

qiwalker, 2019-04-20:
Hi mr mitch

Can you show me some configuration for g2core before i compailed the file

Use nema 23
Gt2 timing belt 30 teeth

Thanks 
Rizki

MitchBradley, 2019-04-20:
It seems that you are trying to build a complex system without understanding how it works.  In order to create a g2core configuration, you need to understand a lot of different things, including, but not limited to:

- What kind of machine - mill, lathe, 3D printer, plotter, router, laser engraver, ...
- What kind of driver board is attached to the microprocessor
- Which microprocessor pins are connected to which functions on the driver board
- Which channels on the driver board are connected to which motors
- The polarity of the motors
- The strength of the motors (just saying Nema 23 doesn't help - Nema 23 motors come in different strengths)
- The coupling between the motors and the mechanical system (Gt2 timing belt 30 teeth is part of that, but not all)
- The allowable accelerations and speeds for the mechanical system
- The configuration of limit and homing switches, if any
- The way the spindle (or laser, or extruder, or pen, or knife, or whatever) is controlled

The g2core source code contains example settings files for several different types of machines.  I have no way of knowing which one of them might apply to your system. It as is if you had asked me how to operate a machine with a 1 kilowatt motor, but I don't know if your machine is a lawn mower, or a scooter, or a washing machine, or a bicycle, or an airplane, or a saw.  And the fact that you don't seem to know which information to supply makes me think that you need to do a lot more study before taking the next step.

qiwalker, 2019-04-21:
Hi mr. Mitch

[image]

I want to build screw fastening machine like above picture sir
For the motor and machine

1. Nema 23 3,2A
2. Gt2 30 teeth pitch 2mm
3. Arduino Due

Thanks
Rizki

MitchBradley, 2019-04-21:
How will you drive the motors?  You can't connect them directly to the Due.  You need some sort of driver board.  You also need some way to control the screwdriver precisely.

qiwalker, 2019-04-22:
for driver i use smart stepper with magnetic encoder
for screw driver i use atlas copco
the intelegent screw driver will combine with festo cylinder (down mechanism) + z axis

can i configure the firmware setting after flashing to the board ?
will the configuration save as i know due dont have EPROM ?

thanks
Rizki

qiwalker, 2019-04-22:
for cylinderl i will use gcode like M3, M4, M7, M8 for the trigger

MitchBradley, 2019-04-22:
You must configure at compile time.

Go into g2core/board/ArduinoDue/, make a copy of gShield-pinout.h under a different name, and edit it to set the pins that connect to your steppers.  The lines you need to change have names like kSocket1_StepPinNumber, kSocket_DirPinNumber, etc.  kSocket1 is for the first motor, kSocket2 is for the second motor, etc. You can set the  Microstep PinNumber entries to -1 under the assumption that your smart steppers have some external setting for their microstepping configuration.

If you want to use spindle control gcodes to turn the screwdriver on and off, set kSpindle_EnablePinNumber and kSpindle_DirPinNumber.

Set the kInput _PinNumber to the pins you will use for limit switches or other inputs to the controller.

Set kOutput _PinNumber to any other pins you will you for other outputs.

Then go into g2core/settings and make a copy of settings_othermill.h under a different name.  Edit it to configure all aspects of your machine.   For example, the setting M1_STEP_ANGLE applies to motor 1, which corresponds to kSocket1 in the pinout file.  For a 200 steps/rev stepper, its value is 1.8 because 360 degrees / 200 is 1.8 degrees/step.  M1_TRAVEL_PER_REV is the number of milllimeters the associated axis moves when the motor turns one rev. M1_MOTOR_MAP tells which axis that motor controls.

The settings like X_VELOCITY_MAX control how fast that axis is allowed to go.  VELOCITY is the speed for rapid moves (G0) while FEEDRATE is the maximum speed for G1 moves (but you can make it go slower with the F word in the GCode).  The TRAVEL settings tell how far the axis is allowed to move; set them to the actual size of your machine.  The JERK setting control how quickly the machine will accelerate.  If you have smart steppers the JERK probably doesn't matter because the smart stepper will figure it out.

The settings like DI1_MODE let you configure inputs like limit switches or probes.

Finally, edit g2core/boards.mk and add a section for your boards.  Make a new CONFIG name for your machine (for example Screwdriver), set BOARD to the name of the pinout file you created in the first step, and set SETTINGS_FILE to your settings file.

Then you can compile with

make CONFIG=Screwdriver

qiwalker, 2019-04-22:
thanks you very much mr. Mitch, il will follow your guidance
for VELOCITY how to calculate it Sir ?

Thanks 
Rizki

MitchBradley, 2019-04-22:
Experiment with different values to see how fast the machine will go without stalling the motors or causing mechanical instability, then reduce the number to give a safety factor.
The number will probably be different for different axes.

MitchBradley, 2019-04-22:
You can experiment with different velocity settings without recompiling.  In a console window, you can type, for example:

```
$xvm=1000
G0 X100
G0 X0
```
That will set the max velocity for the X axis to 1000 mm/min, then go at the rapid rate to X=100, then go back to X=0.  Then when you have found a good value, put it in the settings file and recompile to make it permanent.

qiwalker, 2019-04-23:
Ok mr.Mitch

thankyou very much for your guidance and advice
will update my work to you

thanks
Rizki

qiwalker, 2019-05-08:
Dear Mr. Mitch

i change my idea of robot by build scara robot
below picture for my scara
kindly please help me again :D
should i change x, y to rotary for move my scara and let z axis like usual
and i very thanks for your help before now i can compile g2core.. :)

[image]

thanks
Rizki

MitchBradley, 2019-05-08:
I'm sorry, I have no experience with non-cartesian setups so I can't help.  Perhaps you should use firmware that already supports SCARA.  A web search showed a few possibilities.

qiwalker, 2019-05-13:
it's ok mr.mitch

for cartesian unit for screw fastening on progress,

it would be great if i can make screw fastening robot and soldering robot in one machine
but for soldering robot need feeder for solder wire, 
can i add extruder setting (use for feeding solder wire) ?
now i already use 5 axis (3 Axis_standar, 2 Axis_radius ) and want to use axis number 6 for extruder is it possible sir ? 
if yes, kindly please help me to get the setting

thank you very much
rizki

MitchBradley, 2019-05-13:
The instructions that I posted three weeks ago, telling how to edit the configuration files to set pin numbers and motor speeds for different axes, apply to axis 6 the same as any other axis.

qiwalker, 2019-05-17:
Dear Mr. Mitch

Noted sir
can i check pin use using in Due using command (json)

thanks
Rizki

MitchBradley, 2019-05-17:
https://github.com/synthetos/g2/wiki/Digital-IO  I found that by searching for "g2core GPIO"

qiwalker, 2020-04-30:
Dear Mr. Mitch

how are u Sir ?

MitchBradley, 2020-04-30:
Do you have a question?

qiwalker, 2020-04-30:
can i use eeprom with g2core ?

MitchBradley, 2020-04-30:
> can i use eeprom with g2core ?

Due does not have EEPROM. Synthetos is working on some new boards that have EEPROM but they are not for sale yet.

qiwalker, 2020-04-30:
ouh oke thanks for ur answer, 
how to calculate step per mm ?

MitchBradley, 2020-04-30:
Look it up on the internet. Here is one reference that shows the math http://diymachining.com/grbl-settings-101-a-how-to-guide/

qiwalker, 2020-05-01:
thanks mr. Mitch
btw i already flashing gcore to arduino DUE and connect to use CNCJS but jog is disable :(

[image]

MitchBradley, 2020-05-02:
I don't think it is succeeding in connecting to g2core because there is no response to those commands in the Console window.  Maybe you are using the wrong COM port, or the wrong usb port on the DUE.

qiwalker, 2020-05-02:
i try to connect using arduino IDE
here is the output

[image]

justinclift, 2020-05-02:
Interesting.  The COM6 option seems correct from the Arduino IDE, but the output showing in the console of your CNCjs screenshot shows it wasn't working.

As a concept, Windows only lets one application access a given COM port at once.  Is there any chance some other applications (Arduino IDE or anything) was using COM6 already, when you tried it with CNCjs?

If you're not sure and want to test again, then try rebooting your computer and using CNCjs right away before anything else.  If that works, that might help figure out a real solution. :smile:

qiwalker, 2020-05-02:
already follow your instruction still cannot connect using CNCjs

but i try using g2core-api its working.

any idea ?

justinclift, 2020-05-02:
Hmmm.  This is being an interesting challenge.

With CNCjs, try uninstalling the one you have, and using this exact download instead:

&nbsp; &nbsp; https://github.com/cncjs/cncjs/releases/download/v1.9.22/cncjs-app-1.9.22-win-x64.exe

Saying that just because "something" seems to be not happy with your current CNCjs install.  Hopefully this new one fixes it. :smile:

If that doesn't work, then I'd probably try Universal Gcode Sender instead, and see if that works:

&nbsp; &nbsp; https://winder.github.io/ugs_website/download/

In the list of downloads there, you'd want the "2.0 Platform" one.  Either the "Nightly build" or the "Beta" one.  Either should work. :smile:

qiwalker, 2020-05-02:
Universal GCode Sender work perfecly
thankyou very much for your suggestion

for now im still curious how to use g2core-api, im very confuse because my first time to deal with API ahahhaha

justinclift, 2020-05-02:
Awesome, glad you got it working. :smile:

With g2core-api... I'm not personally sure as it's not something I've really touched. :wink:

qiwalker, 2020-05-04:
btw, is there any setting corexy in g2core ?

justinclift, 2020-05-04:
I **think**, but am not 100% sure, there's is initial CoreXY support in the `edge-preview` branch:

* https://github.com/synthetos/g2/tree/edge-preview

It's probably using the same config options as the example from the older 168-gquinitic branch here:

* https://github.com/synthetos/g2/blob/dev-168-gquintic/g2core/settings/settings_axidraw_v3.h#L92

qiwalker, 2020-05-05:
thank you so much @justinclift it running perfectly.
for g2core GUI, is there someone build it using C# or VB ?

justinclift, 2020-05-05:
Not as far as I know.  The only one's I've seen so far are in javascript (CNCjs) or java (UGS, [Goko](https://github.com/cncgoko/Goko)).

qiwalker, 2020-05-13:
i found liteplacer but its for pnp machine

ril3y, 2020-05-13:
@rickcaddell wrote a program called "CHIP" in VB a few years ago for "TinyG". Might take a little work to get it working properly with g2.

https://github.com/rickcaddell/Chip

qiwalker, 2020-05-14:
hi @ril3y thank you for ur reply, i already look at this link but there's no source code :(

ril3y, 2020-05-14:
I have the source code.  I can clone his repo and post it.  I know rick did not care if it was shared.

ril3y, 2020-05-14:
https://github.com/ril3y/Chip

justinclift, 2020-05-14:
Oh, that's cool. :smile:

I had the same thought as @qiwalker when you posted the initial link.  "Just an exe, no source. :frowning:"

Symeth, 2020-10-26:
hi @justinclift 

is there any stable release for corexy machine ?

justinclift, 2020-10-26:
@Symeth Haven't personally taken a look at the CoreXY stuff, but it still seems to be only in the `edge-preview` branch.

So, code is there but not well tested.  aka Definitely not yet a **stable** release. :shrug:

closed 2023-02-16 by ril3y