synthetos/g2 #412: Help getting started with a Due

SketchThis, 2019-04-19:
Hey all, I'm graduating myself from GRBL and looking to build a 5 axis system.

I've built several control systems before, GRBL included, but this is my first G2 system. I was drawn to it after using (and owning) a Handibot that's got G2 on it.

My setup right now is a bench test where I've got 5 stepper drivers, 5 steppers, some limit switches and a relay board. I've also got a Due, and will be using CNCjs on a PC/Raspberry Pi.

Here's where I'm stuck: I flashed the Due with this: https://github.com/synthetos/g2/releases/download/101.03/g2core-TestV9-g2v9k-101.03.bin

I used Chilipepprs online hardware fiddle to do it (because it was so easy) I've got the Due connected to CNCjs and they're talking just fine.

I used the pinout diagram I found here: https://github.com/synthetos/g2/wiki/Arduino-DUE-Pinout-for-g2core

When I wire in any of the steppers I'm not getting any movement of any kind.

I suspect that it's because I flashed something that wasn't configured.

I've got the repository cloned, and I have X Code installed and I'm looking through the files to see if I can figure out what the stock setups/pin assignments are, and I can't seem to figure it out. I've also looked all over this github, and I just can't find the information I need to compile a version for my needs.

I'm either missing something obvious, or because I'm not actually a developer it's just not making sense to me.

Could someone head me in the right direction?

karoria, 2019-04-20:
1. By default, g2 keeps motors disabled. So you need to enable them (see for motor_disabled and change it to motor_always_powered) in settings_default.h and set other machine specific parameters in this file before compiling.
2. Compile (I suggest master branch as you are first timer. You also need to download motate directory separately and paste contents in the empty motate folder of g2).
3. Flash newly generated bin to due. That's it!
Please remember that unlike grbl you cannot set machine parameters after flashing the firmware. You have to do it before compiling in the above mentioned file. That's why you almost cannot use a readymade firmware like you did.

Hope I am clear.

SketchThis, 2019-04-20:
Ok, that sounds reasonably simple. I didn't know that you couldn't configure afterwards like you do on GRBL, or that the motors are disabled by default... I get why, but that would have taken me a while to figure out!

I'm assuming the compiling errors I was getting in the screenshot were because I didn't have the motate directory downloaded? (I thought I did as I see some of the motate settings in there...)

karoria, 2019-04-20:
Most probably the compile errors may be due to empty motate directory. Remember to use gshield option for compiling. Optionally, you can also use command line for compiling in mac. Rest, you are on right path. All the best...

SketchThis, 2019-04-23:
Ok, I figured out the empty Motate directory, running the X Code file that was in the G2 directory, not the main directory of G2. I made a little progress, but I'm still really stuck on this compiling thing... I tried to compile a stock Gshield version from the master branch just to see if I could get it to compile and this is what happened:

https://drive.google.com/file/d/1zYeSXztFbKogHjmHGWCbOCyDb_mphR-n/view?usp=sharing

Hopefully that video explains much better what I'm failing at. I fully realize that the problem is very likely me, I just don't know where I'm going wrong... Is X Code even the right utility to do this?

SketchThis, 2019-04-24:
Ok, bit of an update, I got it to compile, I'm not sure how but I did!

[image]

I'm assuming this is the setting(s) I'm looking for.

I see that there are settings for different machine types, ShopBot, Ultimaker, etc... If I change the default settings, how do I know that when I do a build it's going to pull those default settings? Is there another "settings for my settings" somewhere?

Also, while I do have a pinout diagram, it does note that the pins are changing a lot. The latest diagram is a lot older than this build. I feel like it's a good idea to check which pins are assigned when I build.. I looked through and found a "Motate Pins" file.. Is that what assigns the pins? I couldn't really tell from the file...

karoria, 2019-04-24:
You have almost done it. Have you flashed it to due?
1. Yes, you can define the settings file while compiling. I am not experienced user of xcode. So cant exactly tell you where to define it. I always use settings_default.h to be safer as I am not a programmer.
2. The only pinout diagram availabe for g2core still stays true for most of the default themes. So don't worry. But yes, there are instructions available if you want to change it. I suggest to use default pinout now, till you become an experienced user of g2. This is because there is still lot of development and changes going on and we may have additional issues with pinout changes. I am not sure but I suspect.
3. Now, as you succeeded compiling it, the only experimentation you need is changing machine parameters and see which suits best to you.

karoria, 2019-04-24:
Sorry, forgot to mention that the settings you are in as per screenshot is right.

SketchThis, 2019-05-01:
I haven't flashed it just yet, but I'm going to take a crack at it this afternoon. I'll post the results!

SketchThis, 2019-05-05:
Ok, so I'm back at this again and ran into this:

In that settings.default..h file I changed MOTOR_DISABLED to MOTOR_ENABLED and I get this error when I try to build:

[image]

Is there a good tutorial out there for someone like me that would let me more easily get this up and running? I feel like my lack of real coding experience is making this really difficult :)

SketchThis, 2019-05-05:
Ok, I'm getting a little closer, I was able to alter the settings.config.h file and enable the X motor, and when I compiled it worked. I also used the command line tool to flash to the DUE, so that's working as well. I'm using CNCjs to connect to the DUE to control it, when I connect, that also works (I can do the ?? command to see all of the settings, etc so I know it's working)

However, I'm still getting no motor movement. On top of that, when I issue move commands, (X1 X2, etc) the DRO isn't changing.

So for some reason it's not taking the move commands.

I see in these settings files there are a bunch for different machines. How do I know that when I compile it's using the settings file I altered? Is there a way to check/change that? I'm thinking that this might be my problem...

BTW I'm using X Code to compile...

karoria, 2019-05-06:
You figured it right. compile time file selection is as per "makefile". For time being, please don't experiment with other settings. "Motors not moving" is the problem of settings file only as I feel. Your compiling and flashing process doesn't have any issues what I can understand from your description. Kindly share your settings_default.h so I can see if there can be any problem with it.

SketchThis, 2019-05-06:
There's my settings file.

First I started with ONLY changing the motor_disabled value, and that didn't give me anything. Then I added some random numbers to microsteps/steps per unit just to see if I could get some reaction.

Still nothing.

I have a feeling like this is a configuration problem and not a wiring problem because I'm not seeing any movement on any axis in the DRO. Once I see that, hopefully I'll get some physical motor movement, and if I don't at least seeing it in the DRO will convince me that the board is likely putting out steps...

[settings_default.h.zip](https://github.com/synthetos/g2/files/3147913/settings_default.h.zip)

karoria, 2019-05-06:
OK. Got it. Please replace AXIS_DISABLED with AXIS_STANDARD for x, y and z axis and use AXIS_RADIUS for a, b and c axis if you want to use any of them. Rest seems ok. Compile and flash with this changes done and share the results..Hopefully you will get the axis moving.
I suggest to enable all 3 axis (motors and axis both) in one go and then compile and flash. Best of luck.!!!

SketchThis, 2019-05-08:
Ok that worked! I can now connect to the board and get "movement" in the DRO!

THANKS!!

However I'm still not getting physical motor movement. I know my bench setup works because I can use other controllers with it. Is there a way I can verify that the pins I'm using are accurate? I'm following the posted pinout PDF, but that PDF says that there are likely pin changes. Can I see in the firmware build where the pin assignments might be so I can make sure I'm using the right ones?

rkoe, 2019-05-08:
I'm successfully using g2 with a Due on a milling-machine for quite some time now. I could probably check and send you some information in a few days.
(But note that I'm using my own G-code-sender, since none of the GUIs I've tested worked well with g2. But this may have changed since then.)

Regards,
Roland

SketchThis, 2019-05-08:
> I'm successfully using g2 with a Due on a milling-machine for quite some time now. I could probably check and send you some information in a few days. (But note that I'm using my own G-code-sender, [...]

That would be awesome. Mainly what I'd be interested in knowing is how you have the axes wired (specifically what pins you're using). 

Right now I can't tell if I'm using the wrong pins, or if I'm not getting any outputs at the pins. I suppose what I really need is a scope to check the pins on the board, but barring that making sure I've got everything setup properly seems like a prudent first step.

I'm using some basic open loop stepper drivers (cheap Chinese ones I got on Amazon for bench testing). For Step+ and Dir+ I've got them connected to pins 2 and 5 respectively. (I used this: https://github.com/synthetos/g2/wiki/Arduino-DUE-Pinout-for-g2core as a guide)

karoria, 2019-05-09:
You are using right pins for x axis. So no worry. Still you can check the pulse with multimeter as you can be able to see the meter display some random numbers when the pulses actually being sent.

mhlong10, 2019-05-09:
One thing to double check is the ENABLE_POLARITY.  Depending on how your hardware looks it might be backwards.  I went thru a similar situation, scoped out the signals and then as it turns out it was simply that my enable was backwards.

SketchThis, 2019-05-09:
Does that matter if I'm not using the enable pins on my stepper drivers
though? Right now the way they're setup on the bench is that I don't need
to hook up the Enable positive or Negative at all.

mhlong10, 2019-05-09:
I didn't realize you are not using the enable pins.  It shouldn't matter in that case.

SketchThis, 2019-05-09:
Ok, I was worried that beyond the actual enable pins that the software looked at that setting... I'll continue to poke around at the hardware to see if I can work this out...

ril3y, 2019-05-09:
High jacking thread.

Roland would you share your GUI sender?  Also CNCjs works with G2.  Thanks
for helping us out on the issues.

Riley

turboproc, 2019-05-11:
You should be able to make this work in the end. Running here a WorkBee with g2core on a DUE and LaserWeb4 as the gui. The DUE has been complimented with a rather trivial shield to allow easy connection to stepper drivers and end switches.

Below my config file. It's a bit long and could be cleaned up, nvertheless it;s the version that compiles without complains makes my machine move.

Regarding Laserweb4 I had to make some changes as this didn't work out of the box. Let me know in case you want the details.

`/*
 * settings_default.h - default machine profile - Set for Shapeoko2
 * This file is part of the g2core project
 *
 * Copyright (c) 2012 - 2018 Alden S. Hart, Jr.
 *
 * This file ("the software") is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License, version 2 as published by the
 * Free Software Foundation. You should have received a copy of the GNU General Public
 * License, version 2 along with the software.  If not, see <http://www.gnu.org/licenses/>.
 *
 * As a special exception, you may use this file as part of a software library without
 * restriction. Specifically, if other files instantiate templates or use macros or
 * inline functions from this file, or you compile this file and link it with  other
 * files to produce an executable, this file does not by itself cause the resulting
 * executable to be covered by the GNU General Public License. This exception does not
 * however invalidate any other reasons why the executable file might be covered by the
 * GNU General Public License.
 *
 * THE SOFTWARE IS DISTRIBUTED IN THE HOPE THAT IT WILL BE USEFUL, BUT WITHOUT ANY
 * WARRANTY OF ANY KIND, EXPRESS OR IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES
 * OF MERCHANTABILITY, FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT
 * SHALL THE AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
 * LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM, OUT OF
 * OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE SOFTWARE.
 */
/* Note: The values in this file are the default settings that are loaded
 *       into a virgin EEPROM, and can be changed using the config commands.
 *       After initial load the EEPROM values (or changed values) are used.
 *
 *       System and hardware settings that you shouldn't need to change
 *       are in hardware.h  Application settings that also shouldn't need
 *       to be changed are in g2core.h
 */
/*
 * This file is included to define settings that were not present in the
 * SETTINGS_FILE, or if SETTINGS_FILE was omitted.
 *
 * It does the following for each section
 *
 *  - Define rational machine defaults and Gcode power-on defaults
 *  - Define rational communications and reporting settings
 *  - Disable all motors. Motors used must be enabled in other settings file
 *  - Disable all axes. Axes used must be enabled in other settings file
 */
/***********************************************************************/
/**** Default profile for screw driven machines ************************/
/***********************************************************************/

// ***> NOTE: The init message must be a single line with no CRs or LFs
#ifndef INIT_MESSAGE
#define INIT_MESSAGE "Initializing configs to WorkBee 50x75 settings"
#endif

//*****************************************************************************
//**** GLOBAL + GENERAL SETTINGS **********************************************
//*****************************************************************************

// *** Machine configuration settings

#ifndef JUNCTION_INTEGRATION_TIME
#define JUNCTION_INTEGRATION_TIME   0.75    // {jt: cornering - between 0.05 and 2.00 (max)
#endif

[... 2159 lines cut ...]
#define USER_DATA_C0 0
#endif
#ifndef USER_DATA_C1
#define USER_DATA_C1 0
#endif
#ifndef USER_DATA_C2
#define USER_DATA_C2 0
#endif
#ifndef USER_DATA_C3
#define USER_DATA_C3 0
#endif

#ifndef USER_DATA_D0
#define USER_DATA_D0 0
#endif
#ifndef USER_DATA_D1
#define USER_DATA_D1 0
#endif
#ifndef USER_DATA_D2
#define USER_DATA_D2 0
#endif
#ifndef USER_DATA_D3
#define USER_DATA_D3 0
#endif
`

StanislavPr, 2020-03-01:
Not sure where you're at in this, but you should change the power level to your motors. 
#define M1_POWER_LEVEL 0.0 // {1pl: 0.0=no power, 1.0=max power
Change that too M1_POWER_LEVEL 1.0 and all other motors of course M2 etc...
That should give you motor movement.

closed 2023-02-16 by ril3y