synthetos/g2 #417: Feedrate Override Solution

karoria, 2019-07-04:
@aldenhart @giseburt @MitchBradley @justinclift 
Hi all wonderful developers...
I am building a VMC machine for which I am using g2core with latest master branch (I am flexible to use edge also). All the other functions of machine works great except "FEEDRATE OVERRIDE" which is badly needed for this machine. I have gone through the documentation in which it is mentioned that this feature is experimentally implemented but somehow I am not able to do it. I am proposing several cases as under and seek your expert guidance for which suits me the best:

CASE 1: USB DUAL ENDPOINT MODE WITH ONLY NATIVE USB CONNECTED
I am unable to get the second port connected with communication. It just connects but doesn't accept any communication like jog or other commands. If I use software for it, the buttons stays disabled even after connecting. So, I am getting only the first port connected with communication.
Am I missing anything here? or I have to alter my settings file or something else before flashing the firmware? If somebody has done this successfully, would you mind sharing your firmware to me?

CASE 2: USB DUAL ENDPOINT MODE WITH NATIVE AND PROGRAMMING PORT CONNECTED WITH 2 DIFFERENT USB CABLES
Same result as of Case 1

CASE 3: USB SINGLE ENDPOINT MODE
Same as above. In all these cases I am able to override the feeds while jogging only (not realtime). While running any program, it doesn't accept the overrides. I have noticed that when I issue feedhold (!) the queue flush (%) and then say {mfo:0.5}, it accepts the override within program also but those program lines are lost while flushing the queue.
Any suggestion/guidance which is workable?

CASE 4: ADDING ANOTHER CONTROLLER BOARD FOR ONLY OVERRIDES
Sorry for this "WILD IDEA" but if there is possibility to input the axis pulses to another controller and simply "SCALE" those pulses and make it out from any pin of that second controller. I know, getting the machine state in that case must be a big hurdle.

I am desperate using g2core for my machine and that is why I am trying every possible solution. Waiting for expert advice from all of you.

Best regards,
Ravi Karoria

justinclift, 2019-07-04:
Hi Ravi,

The main g2 developers have been focused on other projects for a few months, not really putting time into g2. :frowning:

If you're up for learning C++ (?), that would give you the ability to debug and fix the problem yourself.  It's not a simple thing, but it is doable. :smile:

@cmcgrath5035 Any other ideas spring to mind?

karoria, 2019-07-04:
Noted @justinclift. Thanks for reply. 
@MitchBradley any suggestion in this regard?

cmcgrath5035, 2019-07-06:
Sorry Ravi, nothing I can add.

MitchBradley, 2019-07-06:
I am experimenting with adding real-time single character rate adjustment commands compatible with GRBL. I have run into the problem that mp_start_feed_override() causes the system to hang.  I have tracked the hang down to the line "mp->p = mp->c;" which sets mp->p to a null pointer since mp->c is never set anywhere.  Later dereferencing sends the code off into the weeds. 
I think the intent was to have some preceding code to find a "break point" at T+20ms in the planner queue, setting mp->c to the appropriate mpbuf struct, but that code was never written.  I'm doing a deep dive into the planner algorithm in hopes of figuring out how to do this, but it may be a long time underwater and I hope I can hold my breath that long.  If anybody who knows the planner seascape wants to weight in or leave some air tanks lying around, please do so.

MitchBradley, 2019-07-06:
If can get the single-character commands to work, perhaps I can then arrange for the JSON versions to queue-jump well enough to work.  I still don't understand all the timing implications of the various pipeline stages.

karoria, 2019-07-08:
Glad to know about it @MitchBradley. Would you please elaborate advantages of such single character commands over the ascii commands at present in grbl control system? This will give me little insight and direction to my project. As for grbl, i wish to make a knob which will send required ascii  for overrides.
Further, will you please clarify that have u got succeeded (or anybody you know) to get feedrate overrides for g2core as mentioned in the wiki. I cant get it right when a program is running as mentioned above.

karoria, 2019-07-08:
> Sorry Ravi, nothing I can add.

Thanks for reply.

karoria, 2019-07-08:
@MitchBradley 
Is it possible that instead of finding a way through busy planner queue, can we define a physical analog or digital pin to send a variable voltage or pwm to control overrides?
Like on a scale of 0 to 5V analog input, can we define 10% to 200% feedrate overrides? Obviously Uno board has limited resources, so we can think with mega.
Sorry if my thought seem too ambitious as I don't know programming/coding. Just sharing which came to my mind.

MitchBradley, 2019-07-08:
The pin idea doesn't really help. We already have a way to get the desired override factor into g2core in a timely fashion.  The problem is to get the override applied to the motion quickly enough to be useful.  For that to happen we need to interact with the planner queue.  My goal at this moment is to have the override apply to the next move after the one that is currently running the motors (unless that move is going to end very soon, in which case the move after that).  In an ideal world, the override would apply instantly, but that would require a lot of super-gnarly math to revise an acceleration plan starting in the middle. So I am focusing on revising the next acceleration plan.  There is some existing code for doing that but it is incomplete and doesn't quite work right, so I am trying to reconstruct the thought process of the developers to try and get it going.
Regarding the idea of "scaling the pulses" with an outboard controller: Imagine that you are in a car and you are trying to duplicate the path of a second car, only at a different speed.  The only information that you have comes from watching that other car, perhaps by tracking their GPS.  If you want to go faster, the only way to do it is to wait for them to arrive at the destination, all the while tracking their position and speed, then duplicate their path at a fixed multiple of their speed, but limited by factor like speed limits, the maximum speed of your car, while avoiding unsafe or impossible conditions such as going around corners too fast and going off the road.  And even then you would not arrive sooner than them.  Your trip might take less time but you have to wait to start.
If you want to go slower, it is somewhat easier.  You can start right away and follow them and the limiting cases are much easier, but you still have to track them with GPS or something, because they will soon go out of sight.  To reproduce their track accurately, you will need to record what they are doing when they get arbitrarily far ahead, so you will need a lot of storage.
The only real way to do this is for both cars to have the same route map in advance, then they can both follow it at whatever speed they like.  The route map in this case is the GCode program.  The current problem I am working on is analogous to a driver following a route with a passenger who sometimes says to go faster or slower.  If you are on a straight section of road, the driver can easily follow the instructions - except that doing so might exceed the speed limit or result in unsafe driving conditions relative to other cars - so the driver can't just blindly follow the instructions.  And if the driver is in the middle of a set of curves, where he carefully planned the speed and line to negotiate the turns within the traction limit of the tires, it might not be possible to change speed until the next segment of the route.
Note that I haven't invoked any knowledge of coding or programming in that analogy.  Machine tools are very much like cars - they have a fixed amount of power, mass which causes inertia, limits on speed and curvature before something slips, and drivers/controllers that have to figure out a way to approximate a specified route given those and other constraints.

karoria, 2019-07-09:
You are a best teacher @MitchBradley 
Got it completely. I can also understand the task you have dived in is probably the most difficult one in CNC controllers as it has to respond correctly to something "UNPLANNED".
I am waiting for the day you will come with the desired result with g2.
BTW, they have mentioned that feedrate override is already implemented and works well. Only the line in planner buffer will remain unaffected. While I am getting no results at all once the program is started. Is it the same with you?

MitchBradley, 2019-07-09:
Where did you hear that the override works well?  Perhaps it is in a different branch?  I would like to read what they said.

karoria, 2019-07-10:
Sorry, apologies. Somehow I got that way. Actually its well working is not mentioned in its only wiki. But they have mentioned that the moves which are in queue will be unaffected. But I don't get the result at all till the program ends. ;)