synthetos/g2 #418: Axes pairing?

SketchThis, 2019-07-23:
Im getting started with G2 core. I've got an Arduino Due setup on a bench driving a few external drivers and steppers.

The gantry machine I'm planning on installing this system on has dual X motors. This keeps the gantry running straight.

Is it possible to configure G2 so that two axes are paired together?

karoria, 2019-07-23:
Make following settings in settings_default.h or whatever file you are using for compiling:
Define motor 1 as Axis_X
Define motor 2 as Axis_X
Define motor 3 as Axis_Y
Define motor 4 as Axis_Z
and make subsequent changes according to your mechanics. Compile and Flash.. That's it...

SketchThis, 2019-07-23:
That's easy! Now the next question:

With both motors paired is there a way I could unlock them from each other temporarily?

On my current gantry machine I have dual X motors that are paired but I have a squaring routine that unlocks them, measures the distance between a sensor mounted on each side of the gantry and then "twists" the gantry into square... Is there some special g code I could use to address each motor at a time to accomplish this?

ril3y, 2019-07-25:
Yah this is possible.  Check out this.
https://github.com/synthetos/g2/wiki/Power-Management#md-motor-disable

Should work for you.

Riley

SketchThis, 2019-07-25:
If I'm reading that correctly, that should power off the motor which is actually what I don't want. If a motor is powered off that side of the gantry has a potential to freely move.

Here's what I'm trying to do:

When you first power up the machine you zero the axes. On X (where there's a motor on each side of the gantry) you have a proximity switch on both sides. The X motors go past the switches and the distance between those two switches is recorded, and compared against a stored value.

If the actual distance is different than the store value, both steppers should stay locked (IE powered up) but move commands are only sent to one stepper to increase or decrease the distance between the two sides of the gantry to match that stored value.

So to boil it all down: I want to be able to have both X steppers run in unison, unless I issue a special command to move one and not the other (while both remain powered up and locked)

ril3y, 2019-07-25:
So no this is not supported in software for dual homing as of now.  However, I think there could be a few hacks that you could do.  If your system is ok with crashing into one of the edges for a limited time (IE motor stalls but the machine is fine) then you could wire up your limit switches so there were in series meaning both have to be closed to send the signal to the board.  Its a bit hacky but I have thought about doing this on the smaller belt driven machines.  

@giseburt @aldenhart I updated this issue but any input on where / how we could do something in software were, we could use 2 inputs for individual motor homing for dual-motor axis setups would be appreciated.

SketchThis, 2019-07-25:
For reference this is how I solved it on a ShopBot controller that didn't support unlocking a paired motor:

https://youtu.be/RGEmztTTnIE

I was a little nervous about running a relay on the step wire for an axis, but it totally worked. I suppose I could do that here, but I really think a software solution is much more elegant.

On this machine I'm currently running a Centroid system that has software squaring built in. It's just one less thing to wire/go wrong.

I could also see this being used on larger 3D printers as well.

giseburt, 2019-07-26:
What Riley said about the motor disable function does in software what your relay system did for the ShopBot.

So say your two Y motors are on motors 2 and 3, and you normally use power mode 1 (always enabled, always on). (Power mode 2 would work too.) Then we do:

```gcode
M100 ({2pm:1})
M100 ({3pm:1})
G28.2 Y0
; measure difference here if you wish
; or if you want to home them to blocks individually
M100 ({3pm:0}) ; disable second Y
M28.2 Y0 ; home first Y
M100 ({3pm:1}) ; enable second Y again
M100 ({2pm:0}) ; disable first Y
M28.2 Y0 ; home second Y
M100 ({2pm:1}) ; enable first Y 
; all squared up!
```

If you wished you could also use probing but you would need to set your probe input and then manually adjust your coordinate system.

ril3y, 2019-07-29:
I think the only issue with this @giseburt would be that when you disable the second y after you homed it could move when you try to home the other y motor?

SketchThis, 2019-11-09:
You wouldn't want to cut power to the second motor so it stays locked...you would not want to send any step signals to it

UweSauter, 2019-11-24:
I'm also currently building a gantry setup, with two shafts for X and I'd like to see this functionality as well.

What do you think about this:
Instead of having two limit switches per direction (Xmin, Xmax, etc.) they could be couple to a motor/shaft. The logic then would be to home one axis until the first limit is reached and then put that motor to stop and move only the second motor until its limit is reached as well.

I'll try to create my own pinout for the Due as I don't need many of the functions that are connected to outputs on the original pinout. So in my case, I'd be free to assign a separate X2min pin.
But many machines only have one limit switch per axis and thus it might just be a configuration issue to repurpose Xmax as X2min.

UweSauter, 2019-11-24:
Seems like #359 is about the same functionalty…