synthetos/g2 #419: B Axis not moving

SketchThis, 2019-07-25:
Hey all, let me preface this with saying that I'm a bit of a rookie with G2. However, with the help of folks here I've got a bench test up and running. 

I've got a 5 axis build that I've flashed to a Due. I'm using the G Shield/default settings.h file to create this build (that's all I was able to get to work). I've altered the settings.h file to enable motors 1-5. I'm treating XYZ as linear (cartesian) and AB as rotary axes. Really the goal at this stage is just to get everything spinning.

XYZ and A work, but I can't get B to work. I've ruled out a bad stepper motors/drivers/wiring as I've tested the wires, swapped the steppers, and swapped the wiring. It's not a hardware issue.

It appears as though the Due isn't outputting anything to my drivers. I followed the pinout diagram here: https://github.com/synthetos/g2/wiki/Arduino-DUE-Pinout-for-g2core

My B Axis has step+ connected to pin 37 and direction+ to pin 38. Step/Direction- are grounded to the DUE.

I tried enabling the C axis and connecting something to that and I wasn't able to get any movement either. This all points to a mis-configuration in what I flahsed.

I've looked over my settings.h file several times to spot something wrong and I can't see anything. Can anyone help me figure this out? I've attached my settings.h file.

[settings_default.h.zip](https://github.com/synthetos/g2/files/3432855/settings_default.h.zip)
 ##
[image]

giseburt, 2019-07-26:
Hello,

So the Pinout is right, and you probably have it wired correctly. The problem is only 4 motors are enabled in the code. If you’re comfortable with modifying the code and recompiling you can make these changes:

At https://github.com/synthetos/g2/blob/edge/g2core/board/ArduinoDue/hardware.h change `#define MOTORS 4` from `4` to `5` which enables various configuration options (`5ms` etc.) all over the system.

Uncomment lines 71-78 of [`board_stepper.h`](https://github.com/synthetos/g2/blob/edge/g2core/board/ArduinoDue/board_stepper.h#L71-L78) and the corresponding lines 69-77 in [`board_strpper.cpp`](https://github.com/synthetos/g2/blob/edge/g2core/board/ArduinoDue/board_stepper.cpp#L69-L77) . That creates the `motor_5` object.

Now we need to add `motor_5` to the `Motors` array on line 89 of [`board_stepper.cpp`](https://github.com/synthetos/g2/blob/edge/g2core/board/ArduinoDue/board_stepper.cpp#L89) so that it looks like:

```c++
Stepper* Motors[MOTORS] = {&motor_1, &motor_2, &motor_3, &motor_4, &motor_5};
```

I believe that will do it. Your settings file may need to have motor 5 settings added, and the defaults that will be used when those are missing disable the motor (for safety reasons.)

giseburt, 2019-07-26:
Slightly unrelated, you hardware is looking good! I’d recommend in the final assembly to utilize both sets of screw terminals off your power supply to the power breakouts, so you have redundancy and double the cable handling the load.

IOW, your power supply has (at least) two sets of DC+ and DC- terminals, so have a cable from one DC+ to once terminal on the breakout, another cable from the other DC+ to a different terminal on the breakout, and so on for the DC-.

And don’t forget to ground the whole thing.

SketchThis, 2019-07-26:
That totally led me to the right answer! I now have all 5 axes working!!

Thanks for the comments on the setup! I used a bunch of old stuff that I had laying around to put that together. The point of this really is to just learn the software side of things. I'm likely going to transfer this setup over to a machine that's already got a power supply and drivers working on it.

justinclift, 2019-11-15:
@SketchThis How'd this end up going? :smile:

closed 2019-12-12 by ril3y

SketchThis, 2019-12-13:
It worked!!

justinclift, 2019-12-13:
Cool!

Any photo's of the final machine, and/or video's of it in operation? :smile: