synthetos/g2 #420: Can g2core run on Teensy 3.2

UnfinishedBusiness, 2019-07-29:
Can g2core run on Teensy 3.2? If so where could I find build instruction?

If not, what would it take to make it happen?

Thanks,
Travis

justinclift, 2019-08-04:
It may, though I don't remember anyone having done so (yet).  The Teensy seems to have a compatible CPU and enough ram, so "should be doable".

For building... good question.  Initial thought would be to start with the standard Arduino Due approach (eg https://github.com/synthetos/g2/wiki/Getting-Started-with-g2core-Development#compiling-the-code), and adjust the pinouts to work with the Teensy.

The reference pinouts for the Due: https://github.com/synthetos/g2/wiki/Arduino-DUE-Pinout-for-g2core

Hopefully that helps? :smile:

UnfinishedBusiness, 2019-11-19:
I somehow missed your comment 3 months or so ago... Anyways, I wasn't able to get g2core to run on the Teensy 3.2. I ended up writing my own motion control firmware that is highly cross-platform. It uses the Arduino wiring framework and platform IO. It will run on virtually any MCU with an interrupt timer that can run raster that 10uS periods. Faster the timer, higher the step train frequency capability... I was hoping to avoid doing all that work and just use g2core but it's up and running now on a dozen or so machines and ultimately I'm happy to have a motion control system I have full control over. Thanks, I appreciate your response!

justinclift, 2019-11-19:
Interesting.  Is the project OSS and online somewhere? :smile:

UnfinishedBusiness, 2019-11-19:
Sure, check it out.

The Firmware is here
https://github.com/UnfinishedBusiness/XmotionFirmware

The Machine Interface Software is here
https://github.com/UnfinishedBusiness/XDesign

It's all designed for CNC Plasma cutting and has built-in automatic torch height control. I'll be adding rotary axis capability soon, right after I get my inexpensive BLDC servo drivers wrapped up (Basically what ODrive is doing except I'm adding the driver, motor, and encoder into a nice tidy little unit that reads step and dir pulses, and has a feedback interface...)

justinclift, 2019-11-20:
Oh, that's cool.  I was thinking about CNC Plasma recently, but it's likely not a short term thing to start doing in my case. :wink:

With the JetCad thing - written in C++? - is that compiling to WebAssembly?

UnfinishedBusiness, 2019-11-20:
Thanks! My online JetCad software is all Javascript but is only 2D right now. It's great for creating 2D geometry for plasma or any other jet cutting. The offline JetCad3D is parametric 2D and 3D (https://github.com/UnfinishedBusiness/JetCAD3D) written in C++ and still cross-platform. I big goals of making JetCad3D replace Fusion360 in the shop (I don't like windows in my shop, will force an update while you're in the middle of a project time crunch and can't justify Macs financially lol) and be suited for complicated CAM Milling and Turning toolpaths. But Jet cutting toolpaths will be coming first and function with the same workflow as Sheetcam nesting wise, but more automated as far as tool management goes.

I have not messed around with WebAssembly at all

closed 2019-11-20 by UnfinishedBusiness

justinclift, 2019-11-20:
> I don't like windows in my shop, will force an update while you're in the middle of a project time crunch ...

Yeah.  A friend (professional translator doing contract work) was staying over a few weeks ago, and his boss asked him to get an urgent news piece translated.  When he booted up Win 10... it then proceeded to install some update for *6 hours*.  No way to get around it.  Beyond unimpressed.

> But Jet cutting toolpaths will be coming first and function with the same workflow as Sheetcam nesting wise, but more automated as far as tool management goes.

Sounds like it could be useful for lots of people.  With the "cross platform" bit, looking through the project Makefile there it has references to Win/Apple/Linux/BSD.  That sounds hopeful (Linux user here). :smile:

> I have not messed around with WebAssembly at all

Keep an eye out for it.  It's a widely adopted up-and-coming alternative execution engine to Javascript.

Instead of having to develop stuff in Javascript, Typescript, or similar (crap languages IMHO), you just use whichever (LLVM supported) language you're familiar with (Go in my case) and compile it to WebAssembly (Wasm).  Early wasm stuff was pretty slow, but due to the wide adoption and large amount of ongoing development in Firefox, Chrome, etc, it's becoming fairly decent now.  Debugging is still not really workable yet, but people are working on that specifically to bring it up to speed. :smile:

This might be useful: [WebAssembly Guide for C/C++ developers](https://webassembly.org/docs/c-and-c++/)