synthetos/g2 #423: How to Query Homing/Limit switch state?

BDFife, 2019-08-18:
Forgive me for asking this question here (I'm happy to use another channel if you prefer).

I'm trying to debug an issue with homing on my Printrbot CNC (v1) using a TinyG v9 board. My configuration file is here:

https://github.com/BDFife/MarlinConfigs/tree/master/PBCNCv1/settings

How do I query end stop state via TinyG syntax? I read through and searched the wiki but if the answer is there, I didn't find it. I'm looking for an equivalent to M119 in Marlin gcode. 

Thanks,
Brian

BDFife, 2019-08-18:
Aha: 

{in1:n}  Read digital input 1, Returns 0, 1, or null
{in2:n}
...
{inN:n}

{in:n}   Read a all digital inputs as a single JSON object

I will look into this, and my GPIO assignments.

closed 2019-08-18 by BDFife

justinclift, 2019-08-18:
No worries.  If you don't figure it out, you're welcome to reopen this/ask again (etc). :smile: