synthetos/g2 #424: Has this project stalled?

norru, 2019-09-01:
Hello,

I am at the point where I need to make a committment to a firmware for my CNC build with basically two candidates: g2core and some flavour of GRBL 1.1.

While g2core looked extremely promising, it does not seem to have been very active for the last few years.

There is clearly a lot of work in progress so the reason can't be stabilisation.

Have the original maintainers left or moved on to pastures greener? Unfortunately I have no spare resources to write large bits of functionality so "maintain it yourself" is kind of a dead end.

justinclift, 2019-09-02:
Development is still happening... just very, very slowly at the moment around people's other commitments. :frowning:

norru, 2019-09-02:
Is Synthetos not pursuing this "commercially" any longer (as it did with tinyG boards?). Or was it never supposed to be a business enterprise?

norru, 2019-09-02:
I wonder, because "on paper" this is a killer project. The consensus seems to be that it has great potential but it's not good for production quite yet due to bugs and missing fieatures such as configuration persistence which forces every user to build their own firmware, which is itself a very big barrier to entry.

karoria, 2019-09-02:
Hi norru.
I was in exactly same situation one month back and i chose grbl with esp32 for time being. I am a hardcore fan of g2core but feed overrides are still not available with g2 which is main requirement for my VMC project. Once g2 team cracks overrides, i will definitely switch back to g2.
Regards,
Ravi

yck011522, 2019-09-02:
@norru I would love to just add a +1 to the 

> configuration persistence 

I'm really hoping to see this come to daylight at some point.

justinclift, 2019-09-02:
Not being a member of Synthetos (more just helping out here), I'm not really sure of the long-ago background.

That being said, a prototype [gQuintic board](https://github.com/synthetos/g2/wiki/gQuintic-Specs) turned up in the post about a week ago.  Haven't yet powered it up, as it doesn't seem to accept power over USB (first thing I tried).  (It likely just needs me to spend some time looking for the right info).

So... there's "stuff" still happening. :wink:

norru, 2019-09-02:
Hardly "new":

>gQuintic Specs
>Alden Hart edited this page on 1 Oct 2018 · 25 revisions

norru, 2019-09-02:
@karoria 

> I was in exactly same situation one month back and i chose grbl with esp32 for time being

This actually tickles my curiosity. Care to share details about this project? I've got a few ESP-WROOMs idling in a drawer somewhere :)

justinclift, 2019-09-02:
> Hardly "new"

???

karoria, 2019-09-02:
@norru
I am using this source
https://github.com/bdring/Grbl_Esp32
I chose this because it runs on 32 bit controller ESP32 and you can get stable pulse train of approx 120kHz against 30kHz of uno board. And of course with best in class realtime overrides!
Next good thing is pin assignment is almost flexible.
The idea of this whole project is you can run this controller without a host pc because of wireless server capability of esp32.
That was enough to choose it for my project.

norru, 2019-09-02:
@karoria yep, I had just found it myself googling. Looks very intriguing! Are you building the board yourself?

karoria, 2019-09-02:
yes. But The developer Bart Dring has already developed a board with builtin stepper drives useful for small machines or 3d printers. That board is being sold online. However i am using industrial quality drives as my project is of VMC so i am building my own board as per my need.

manmountain, 2019-09-14:
@justinclift 
> That being said, a prototype gQuintic board  turned up in the post about a week ago

Where can non  Synthetos members get a board or the schematics for building one?

justinclift, 2019-09-20:
@manmountain Good question.  Email me directly (justin@postgresql.org) and lets have a discussion about intended use case, etc.  Not sure how many of the dev boards don't have an owner yet :wink:, but if there are unallocated ones we can probably work something out. :smile:

closed 2020-01-13 by ril3y