synthetos/g2 #426: power mode settings have no effect

arcadeperfect, 2019-09-22:
Hi,
Running v0.99 on Due with Gshield.

It works fine except changing the power mode for the steppers has no effect.

I would like t have them power down when not actually moving, currently they are permanently powered.

Thanks for any help

justinclift, 2019-10-02:
Did you get this figured out?

If not, which power mode settings have you tried? :smile:

arcadeperfect, 2019-10-02:
No, didn't figure it out!

I tried all three settings on all three motors, ie

$1pm=0
$1pm=1
$1pm=2
$1pm=3
$2pm=0

and so on. In all cases, the motors were permanently energized.

giseburt, 2019-10-02:
Just as a sanity check, were you able to confirm that the “Enable” pins for each motor is or is not changing when the power mode is changed?

-Rob

arcadeperfect, 2019-10-03:
Hi,
I checked and the enable pin behaves as expected. That is to say high when the motor is expected to be disabled, low when motor is expected to be enabled, and responds appropriately for all power modes. 
Motors remain energized at all times however!

arcadeperfect, 2019-10-03:
I just realised that the enable pins aren't even wired into the gshield PCB. Does this mean the [https://synthetos.myshopify.com/products/gshield-v5](url) doesn't support disabling motors?

ril3y, 2019-10-15:
I will check on this, but if that is the case then yes you would not be able to disable the motors.
@aldenhart does this sound right?

arcadeperfect, 2019-10-15:
As far as I can see it turns out that GRBL uses one global pin to disable all motors, whereas tinyG can disable them individually and uses pins that are not used by the gshield. 

However this means that pin 8 is presumably not used by tinyG.

So I guess the real subject of this issue should be: can tinyG be modified to pull pin 8 low if any of the motors are disabled? Unfortunately I don't have the knowledge to try to implement that myself!