synthetos/g2 #427: Striations in 3D Carvings

HartLabs, 2019-10-11:
I am having issues with the highly detailed 3D carvings.

I am getting striations in the raster direction that are not in the file, are not scallops. Just to say again, 100% sure they are not scallops. The same files did not produce the same striations when using GRBL.

I have the chordal tolerance set to 0. Have played around with feeds/speeds, max jerks, junction integration time.

If I want g2core to be controlling the machine in a way that it is hitting the points in a gcode file exactly, and moving with a straight line in between, what settings would that entail?

I understand that might be slow, but looking for the starting point before dialing back to where I get the trade off of precision and speed.

giseburt, 2019-10-12:
I’m sorry to hear that. To be sure, this is not an inherent problem in the firmware, as we’ve done extensive testing with 3D carving and 3d printing, both requiring detailed motion in all directions without slippage or striations.

Can you post some photos and perhaps some gcode that illustrates the issue? (GitHub allows you to attach files to issues.)

Chordal tolerance only applies to how arcs are handled, so I doubt that’s the issue, unless your gcode consists mostly of arcs.

There are several things that can change what we’re looking for here. Can you describe your setup more?

* What branch and hardware are you using? Is this a machine with one motor per axis, or do you have two motor on Y os such?
* What drivers, at what micro step level, etc? 
* Since you bring up GRBL, what all changed between that test and using g2core other than firmware? Is this the same mechanical machine, or a different one?
* I have to ask: Have you eliminated all possible mechanical issues? There’s no (perhaps newly introduced) backlash or flex in the mechanics? In particular, look for a single axis issue.
* How did you generate the “raster” gcode? (Raster being somewhat of a misnomer when related to gcode, so there are several techniques available to make raster-like input work with gcode.)

-Rob

HartLabs, 2019-10-12:
Appreciate the quick reply!

Machine started as a Shapeoko 3 XL, has the upgraded Beaver HDZ Z axis with an 800w water cooled spindle. So dual Y motors. Had no issue with this at that point, but wanted to try to increase machine speed without losing quality. Though g2core's motion profile might help.

For g2core it is now running off an arduino due, controlling 4x StepperOnline DM556T drivers, running at 48v instead of the stock 24v.

So the controller board, firmware, drivers and voltage is new.

{"r":{"sys":{"fb":100.26,"fbs":"100.26-10-g1ff7-dirty","fbc":"settings_shapeoko3_Custom.h","fv":0.99,"hp":3,"hv":0,"id":"0084-084a-0084-18c","jt":0.2,"ct":0,"sl":1,"lim":0,"saf":1,"m48e":1,"mfoe":1,"mfo":1,"mtoe":1,"mto":1,"mt":2,"spep":1,"spdp":0,"spph":1,"spdw":1,"ssoe":1,"sso":1,"cofp":1,"comp":1,"coph":0,"tv":1,"ej":1,"jv":4,"qv":0,"sv":1,"si":250,"gpl":0,"gun":1,"gco":1,"gpa":2,"gdi":0}},"f":[1,0,3]}

I am using the Master Branch current as of maybe 2 weeks ago. I had tried the Edge and was having a bunch of communication issues with Universal G-Code Sender so stayed with the Master.

Microstep level is 8

Everything mechanically seems as good as has ever been. Rigid, tensioned, trammed, clean. Motors are always powered on, not using the enable pins at the moment.

The carvings I make are terrain maps. I have custom software that generates the code from satellite point cloud data. Looking at it in a visualizer the paths all look perfect. This is confirmed as the striations change position when re-running the same code.

Attached an image. It shows 3 different passes one rough, and 2 finals that I was testing and stopped before they finished. You can see it best in the center pass, near the bottom. There are 2 places where the pass was especially out of sync. I am doing 0.125mm step overs with a 0.75mm radius bit here. In the past it would be practically smooth.

It might look like scallops/step-over but carving the same thing in GRBL did not produce the same results. I have done close to 100 of these by now. It does look like it falls under the 0.2mm step size (40mm/rotation (20 tooth, 2mm pitch ), 200 step motor). 

[image]
.

giseburt, 2019-10-12:
That extra info really, helps, thank you.

I would recommend trying Edge for this case. There are a couple of issues fixed in edge that may be tripping you up here.

Which leaves you with the issue of a working sender. Since I (personally) wasn’t aware of Universal gcode Sender working for g2 at all, I had no idea it might have issues with edge vs master. :) We’ll look into seeing what might be the problem there.

I would recommend upgrading to edge and using [CNCjs](https://cnc.js.org) at least to test and see if edge fixes this issue.

Please let us know either way.

HartLabs, 2019-10-12:
No change with the edge branch and CNCjs.

Downloaded CNCjs and tried loading one of my medium size files ~250mb and it can't handle it. My large files are close to 1GB. So will be a no go for what I am trying to do anyway.

Maybe should open a different ticket, but UGS seems to have some type of handshake issues with edge. Connects and the UGS buttons etc work but no console. Or reconnect and the console works but no buttons. Makes it pretty impossible to test and use. Seems to work fine with Master.

justinclift, 2019-10-12:
As a general thought, what are you using to generate the G-Code?

Wondering if it has the ability to generate specifically for TinyG / g2core, instead of for GRBL?

For example, with Fusion 360 it has post processors for both and the TinyG one is the correct one for g2core. :wink:

justinclift, 2019-10-12:
@giseburt UGS has supported g2core since last year.  It's on our recommended list, though it doesn't work well with the edge branch due to bugginess of our edge branch when they were testing.

@breiler (UGS developer) is on list of people to send gQuintic boards to.  Which reminds me, we need to double check people's addresses before sending them out due to our sending delay.  Some people have definitely changed address in the meantime. :wink:

justinclift, 2019-10-12:
>  I have custom software that generates the code ...

Oops, somehow didn't grok that bit. :frowning:

HartLabs, 2019-10-12:
Yeah nothing in the gcode should cause this. It is very basic, a header and then nothing but coordinates once it gets going.

Really seems to me without knowing what is going on in the code, just guessing, like it is something about changing directions on a microstep instead of a full step. With the machine losing that precision on occasion. Like moving +X to 100 full steps 3 micro steps, then starting to move -X and having it start at 101 full steps losing 5 microsteps etc.

HartLabs, 2019-10-12:
In order to keep working, I switched the machine back to GRBL this morning and the issue is gone with the same file, same sender.

As stated before the only differences are g2core/arduino/StepperOnlineDM556T/48v vs GRBL/Stock Shapeoko3 Board with Drivers/24v.

karoria, 2019-10-13:
@HartLabs
I also strongly doubt microstepping as you have pointed out. I recommend you to test the same program with g2core vs grbl but with same hardware, if possible.
I mean machine, motors, drives and operating voltage (24 or 48) should be same for one to one comparison.
It seems, any or combination of the above hardware elements are responsible for this behaviour. Any stepper will do perfect microstepping when not in load. But when running with load it can miss some steps. Only support you can add to your hardware is to increase operating current and voltage of motors to reduce the chance of missing steps.

justinclift, 2019-10-13:
Interestingly, I used to run my Shapeoko 3 (standard size) with g2core + Arduino Due too.  This issue didn't show up for me, though I don't remember the voltage I was using to drive the motors.  May have been 12v, but not 100% sure.

HartLabs, 2019-10-13:
Wish it was easy to test the components I have without buying and assembling more hardware. Would be nice if there was a GRBL build that ran on the due with matching pinouts but doesn't seem to be any way to run GRBL on the due at all from my searching.

Will have to order an uno and shield, wire up circuits for all the pull ups, connectors etc. Figure out the software build process for it. Not likely to happen for the next month have too much on my plate.

I can probably find some time to lower my power supply voltage to 24v and test sooner, that should eliminate interference. Which would leave it to a bad driver.

The nature of the problem seems like something else to me though. Like if it was loosing steps through interference or a bad driver I would expect the drift to be larger. As in it shouldn't stay within a narrow margin of error of the expected output. I would expect it to have lost 2 step in the same direction at some point. Also wouldn't expect the same type of error in both X and Y rasters.

justinclift, 2019-10-14:
Yeah, this is a weird problem. :man_shrugging:

giseburt, 2019-10-14:
The specific code that I expected to fix this problem somehow got dropped out of Edge in a merge. I’m working on putting it back but it’s part of a larger merge. I’ll report back here once I think it’s ready for testing (and where it is).

closed 2023-02-16 by ril3y