synthetos/g2 #428: Viable alternative to Due Board

mnesarco, 2019-10-21:
Hello friends, this project is amazing, I am using the edge branch in my DIY CNC with cncjs and everything works very well. I am planning to design CNC milling machines and start a small business around it, g2core is my first choice, but now that DUE is a discontinued board for a long time, I wonder if this project will also end soon or if it is possible to execute it on another 32 bit board. Did anyone succeed in running g2core on something else?

justinclift, 2019-10-21:
Looking at the Arduino website, the Due still seems to be available:

&nbsp; &nbsp; https://store.arduino.cc/usa/due

Looking in Australia (my present location), they're definitely available here too.

That being said, the Due is just the default board.  There have been a few other boards around which people have done things with, though I've not personally tried them.

There's also an "[official](https://github.com/synthetos/g2/wiki/gQuintic-Specs)" board in development (for a while, with several revisions so far), though it's not yet released.

With the Arduino Due discontinuation... looking around online it seems like it was announced to be discontinued a few years ago, when there was some Arduino brand confusing (competing camps or something from rough memory).  Seems to have been picked up again though, judging by the continued availability. :smile:

ril3y, 2019-10-21:
We should be releasing a g2 board for sale in the near future.

What is your timeline?

mnesarco, 2019-10-22:
Hi @justinclift, yes, it is available in the store, but as far as I know, they are no longer producing it. It is difficult to know because there is no official announcement of arduino.cc clarifying if they continued the production of the Due after the reunification.

@ ril3y I don't have an official timeline yet. I'm just designing and I love using g2core. I have heard about the quintic for a long time, but I have lost hope, it is good to know that it is still on its way, but I suspect it will be difficult (or expensive) to get one here in Colombia (South America). I have read about the teensy 3.6 and it looks great because have enough pins, but now that 4.0 was released and they decided to remove many pins, I doubt I can do with the teensy 4 what is possible with the Due, and probably the 3.6 will be discontinued soon because they are adding power but reducing io (a move that i don't understand).

The best part of using generic boards like Due or Teensy or STM32 is that we can get one anywhere at decent prices and we can design our own shields...

ril3y, 2019-10-22:
You should take a look at Motate project we have.  @Rob Giseburt
<rob@tenmilesquare.com>  is the author and its a generic hal abstraction
layer for G2 to be ported to other hardware.

BTW they announced that they will continue to sell the Due I think a few
months back.

Riley

mnesarco, 2019-10-22:
@ril3y Yes!, I have been reading the Motate code, but to be honest, it is far beyond my actual knowledge. One stupid question: In order to port g2core to another board (ie. Teensy 3.6) is it required to modify g2core's code AND Motate's code or only g2core?

BTW if it is safe to continue depending on the Due for the next years, there are no hurries to do any porting work.

closed 2020-01-13 by ril3y