synthetos/g2 #430: Stuck in run

mhlong10, 2019-11-12:
While running gcode the machine will stop and be stuck in the run state.  The only way out is hard reset.  No error code or way to get additional information.

I wrote my own simple gcode sender which seems to work fine, however while running the attached gcode file machine just stopped.  I decided to run the same file on CNCjs (version 1.9.20) to see if it was my sender.  It got stuck again.  

The only thing I have modified in the build is settings_default.h for my machine config.

I have a raspberry pi running connected via USB to a DUE running g2core.  I have a duplicate setup in my office although the DUE isn't connected to a shield.  I use this simply for testing software changes and to have a backup set of hardware just in case.  This has failed on both setups - which are running same version of g2core.

Plan buffer was at 5 and I was at line 5391 (not always same line) of the following attached file.

[3D Roughing 6.0mm Flat.zip](https://github.com/synthetos/g2/files/3834497/3D.Roughing.6.0mm.Flat.zip)

CNCjs 1.9.20 [TinyG]
Connected to /dev/ttyACM0 with a baud rate of 115200
{"r":{"fv":0.99,"fb":101.03,"fbs":"101.03-6-g7a25d-dirty","fbc":"settings_default.h","hp":"ArduinoDue","hv":"na","id":"0084-084a-18c6-5ad","msg":"SYSTEM READY"},"f":[1,0,1]}
feeder> {ej:1}
feeder> {jv:4}
feeder> {qv:1}
feeder> {sv:1}
feeder> {si:100}
{"r":{"ej":1},"f":[1,0,8]}
{"r":{"jv":4},"f":[1,0,8]}
{"r":{"qv":1},"f":[1,0,8]}
{"r":{"sv":1},"f":[1,0,8]}
{"r":{"si":100},"f":[1,0,10]}
feeder> {spe:n}
{"r":{"spe":null},"f":[1,100,9]}
{"err":{"code":100,"msg":"Unrecognized command or config name"}}
feeder> {spd:n}
{"r":{"spd":null},"f":[1,100,9]}
{"err":{"code":100,"msg":"Unrecognized command or config name"}}
{"sr":{"posx":0,"posy":0,"posz":0,"posa":0,"mpox":0,"mpoy":0,"mpoz":0,"mpoa":0,"stat":1,"vel":0}}
feeder> {spc:n}
{"r":{"spc":0},"f":[1,0,9]}
feeder> {sps:n}
{"r":{"sps":0},"f":[1,0,9]}
feeder> {com:n}
{"r":{"com":0},"f":[1,0,9]}
feeder> {cof:n}
{"r":{"cof":0},"f":[1,0,9]}
feeder> {sr:{stat:t,line:t,vel:t,feed:t,unit:t,coor:t,momo:t,plan:t,path:t,dist:t,admo:t,frmo:t,tool:t,posx:t,posy:t,posz:t,posa:t,posb:t,posc:t,mpox:t,mpoy:t,mpoz:t,mpoa:t,mpob:t,mpoc:t,spc:t,sps:t,com:t,cof:t}}
feeder> {sys:n}
feeder> {mt:n}
feeder> {pwr:n}
feeder> {qr:n}
feeder> {sr:n}
{"r":{"sr":{"stat":1,"line":0,"vel":0,"feed":0,"unit":1,"coor":1,"momo":4,"plan":0,"path":2,"dist":0,"admo":1,"frmo":1,"tool":0,"posx":0,"posy":0,"posz":0,"posa":0,"posb":0,"posc":0,"mpox":0,"mpoy":0,"mpoz":0,"mpoa":0,"mpob":0,"mpoc":0,"spc":0,"sps":0,"com":0,"cof":0}},"f":[1,0,206]}
{"r":{"sys":{"fb":101.03,"fv":0.99,"fbs":"101.03-6-g7a25d-dirty","fbc":"settings_default.h","hp":"ArduinoDue","hv":"na","id":"0084-084a-18c6-5ad","jt":0.75,"ct":0.01,"zl":0,"sl":0,"lim":0,"saf":0,"m48":1,"froe":0,"fro":1,"troe":0,"tro":1,"mt":2,"tv":1,"ej":1,"jv":4,"qv":1,"sv":1,"si":100,"gpl":0,"gun":1,"gco":1,"gpa":2,"gdi":0}},"f":[1,0,9]}
{"r":{"mt":2},"f":[1,0,8]}
{"r":{"pwr":{"1":0,"2":0,"3":0,"4":0}},"f":[1,0,9]}
{"r":{"qr":48},"f":[1,0,8]}
{"r":{"sr":{"stat":1,"line":0,"vel":0,"feed":0,"unit":1,"coor":1,"momo":4,"plan":0,"path":2,"dist":0,"admo":1,"frmo":1,"tool":0,"posx":0,"posy":0,"posz":0,"posa":0,"posb":0,"posc":0,"mpox":0,"mpoy":0,"mpoz":0,"mpoa":0,"mpob":0,"mpoc":0,"spc":0,"sps":0,"com":0,"cof":0}},"f":[1,0,8]}
{"sr":{"stat":1,"unit":1,"coor":1,"momo":4,"path":2,"admo":1,"frmo":1}}
>

breiler, 2019-11-12:
This sounds a lot like the problem I had in #367. I tried compiling the latest code and the problem is now gone (thanks @MitchBradley). 

The latest official build that I got to work when implementing the support for g2core in "Universal Gcode Sender" was 100.26.

mhlong10, 2019-11-12:
@breiler - Thanks for bringing this fix up.  Actually my build does have the fix for https://github.com/synthetos/g2/issues/367 - I pulled it so long ago I forget.

**Update** - I have run this file several times on CNCjs and it has stalled for a long period of time on a couple of runs (GUI and not machine - so I think g2core firmware is good).  I've narrowed it down to running on a wifi.  I connected the Raspberry pi to hardwired network and have had nothing but successful runs... no stalls.  **I'm going to give it a few more runs and if it continues to run fine I will close this.**  I've also run this on my sender successfully a few times... I think I might have an issue with pause/resume since a couple times I've seen it hang minutes after a pause/resume.

**Going off topic for a moment** - I'm going to start using CNCjs and customize and contribute to that project.  I really like my Web based gcode sender, it has been working flawlessly for over a year now and is customized just how I like, however it doesn't make sense to maintain that seeing how nice CNCjs is.  There are a couple of minor additions would like to see/implement if it makes sense:

1. Ability to enable a separate jog distance setting for Z axis.  Given the travel in Z axis is typically so much less than other axis, this avoids constantly modifying the jog distance for some use cases (like zeroing using paper method).  With the probe feature might not be doing this as often now. :)
2. Ability to remove (not just add) jog distances.  I typically only use x10 increments, others get in the way.
3. A reset button which can toggle a raspberry pi GPIO pin.  I see that CNCjs is very customizable and maybe there is a way to add a custom button or widget with the existing UI.

Here is a clip of my UI's jog control - not as condense or customizable as CNCjs but I find it very effective - maybe I'll be able to add this as a custom widget:
[image]

justinclift, 2019-11-12:
> The latest official build that I got to work when implementing the support for g2core in "Universal Gcode Sender" was 100.26.

Oops.  Good reminder.  We need to create a new build for people, now that fix is in and no-one's reported problems with it.

mhlong10, 2019-11-12:
I haven't been able to reproduce on multiple runs and using hardwired network removes any long pauses or apparent hangs in CNCjs as well.  I'm going to close this.

closed 2019-11-12 by mhlong10

MitchBradley, 2019-11-12:
@mhlong10 If you want to open an issue on cncjs/cncjs, we can discuss your ideas.  In particular, the GPIO thing is pretty easy to do with existing functionality.