synthetos/g2 #432: Current state of g2core SPI support on Arduino Due

tfeldmann, 2019-11-25:
Hello,
I trying to adapt the edge branch of g2core to my Arduino Due shield which controls four stepper driver (St L6474) via SPI.
I found the Motate submodule is many commits behind its master branch and it seems that in the meantime a lot of work has been done to support SPI on the sam3x (in Motate @ 41e5b92 the whole sam3x SPI code is `#if 0`ed). I tried compiling g2core with the current version of Motate but failed miserably.

So my question is: Is this feasable at all? Should I try to get the newest Motate version to run or use another SPI library?

giseburt, 2019-11-25:
There is a newer version of g2 that hasn’t been pushed yet (we’re still testing) that uses that newer Motate and will have the SPI support you’re looking for. I’ll leave this issue open until we push that code.

tfeldmann, 2019-11-25:
Thank you for the quick response. Looking forward to it 👌

tfeldmann, 2019-11-28:
Do you have a estimate when this version is available? Would be very interested in testing this!

karoria, 2019-11-28:
I would like to know if it addresses feedrate overrides. Desperately waiting for it.

giseburt, 2019-12-05:
I’m hoping to get this pushed in the next few weeks.

@karoria Feedrate overrides are in there, but need testing.

karoria, 2019-12-06:
Great news...Thanks a lot...