synthetos/g2 #437: Compile Error in AtmelStudio7

Vellrock, 2019-12-12:
Hello everyone,

I have an error during compile of g2core (edge) in Atmel Studio 7 on Windows 10 x 64.

Project was taken from GitHub Desktop, so I assume that it was most up to date.

Here are the Errors:
recipe for target 'win32/gcc-arm-none-eabi-5_4-2016q2-20160622-win32.zip' failed
recipe for target '../Motate/MotateProject/motate/../../Tools/win32/gcc-arm-none-eabi/bin' failed

I followed those instructions:
https://github.com/synthetos/g2/wiki/Compiling-G2-on-Windows-10-and-Atmel-Studio-7#git-for-windows

And I also found the same issue here:
https://github.com/synthetos/g2/issues/163

Any clue of how to solve this?

Best regards

p.s.: log from atmel studio 7

------ Build started: Project: g2core, Configuration: gShield ARM ------
Build started.
Project "g2core.cppproj" (default targets):
Target "PreBuildEvent" skipped, due to false condition; ('$(PreBuildEvent)'!='') was evaluated as (''!='').
Target "CoreBuild" in file "C:\Program Files (x86)\Atmel\Studio\7.0\Vs\Compiler.targets" from project "C:\Users\vellrock\Documents\GitHub\g2\g2core\g2core.cppproj" (target "Build" depends on it):
	Task "RunCompilerTask"
		Shell Utils Path C:\Program Files (x86)\Atmel\Studio\7.0\shellUtils
		C:\Program Files (x86)\Atmel\Studio\7.0\shellUtils\make.exe -C "C:\Users\vellrock\Documents\GitHub\g2\g2core" -f "Makefile" BOARD=gShield CONFIG=ShapeokoDualY COLOR=0 VERBOSE=1 
		make: Entering directory 'C:/Users/vellrock/Documents/GitHub/g2/g2core'
		"Found that we're in Atmel Studio"
		process_begin: CreateProcess(NULL, git describe --abbrev=4 --dirty --always --tags, ...) failed.
		make: ../Motate/MotateProject/motate/Motate.mk:302: pipe: No such file or directory
		process_begin: CreateProcess(NULL, git describe --abbrev=0 --tags, ...) failed.
		make: ../Motate/MotateProject/motate/Motate.mk:303: pipe: No such file or directory
		The system cannot find the path specified.
		Installing the necessary tools...
		cd ../Motate/MotateProject/motate/../../Tools && make "ARCH=gcc-arm-none-eabi"
		make[1]: Entering directory 'C:/Users/vellrock/Documents/GitHub/g2/Motate/Tools'
		The system cannot find the path specified.
		The gcc-arm-none-eabi tools need installed. Downloading win32/gcc-arm-none-eabi-5_4-2016q2-20160622-win32.zip and installing now.
		mkdir -p win32 && cd win32 && curl -OL "https://launchpad.net/gcc-arm-embedded/5.0/5-2016-q2-update/+download/gcc-arm-none-eabi-5_4-2016q2-20160622-win32.zip"zip"
		A subdirectory or file -p already exists.
		Error occurred while processing: -p.
		A subdirectory or file win32 already exists.
		Error occurred while processing: win32.
C:\Users\vellrock\Documents\GitHub\g2\g2core\Makefile(143,1): error: recipe for target 'win32/gcc-arm-none-eabi-5_4-2016q2-20160622-win32.zip' failed
		make[1]: *** [win32/gcc-arm-none-eabi-5_4-2016q2-20160622-win32.zip] Error 1
		make[1]: Leaving directory 'C:/Users/vellrock/Documents/GitHub/g2/Motate/Tools'
C:\Users\vellrock\Documents\GitHub\g2\Motate\MotateProject\motate\Motate.mk(404,1): error: recipe for target '../Motate/MotateProject/motate/../../Tools/win32/gcc-arm-none-eabi/bin' failed
		make: *** [../Motate/MotateProject/motate/../../Tools/win32/gcc-arm-none-eabi/bin] Error 2
		make: Leaving directory 'C:/Users/vellrock/Documents/GitHub/g2/g2core'
	Done executing task "RunCompilerTask" -- FAILED.
Done building target "CoreBuild" in project "g2core.cppproj" -- FAILED.
Done building project "g2core.cppproj" -- FAILED.

Build FAILED.
------ Skipped Build: Project: Motate, Configuration: Stub ARM ------
Project not selected to build for this solution configuration 
========== Build: 0 succeeded or up-to-date, 1 failed, 1 skipped ==========

giseburt, 2019-12-12:
Do you have the `Motate` subdirectory? If not see the notes at https://github.com/synthetos/g2/wiki/Getting-Started-with-g2core-Development#cloning-the-repo about submodules.

Vellrock, 2019-12-14:
Dear Giseburt.
Thank you for answer and sorry for my late reply.

Yes, do have a motate sub-directory. 
But problem was solved by installing Git-2.24.1.2-64-bit.exe. I didn't now about this tool before. So, your link has helped.

I have compiled a project, uploaded to ArduinoDUE with Bossac.exe. Works like a charm.

closed 2019-12-14 by Vellrock